john30/ebusd #107: ebusd-3.0_armhf_mqtt1.deb still seems to contain the pre release binary

mike-bike, 2017-09-20:
Hi John, firstly many thanks for your effort and the great program you offer. I was trying to upgrade to v3.0 but found that https://github.com/john30/ebusd/releases/download/v3.0/ebusd-3.0_armhf_mqtt1.deb still seems to contain the pre release ebusd 3.0.595c7c0

Same apply to the other armhf.deb. The wheezy package seems to have v3.0 packaged.

I am also unable to locate most recent config files. The config package 2.1.1 seems to be unchanged but ebusctl i  tells me:

version: ebusd 3.0.595c7c0
update check: revision v3.0 available, broadcast.csv: different version available, vaillant/15.ui.csv: different version available, vaillant/25.solsy.hwc.csv: different version available, vaillant/26.solsy.hc.csv: different version available, vaillant/bai.0010004121.inc: different version available, vaillant/broadcast.csv: different version available, vailla

Many thanks & Kind regards
Michael

john30, 2017-09-21:
The version number of ebusd in those packages didn't change, only the systemd configuration was replaced.
For the packages built on initd, there was no change necessary anyway.
[2.1.1](https://github.com/john30/ebusd-configuration/releases/tag/v2.1.1) is the most current release of the configuration files. I'm already working on the first 3.0 version but this will take some more time.

mike-bike, 2017-09-21:
John, many thanks for the quick response. I thought Jackfruit would be the first 3.0 release. I may have this misunderstood. Anyway, I was just wondering how to make ebusd happy as it recognizes that a newer version is available and that some of the config files are outdated. Besides that, it is working fine and I am currently trying to figure out how to get MQTT working. Currently I only could see the few global topics coming through but no circuits (yet)... Right now my openHAB uses http.
I am looking forward to seeing the next version - Mike

john30, 2017-09-23:
yes, Jackfruit is the first 3.0 version. Then I probably misunderstood you. So what did you mean by:

> still seems to contain the pre release ebusd 3.0.595c7c0

595c7c0 is the last commit on the v3.0 release tag.
For the update check, the message should be fixed in the meanwhile, as the version info on the update check website was not correct back then.
For the configuration files, the most up-to-date version is available in the repo only, no new release yet.

mike-bike, 2017-09-26:
I am too dumb:-) I have done
/usr/bin/git clone https://github.com/john30/ebusd-configuration.git
then make_debian and then  sudo dpkg --force-overwrite -i ebusd-configuration-2.1.edb69c4-en_all.deb. ebusd restart and still get with ebusctl i
version: ebusd 3.0.595c7c0
update check: revision v3.0 available, broadcast.csv: different version available, memory.csv: different version available, vaillant/25.solsy.hwc.csv: different version available, vaillant/50.solsy.mc.csv: different version available, vaillant/bai.0010004121.inc: different version available, vaillant/broadcast.csv: different version available, vaillant/ec.so

Anyway, anything else seems to be running fine, so I am good. Do you have any idea how to measure gas throughput or current burner power? ebusctl read ExternGasvalve seems to provide a numeric value. Do you have a link for Vaillant FAQs or help?

john30, 2017-09-28:
> Do you have any idea how to measure gas throughput or current burner power?

I'm afraid I don't have this information. There was a thread long time ago in the [forum](https://forum.fhem.de/index.php/topic,29737.0.html) regarding that as far as I remember.

closed 2017-09-28 by john30

andig, 2017-09-28:
>Do you have any idea how to measure gas throughput or current burner power?

@mike-bike for my EcoTec Plus with 14.6kWh I'm using this formula for burner power:

    1.46*HcVent()*ModulationTempDesired()

The `HcVent` is only there for differentiating heating power from hot water, it's not relevant for the burner power as such. `ModulationTempDesired` is the current degree of modulation of the burner.

mike-bike, 2017-10-04:
many thanks for your responses.

**@john30** - I scanned the forum article but was not (yet) able to find what I was looking for. But it has more than 160 pages, so I may have missed that. 

**@andig** - I have seen ModulationTempDesired() to vary between 16.0 (min) and 51.3 (max) with my  23.4KWh (21KWh) EcoTec Plus. What confuses me is that ModulatioTempDesired gives 16.0 if the burner is off. I do not get any value for HcVent (ERR:element not found). Do you have any further information about that subject? I'd appreciate if you could point me into the right direction. 

As this discussion is no more than off-topic I'd prefer an PM instead.

Regards,
Michael

andig, 2017-11-07:
@mike-bike replying here as the issues was my first resource for finding a solution, too- hope this is ok.

> I have seen ModulationTempDesired() to vary between 16.0 (min) and 51.3 (max) with my 23.4KWh (21KWh) EcoTec Plus. 

With my 14kW BAI I'm getting between 19 and 100 which is in line with the lowest modulation limit of 19%. 

    14kW x 0.19 = 2.66kW

> What confuses me is that ModulationTempDesired gives 16.0 if the burner is off. I do not get any value for HcVent (ERR:element not found). 

I have the same behaviour. `ModulationTempDesired` is either the "would-if-active" value or the last active, not sure. That's why I'm using `HcVent` to determine if the value is valid. What I forgot to mention is that `HcVent` really doesn't exist (only in my node-red flow) but is part of the 3-stage vent in `Status01`:

	pi@heizung:~ $ ebusctl r Status01
	21.5;21.5;4.250;-;28.0;off

For the last byte, `on` means heating while `4`is warm water (also see https://github.com/john30/ebusd-configuration/issues/38#issuecomment-334060414).

Hope this helps...