john30/ebusd #108: why json value are incomplete?

modenet, 2017-09-23:
I'm testing json output with a VRC700: 
in json output (http://localhost:8080/data) where is no reference to 700; 
instead if I use your script to read all valus ebusd/contrib/scripts/readall.sh I find all 700's values.

in json the 1st colums are: bai, broadcast, scan.08, scan.15, global.
thanks

john30, 2017-09-26:
please post the ebusd log and the curl line and output from the request

modenet, 2017-09-26:
[json.txt](https://github.com/john30/ebusd/files/1332446/json.txt)
[ebusd.log](https://github.com/john30/ebusd/files/1332481/ebusd.log)

added json output (http://xxx/data) and log (--lograwoutput). thanks

john30, 2017-09-28:
it's not incomplete, it just only covers the messages, that were sent/received already. Please [see here for additional request parameters](https://github.com/john30/ebusd/wiki/3.2.-HTTP-client) that allow you to e.g. include all messages as well, even those that were not sent yet. Furthermore you can change a message to be polled automatically by ebusd using the request parameters and if you do so, then some time later the corresponding message values will be available.

closed 2017-09-28 by john30

modenet, 2017-09-28:
thanks, I saw your wiki page for json request but I can't find any parameter that enable inclusion of all messages... can you help me? thanks

modenet, 2017-09-29:
using http://xx.xx.xx.xx:8889/data/700 I obteined  only
{
 "global": {
  "version": "3.0.v3.0-9-gb9c69f0",
  "updatecheck": "revision v3.0 available, broadcast.csv: different version available, vaillant/bai.308523.inc: different version available, vaillant/broadcast.csv: different version available, vaillant/errors.inc: different version available, vaillant/hcmode.inc: different version available",
  "signal": true,
  "symbolrate": 24,
  "maxsymbolrate": 148,
  "qq": 49,
  "reconnects": 0,
  "masters": 3,
  "messages": 211,
  "lastup": 0
 }
}

modenet, 2017-09-29:
even every read on 700 fail:
ebusctl read -c 700 OutsideTempAvg
ERR: element not found

but in the readall.sh output I can see:
...
700 OpMode = off
700 OpModeEffect = alle
700 OutsideTempAvg = 6

john30, 2017-10-04:
that's weird. which version of the CSV do you use?
Anyway, the json output only reveils data from messages, that were already actively sent or were received while sent from another participant. So you'd have to e.g. use the contrib/scripts/readall.sh for ensuring all messages are read.
The 3 messages listed above might all be write messages, which are excluded by default from the JSON output. you can use ?write=true to get those as well

modenet, 2017-10-06:
resolved using lastest version of CSV