john30/ebusd #111: Problem with HTTP request

alexob, 2017-10-13:
Please help me to understand how can I requst by HTTP, for example, Date from my VCR 470?

I set up the work through the command line (works fine)...

> pi@raspberrypi:~ $ ebusctl r -f Date
14.10.2017

But when I trying to do HTTP request it does not work.

http://localhost:8888/data/470/Date
I got error "ERR_INVALID_HTTP_RESPONSE"
Port  and Path by default.

> pi@raspberrypi:~ $ ebusctl info
version: ebusd 3.0.595c7c0
update check: revision v3.0 available, broadcast.csv: newer version available, vaillant/08.bai.csv: newer version available, vaillant/15.470.csv: newer version available, vaillant/bai.308523.inc: newer version available, vaillant/broadcast.csv: different version available, vaillant/errors.inc: newer version available, vaillant/hcmode.inc: newer version avail
signal: acquired
symbol rate: 35
max symbol rate: 133
reconnects: 0
masters: 3
messages: 451
conditional: 3
poll: 0
update: 8
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0518;HW=7401", loaded "vaillant/bai.308523.inc" ([HW=7401]), "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=47000;SW=0130;HW=6002", loaded "vaillant/15.470.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd

I think I do anything wrong, but what? Please help.
Also, I manually created folder /var/ebusd/html and and copied there the contents of the folder \ebusd-master\contrib\html   (maybe it not right).

john30, 2017-10-14:
the /var/ebusd/html folder is not necessary for using the HTTP API under /data/.
What do you use to retrieve from the HTTP port?
When I issue the command `curl http://localhost/data/broadcast/datetime` then ebusd sends an answer like this:

```
{
 "broadcast": {
  "messages": {
   "datetime": {
    "name": "datetime",
    "passive": true,
    "write": false,
    "lastup": 1507974612,
    "zz": 254,
    "fields": {
     "outsidetemp": {"value": 12.875},
     "time": {"value": "11:50:00"},
     "date": {"value": "14.10.2017"}
    }
   }
  }
 },
 "global": {
...
 }
}

alexob, 2017-10-16:
Hi John, Thanks for your answer.
In my case it doesn't work.

> pi@raspberrypi:~ $ curl http://localhost:8888/data/470/Date
ERR: command not found
^C
pi@raspberrypi:~ $ ebusctl r Date
16.10.2017
pi@raspberrypi:~ $

I tried to enter url in the command line: "**curl http://localhost:8888/data/470/Date**"  and in the browsers.
I got  "ERR: command not found". But it work when I enter "**ebusctl r Date**"

The command Date present in the command list. I saw it after this:

> pi@raspberrypi:~ $ ebusctl find
470 ACTOstorDetected = no data stored
470 actoSTOROPMode = no data stored
....
470 currenterror = no data stored
470 Date = 16.10.2017
470 DcfIsValid = no data stored
470 DisplayedHc1RoomTempDesired = no data stored
470 DisplayedHwcStorageTemp = no data stored
470 DisplayedRoomTemp = no data stored
470 EepromUpdateActive = no data stored
....

Maybe I should to copy csv files (like bai.csv etc) to other folder. Look like the it can not find the files with commands.

john30, 2017-10-17:
please post the result of `curl http://localhost:8888/data/470/Date?raw&write&def&full/verbose`

alexob, 2017-10-17:
> please post the result of curl http://localhost:8888/data/470/Date?raw&write&def&full/verbose

```
pi@raspberrypi:~ $ curl http://localhost:8888/data/470/Date?raw&write&def&full/verbose
[2] 28168
[3] 28169
[4] 28170
-bash: full/verbose: Нет такого файла или каталога
usage: write user [tty]
[3]-  Выход 1            write
-bash: def: команда не найдена
pi@raspberrypi:~ $ ERR: command not found
```

> the csvs are already loaded as otherwise the message would not be available through ebusctl find.
> could you check if two ebusd services are running? ps aux|grep ebusd

```
pi@raspberrypi:~ $ ps aux|grep ebusd
root      1088  0.3  0.2  62836  2840 ?        Ssl  окт15   7:37 /usr/bin/ebusd --scanconfig
pi       28176  0.0  0.0   4368   540 pts/1    S+   09:07   0:00 grep --color=auto ebusd
pi@raspberrypi:~ $
```

alexob, 2017-10-17:
Yes, csvs loaded I think

john30, 2017-10-18:
you need to escape the & char in the request or put everything in quotes:
`curl 'http://localhost:8888/data/470/Date?raw&write&def&full/verbose'`

alexob, 2017-10-18:
```
pi@raspberrypi:~ $ curl 'http://localhost:8888/data/470/Date?raw&write&def&full/verbose'
ERR: command not found

^C
pi@raspberrypi:~ $ curl "http://localhost:8888/data/470/Date?raw&write&def&full/verbose"
ERR: command not found
```

john30, 2017-10-18:
sorry, typo:
`curl 'http://localhost:8888/data/470/Date?raw&write&def&full&verbose'`

alexob, 2017-10-18:
```
pi@raspberrypi:~ $ curl 'http://localhost:8888/data/470/Date?raw&write&def&full&verbose'
ERR: command not found

^C
pi@raspberrypi:~ $ curl "http://localhost:8888/data/470/Date?raw&write&def&full&verbose"
ERR: command not found
```

alexob, 2017-10-19:
Hello John,
How can I change the Http Port to 8889 for example?
As I see from wiki, port for command line is 8888 (by default), therefore when I called "curl http://localhost:8888/data/470/Date" I tried to get data with incorrect port.

As I understand, for Http requst, I should use other port. By default HttpPort = 0. So, I should change the port for activate Http functionallity, right?

My ebusd is working as a service, I started it as

`pi@raspberrypi:~ $ service ebusd start`

I tried to change HttpPort with

`pi@raspberrypi:~ $ ebusd --httpport=8889`

but now after run command

```
pi@raspberrypi:~ $ curl http://localhost:8889/data/470/Date
curl: (7) Failed to connect to localhost port 8889: В соединении отказано
```
got the error "Connection refused"

Please, give me instruction how do you start work with Http step-by-step and what the settings you did.

Thanks. 
(sorry for my English) :)

john30, 2017-10-21:
okay that explains it. you have to active the ebusd http port by adding e.g. "--httpport=8889" to /etc/default/ebusd

alexob, 2017-10-21:
Thank you, John.
It work now!

I changed line  to
EBUSD_OPTS="--scanconfig --httpport=8889"

john30, 2017-10-21:
ok great

closed 2017-10-21 by john30