john30/ebusd #112: Question: what is the right way to parse data

guillaumezin, 2017-10-15:
Hello,

I'm trying to write a plugin for Domoticz to bridge to your ebusd daemon. What would be the best way to grabe data types ?

Right now, I'm connecting using telnet and I do a "find -F circuit,id,name,type,comment,fields -r -w", but I don't understand how to interpret data, fields length is not consistent, with ebusctl, for instance, I have something like this:

localhost: find -f HcMc1ConfigCPLP
r,f47,HcMc1ConfigCPLP,BKMK1ConfigCPLP,,15,b509,0d7d00,,s,UCH,,0=off;1=CP;2=LP,,configuration of the VR61 relay-output

localhost: find -f Hc2TempDesiredB50418
r,f47,Hc2TempDesiredB50418,HC2_Desired_TempB50418,,15,b509,0d3200,temp1,s,D1C,,°C,current flow temperature setpoint of the second heating circuit

15 fields on first command, 14 fields on second one. Why this difference? What is your advice to correctly parse data type?

Thank you.

john30, 2017-10-15:
I'm afraid this was a little bug in formatting. fixed with 46078f8

closed 2017-10-15 by john30

guillaumezin, 2017-10-15:
Thank you. For my purpose, I think it will be easier to parse JSON to get fields descriptions. Is it possible to write only one field when there is many, or is it mandatory to write all fields at once ?

john30, 2017-10-16:
you can only write a whole message at once which means all fields of that message as of today. this is a planned feature though but no estimate when it will be available.

commit 46078f8c (2017-10-15): fix for wrong dump of value list and constant fields (fixes #112)