john30/ebusd #113: Improve mqtt support

scantineau, 2017-10-16:
Hello,

Could be nice to update mqtt topic when using the read command (nc or ebusctl, ...)

I suggest an option to the read command such as t (topic) or u (update...).

With this option, we could improve the readall.sh script too.

Thanks

john30, 2017-10-17:
not sure I understand what you mean.
the topics are updated whenever somebody issues a corresponding command and also when you append /get to an mqtt request.

scantineau, 2017-10-19:
Ok I understand. There is a delay between an ebusctl r and the mqtt update.

Anyway, I thought mqtt implementation was the problem but after more tests, my problem is elsewhere...

I don't know if you prefer to open a new thread...

It seems that when heating starts, pump starts too (logic...) but the value is not read "automatically" by ebusd (/var/log/ebusd.log). 
I mean I can see it if I go in the technician menu on my 700 but I need to initiate a read with ebusd (or /get) for the value to be updated in my mqtt topic.

Any idea why ?

john30, 2017-10-21:
you can just use the polling feature of ebusd for that, see here: https://github.com/john30/ebusd/wiki/4.1.-Message-definition#message-definition

scantineau, 2017-10-21:
ebusd is really amazing... You did a really great job.

I think that (almost?) everything is in the wiki but the bottleneck for end users (or idiots like me ;-) ) is the comprehension of it. 

A last thing : is it possible to change this "polling value" without restarting ebusd ?

john30, 2017-10-21:
thanks :-)
yes, you can pass the desired poll priority to the read command, see [read command](https://github.com/john30/ebusd/wiki/3.1.-TCP-client-commands#read).
however, this poll prio change is not made permanent, i.e. when you restart ebusd these prios will be lost

scantineau, 2017-10-21:
Yep but it's possible !

Ok thanks a lot for your help !

closed 2017-10-21 by scantineau