john30/ebusd #115: How to write one field?

biohazardxxx, 2017-10-22:
Hi,

I couldn't find a way on the wiki to write to a field directly. Is that somehow possible?
For example I want to write only to hwc Timer.Monday from.0 
The write command would take from.0 as value since syntax is: write -c CIRCUIT NAME [VALUE[;VALUE]*]

john30, 2017-10-22:
you always have to pass all fields of a write message for the time being. this is subject to be enhanced in future, but not on the roadmap yet as this implies reading the corresponding read message immediately before sending the write message, combining the current read values with the field(s) to set, and then sending the actual write message on the bus. somewhat complex and i'm not sure if its a good idea at all

biohazardxxx, 2017-10-22:
Ok good to know that. Thanks for your reply and great work!

closed 2017-10-22 by biohazardxxx