john30/ebusd #117: Can not write wi  command

alexob, 2017-10-26:
Hello, John

I have CalorMatic 470.  
I am trying to send write command HwcCircuitActive marked as "r;wi", but I get an error

```
pi@raspberrypi:~ $ ebusctl write -c 470#install HwcCircuitActive yes
ERR: element not found
```

I know that the command is present in the list, because I can read 

```
pi@raspberrypi:~ $ ebusctl r -f HwcCircuitActive
no
```

What is wrong in my command line?

Thanks

alexob, 2017-10-26:
I have checked and I see that this command not exists in the list of "write" command

> pi@raspberrypi:~ $ ebusctl find -w
470 ccTimer.Friday = no data stored
470 ccTimer.Monday = no data stored
470 ccTimer.Saturday = no data stored
470 ccTimer.Sunday = no data stored
470 ccTimer.Thursday = no data stored
470 ccTimer.Tuesday = no data stored
470 ccTimer.Wednesday = no data stored
470 Date = no data stored
470 Hc1BankHolidayEndPeriod = no data stored
470 Hc1BankHolidayStartPeriod = no data stored
470 Hc1DayTemp = no data stored
470 Hc1HolidayEndPeriod = no data stored
470 Hc1HolidayRoomTemp = no data stored
470 Hc1HolidayStartPeriod = no data stored
470 Hc1NightTemp = no data stored
470 Hc1OPMode = no data stored
470 Hc2BankHolidayEndPeriod = no data stored
470 Hc2BankHolidayStartPeriod = no data stored
470 Hc2HolidayEndPeriod = no data stored
470 Hc2HolidayStartPeriod = no data stored
470 HcName1 = no data stored
470 HcName2 = no data stored
470 hcTimer.Friday = no data stored
470 hcTimer.Monday = no data stored
470 hcTimer.Saturday = no data stored
470 hcTimer.Sunday = no data stored
470 hcTimer.Thursday = no data stored
470 hcTimer.Tuesday = no data stored
470 hcTimer.Wednesday = no data stored
470 HwcTempDesired = 45.0
470 hwcTimer.Friday = no data stored
470 hwcTimer.Monday = no data stored
470 hwcTimer.Saturday = no data stored
470 hwcTimer.Sunday = no data stored
470 hwcTimer.Thursday = no data stored
470 hwcTimer.Tuesday = no data stored
470 hwcTimer.Wednesday = no data stored
470 LanguageVariant = no data stored
470 LcdContrastValue = no data stored
470 OutsideTempOffset = no data stored
470 RoomTempCorrection = no data stored
470 SummerWinterTimeAdjust = no data stored
470 Time = no data stored
broadcast ident = no data stored
broadcast queryexistence = no data stored

But in the file /etc/ebusd/470.csv  this command marked as "r;wi".

Please explain how can I write data to ebus with this command.

john30, 2017-10-27:
starting with ebusd 3.0, the "install" suffix is subject to the ACL. So you'd either have to define username and secret with [--aclfile=...](https://github.com/john30/ebusd/wiki/2.-Run#daemon-options) or set the [--accesslevel=*](https://github.com/john30/ebusd/wiki/2.-Run#daemon-options) to allow everything to every connected client

alexob, 2017-10-27:
Thanks. It work now. :)
One more question. 
Is it possible to track changes to any value in the cache? For example, changing the activity of the boiler when user changed it in CalorMatic.
Now I check statuses from my app by timer, and may be it's not quite right.

john30, 2017-11-06:
you can use the "listen" command on an interactive ebusctl sesstion for getting notified of changed values

closed 2017-11-06 by john30