john30/ebusd #118: Can't put log options into /etc/default/ebusd

greuff, 2017-10-29:
I'd like to put the logging options (--log) into /etc/default/ebusd but as soon as I do that ebusd won't start anymore. I tried:

```
EBUSD_OPTS="--scanconfig -d /dev/ttyEBUS -c /etc/ebusd --log='all notice' --log='update error'"
```
(and some other combinations of single quotes, double quotes, backslash escaped double quotes etc.) but ebusd fails to start either way (via systemctl start ebusd), until I remove both the --log options again. What is the correct syntax to put them in EBUSD_OPTS?

log output:

```
Oct 28 16:23:02 fhem ebusd[5616]: ebusd: invalid log
Oct 28 16:23:02 fhem ebusd[5616]: Try `ebusd --help' or `ebusd --usage' for more information.
Oct 28 16:23:02 fhem systemd[1]: ebusd.service: control process exited, code=exited status=64
Oct 28 16:23:02 fhem systemd[1]: Failed to start ebusd, the daemon for communication with eBUS heating systems..
Oct 28 16:23:02 fhem systemd[1]: Unit ebusd.service entered failed state.
```

When I start ebusd with those options on the command line it works.

I'm using the ebusd-armhf deb package from release, version 3.0

john30, 2017-11-06:
I would expect `EBUSD_OPTS="--log=\"all notice\""` to work, but I need to check that.

greuff, 2017-11-06:
That's what I tried first, didn't work

jjakob, 2018-02-25:
I've encountered the same issue, so I spent some time trying to solve the issue.
So far I haven't been able to fix using EnvironmentFile with quotes passed in the variable at all.
Trying all combinations of single and double quotes around the whole variable string or after --log,
single, double, triple (even tried quadruple) escaping "s with backlashes, and so on.
Basically I think systemd's parameter expansion and quote stripping is broken, or at least the lack of precise documentation makes it almost impossible to understand. It may even be deserving of a bug report. Some related issues: https://github.com/systemd/systemd/issues/6035 https://github.com/systemd/systemd/issues/624 https://github.com/systemd/systemd/issues/5538 https://github.com/systemd/systemd/issues/4416

So giving up on using the EnvironmentFile, I created this unit drop-in:
```
[Service]
ExecStart=
ExecStart=/bin/sh -c '\
/usr/bin/ebusd --scanconfig --localhost \
--mqttport=1883 --mqttuser=ebusd --mqttpass=pass --mqttjson \
--log="all notice" --log="update error" \
'
```
Place this in /etc/systemd/system/ebusd.service.d/10-ebusd_opts.conf (create the ebusd.service.d directory) where it overrides the default ExecStart line. Tune the parameters to your liking, this is just the example I'm using. /etc/default/ebusd is now superfluous.
Note that wrapping the command in /bin/sh -c was required to get it working, without it the parameters still weren't passed correctly.

P.S. Now that I think of it, something like "--log=all notice", "--log=\"all notice\"" or "--log=\\\"all notice\\\"" might work in the original EnvironmentFile if anybody's up to test it.

sneakin, 2019-03-19:
Playing with afuse. Can't create a service for it.

`
Environment="MOUNT=/usr/bin/sshfs -o default_permissions,auto_unmount,reconnect,
ServerAliveInterval=1 %r %m"
Environment="UMOUNT=/usr/bin/fusermount3 -u %m"
#ExecStart=/usr/bin/afuse -f -o mount_template=\x22${MOUNT}\x22 -o unmount_templ
ate=\x22${UMOUNT}\x22 /%I
ExecStart=/bin/sh -c '\
/usr/bin/afuse -f -o \x22mount_template=/usr/bin/sshfs -o default_permissions,au
to_unmount,reconnect,ServerAliveInterval=1 %%r %%m\x22 -o \x22unmount_template=/
usr/bin/fusermount3 -u %%m\x22 /%I \
'
ExecStop=/usr/bin/fusermount3 -u /%I
`

The mount_template option's value, even with quotes and environment variable, gets split.

systemd 241.7-2-arch