john30/ebusd #119: how to

modenet, 2017-11-02:
hi, in earlier version it was easy to set potentiometer  correctly because I could see "aa" and adjust the screw:

015 - 12 - 24  12 : 07 : 16.955  [ bus  notice ]  < aa 
2015 - 12 - 24  12 : 07 : 17.001  [ bus  notice ]  < aa 
2015 - 12 - 24  12 : 07 : 17.047  [ bus  notice ]  < aa 
2015 - 12 - 24  12 : 07: 17.093  [ bus  notice ]  < aa 
2015 - 12 - 24  12 : 07 : 17.097  [ bus  notice ]  < 10 
2015 - 12 - 24  12 : 07 : 17.102  [ bus  notice ]  < 50 
2015 - 12 - 24  12 : 07 : 17.107  [ bus  notice ]  < b5
2015 - 12 - 24  12 : 07 : 17.111  [ bus  notice ]  < 04

now with raw log actived is a little difficult:
2017-11-02 11:47:15.444 [bus info] poll cmd: 3115b509030d4438
2017-11-02 11:47:15.445 [bus notice] >3115b509030d44303a<0002790131>00
2017-11-02 11:47:15.588 [bus info] poll cmd: 3115b509030d4440
2017-11-02 11:47:15.590 [bus notice] >3115b509030d443832<000282018b>00
2017-11-02 11:47:15.687 [bus notice] >31<10f3
2017-11-02 11:47:15.838 [bus notice] <1008b511010189000934340080ffff0000ff9d00
2017-11-02 11:47:15.928 [bus notice] <7050b50401012000091302000000810000001100
2017-11-02 11:47:16.467 [bus notice] <0364b5120202fe98
2017-11-02 11:47:16.533 [bus notice] <1026b50504270018004800000000
2017-11-02 11:47:16.627 [bus notice] <3753b5040101a80009150300000081000000bf00
2017-11-02 11:47:16.705 [bus notice] <1752b504010d61000500008801135000
2017-11-02 11:47:16.716 [bus notice] >31<1021
2017-11-02 11:47:16.761 [bus notice] <ff
2017-11-02 11:47:16.945 [bus notice] <1053b5040101f40009150300000081000000bf00
2017-11-02 11:47:17.087 [bus notice] <1752b50401325e000a003206030100b00801001c00
2017-11-02 11:47:17.170 [bus notice] <1026b505082b0f1700000000806100000000
2017-11-02 11:47:17.264 [bus info] poll cmd: 3115b509030d4448
2017-11-02 11:47:17.265 [bus notice] >3115b509030d44404a<0002f500e3>00
2017-11-02 11:47:17.338 [bus notice] <1752b505042d000001e100000000
2017-11-02 11:47:17.430 [bus info] poll cmd: 3115b509030d4450
2017-11-02 11:47:17.431 [bus notice] >3115b509030d444842<0002b200dc>00
2017-11-02 11:47:17.578 [bus info] poll cmd: 3115b509030d4458

is possible to have an option to restore that function?

john30, 2017-11-02:
You can simply use --lograwdata=bytes to get the old behaviour

modenet, 2017-11-03:
great, thanks

closed 2017-11-03 by modenet