john30/ebusd #124: async bus communication

danielkucera, 2017-11-04:
Hi,

I'm trying to to use esp8266 for communication on ebus with ebusd as controller. I would be grateful for an option to use async communication: command from ebusd would be sent to esp as whole (without waiting for sync symbol). esp would wait for sync on bus, ten correctly send the command (arbitrating possible conflicts) and ack client response. Everything received on bus would be sent back to ebusd as usual. To get an idea here's lua code for esp https://github.com/danielkucera/esp8266-ebus/blob/master/program.lua 

I tried to understand ebusd code but it's quite complex. Would it be feasible to implement this change?

john30, 2017-11-04:
This is actually one thing that might be added to  https://github.com/john30/ebusd-esp in future

danielkucera, 2017-11-04:
But I still don't see source code there so this is not acceptable for me.

john30, 2017-11-06:
as already mentioned on another discussion, there are no plans of releasing sources yet. This might change in future, but for the time being I wont.

closed 2017-11-06 by john30