john30/ebusd #126: --lograwdata=bytes is silent in combination with -l /var/tmp

nuccleon, 2017-11-08:
Starting ebusd with the option `--lograwdata=bytes` and `-l /var/tmp` (`./src/ebusd/ebusd -f --loglevel=info --lograwdata=bytes -l /var/tmp`) does not give me an output like this:

> 2017-11-08 20:56:02.097 [bus notice] <aa
2017-11-08 20:56:02.140 [bus notice] <aa
2017-11-08 20:56:02.183 [bus notice] <aa
2017-11-08 20:56:02.226 [bus notice] <aa
2017-11-08 20:56:02.269 [bus notice] <aa
2017-11-08 20:56:02.311 [bus notice] <aa
2017-11-08 20:56:02.354 [bus notice] <aa
2017-11-08 20:56:02.397 [bus notice] <aa
2017-11-08 20:56:02.440 [bus notice] <aa

It works fine without passing -l argument to ebusd.

john30, 2017-11-09:
well /var/tmp is usually a directory and not a file ebusd could possibly write to...

nuccleon, 2017-11-09:
You're right. But still no byte log with e.g. `-l ebusd.log`.
The funny thing is, that it starts with the rawdata log after receiving SIGINT.

[lograwdata.txt](https://github.com/john30/ebusd/files/1456607/lograwdata.txt)

john30, 2017-11-17:
what's your OS and gcc version?

nuccleon, 2017-11-17:
> pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.9.59-v7+ #1047 SMP Sun Oct 29 12:19:23 GMT 2017 armv7l GNU/Linux
pi@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@raspberrypi:~ $ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/6/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../src/configure -v --with-pkgversion='Raspbian 6.3.0-18+rpi1' --with-bugurl=file:///usr/share/doc/gcc-6/README.Bugs --enable-languages=c,ada,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-6 --program-prefix=arm-linux-gnueabihf- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-libitm --disable-libquadmath --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-6-armhf/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-6-armhf --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-6-armhf --with-arch-directory=arm --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-sjlj-exceptions --with-arch=armv6 --with-fpu=vfp --with-float=hard --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf
Thread model: posix
gcc version 6.3.0 20170516 (Raspbian 6.3.0-18+rpi1)
pi@raspberrypi:~ $

commit d51b08f0 (2017-11-19): fix for dumping bytes and raw data not initially enabled when requested (fixes #126), compiler warnings
closed 2017-11-19 by john30