john30/ebusd #127: MQTT: wrong values from Master-Slave-Messages

J0EK3R, 2017-11-11:
I'm using ebusd-build [ebusd 3.0.v3.0-12-g77d2a80] with MQTT-features compiled in.
[Additional Infos](https://github.com/J0EK3R/ebusd-configuration-weishaupt/wiki/ebusd-mqtt-fhem)

The only non-default MQTT-option for ebusd I'm using is
`--mqtttopic=ebusd/%circuit/%name/%field`
 **Remember the '%field'-part!**
_I don't use the --mqttjson_.

If I subscribe the topic **ebusd/hc2/Act/RoomTemp** the MQTT-broker sends me a value-mix of the fields RoomTemp and HeatDemandTemp:
```
pi@fhem:/etc/ebusd $ mosquitto_sub -t ebusd/hc2/Act/RoomTemp
24.500;37.0
```
If I subscribe the topic **ebusd/hc2/Act/ProgramState** the MQTT-broker sends me a value-mix of the fields ProgramState and SupplySetTemp:
```
pi@fhem:/etc/ebusd $ mosquitto_sub -t ebusd/hc2/Act/ProgramState
Program;32.0
```
If I subscribe the topic **ebusd/hc2/Act/HeatDemandTemp** the MQTT-broker sends me the value of the field ExternalValue.:
```
pi@fhem:/etc/ebusd $ mosquitto_sub -t ebusd/hc2/Act/HeatDemandTemp
10
```
These are the current valus read by ebusctl:
```
pi@fhem:/etc/ebusd $ ebusctl r -N -v -c hc2 Act
hc2 Act RoomTemp=24.500;ProgramState=2=Program;Byte3=128;HeatDemandTemp=37.0;SupplySetTemp=32.0;SupplyTemp=32.0;ExternalTemp=10;TrendTemp=9;Status=96=heating;ByteA=0;ByteB=0;ByteC=0
```
And there is a [mqtt error] in the log:
```
2017-11-11 13:46:29.999 [update notice] update hc2 Act QQ=70: 24.500;Program;128;37.0;32.0;30.5;10;9;heating;0;0;0
2017-11-11 13:46:34.472 [mqtt error] decode hc2 Act ByteA: empty
```

**I think MQTT-part of ebusd has a problem with master-slave-messages!
My suspect is that the master- and slave-parts of the message are mixed:  
first master-part (->RoomTemp) and the first slave-part (->HeatDemandTemp)  
second master-part (->ProgramState) and second slave-part (SupplySetTemp)**  

I have the following message-definition for this master-slave-message [link](https://github.com/J0EK3R/ebusd-configuration-weishaupt/blob/master/weishaupt/51..hc2.csv):
_The master-parts are marked with 'm', the slave-parts with 's'._
```
*b,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
b,,Act, ,,,"5010",,RoomTemp,m,_16_Temp0,,,Raumtemperatur ,ProgramState,m,_8_ProgramState,,,Status Heizprogramm ,Byte3,m,UCH,,,REQ ,HeatDemandTemp,s,_8_Temp2,,,Wärmeanforderung HK ,SupplySetTemp,s,_8_Temp2,,,Vorlaufsollwert HC ,SupplyTemp,s,_8_Temp2,,,Vorlauftemperatur HC ,ExternalTemp,s,_8_Temp1,,,Außentemperatur ,TrendTemp,s,_8_Temp1,,,Temperaturtrend ,Status,s,_8_Opdataheat2,,,Statusanforderung ,ByteA,s,UCH,,, ,ByteB,s,UCH,,, ,ByteC,s,UCH,,,
```

john30, 2017-11-17:
hm, I need to check that

J0EK3R, 2017-12-10:
Hello John!

I found a quick fix for this issue:

in file **mqtthandler.cpp** in method **MqttHandler::publishMessage(..)** in case "m_publishByField" there is a loop iterating all fields of a message for publishing.
[https://github.com/john30/ebusd/blob/d7400a4155997030b260d03b5794c9c052fe9cee/src/ebusd/mqtthandler.cpp#L629](url)

If you don't call the method **decodeLastData** by field-index but by field-name then the bug is fixed!

_Change_
```
  for (size_t index = 0; index < message->getFieldCount(); index++) {
    string name = message->getFieldName(index);
    result_t result = message->decodeLastData(false, NULL, index, outputFormat, updates);
```
_To_
```
  for (size_t index = 0; index < message->getFieldCount(); index++) {
    string name = message->getFieldName(index);
    result_t result = message->decodeLastData(false, name.c_str(), -1, outputFormat, updates);
```

john30, 2017-12-26:
fixed with commit c680e87

closed 2017-12-26 by john30