john30/ebusd #132: EbusD hangs while scanning

Erikwout, 2017-11-17:
Hi,
When i run ebusd in cygwin64 with the following command
ebusd.exe -d, --device=192.168.xxx.xxx:5000 -f -s -p 8888
it hangs(or a never ending loop as it takes 17% cpu constant) after a it tries to scan a certain device (76)
when i run ebusd without the -s option it runs fine and i am able to see undecoded messages with ebusctl (grab result), any idea how i can investigate why this scan hangs ?

when requesting info from ebusctl i get the following result.
version: ebusd 3.0.p20171019
signal: acquired
symbol rate: 97
max symbol rate: 100
reconnects: 0
masters: 4
messages: 11
conditional: 0
poll: 0
update: 4
address 03: master #11
address 08: slave #11
address 10: master #2
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 71: master #9
address 76: slave #9
address e8: slave

from the ebusd main screen:
2017-11-17 21:58:54.628 [bus notice] scan 08: ;21;17;39;0010019765;1610;005736;N6
2017-11-17 21:58:56.798 [bus notice] scan 15: ;Vaillant;70000;0419;4603
2017-11-17 21:58:57.615 [bus notice] scan 15: ;21;17;40;0020171315;0082;021274;N2
2017-11-17 21:58:57.628 [main notice] read scan config file /etc/ebusd/vaillant/15.700.csv for ID "70000", SW0419, HW4603
2017-11-17 21:58:57.628 [main notice] found messages: 395 (0 conditional on 0 conditions, 0 poll, 4 update)
2017-11-17 21:58:58.127 [update notice] unknown MS cmd: 1008b5110101 / 0949440080ffff0100ff
2017-11-17 21:58:58.202 [update notice] unknown MS cmd: 7108b507030bffff / 0101
2017-11-17 21:58:58.424 [update notice] unknown MS cmd: 1076b5110101 / 09ffff6900ffff00004b
2017-11-17 21:58:59.067 [update notice] unknown MS cmd: 1076b512030f0201 / 070080005d02ff03
2017-11-17 21:58:59.334 [update notice] unknown MS cmd: 1008b5100900004bffffff060000 / 0101
2017-11-17 21:58:59.851 [bus notice] scan 76: ;Vaillant;VWZIO;0111;0103
2017-11-17 21:59:00.735 [bus notice] scan 76: ;21;17;39;0020117049;0082;007597;N8

thanks for any hints or help

kind regards
Erik

john30, 2017-11-19:
that's really weird. I'm already on it and can reproduce the behaviour

john30, 2017-11-19:
please try again with latest commit

closed 2017-11-19 by john30

Erikwout, 2017-11-19:
Hi, 
Thanks for the quick action...
compiled the latest version, now it does not hang and it seems to decode some messages, but when i now ask for info in ebusctl i only see the ebusd device itself, and it thinks there is no signal, in the main screen i see a lot of traffic

localhost: i
version: ebusd 3.0.p20171119
update check: revision v3.0 available, broadcast.csv: different version available, memory.csv: different version available
signal: no signal
reconnects: 0
masters: 1
messages: 11
conditional: 0
poll: 0
update: 4
address 31: master #8, ebusd
address 36: slave #8, ebusd

commit b18d7f38 (2017-11-19): wait some time for initial garbage in tcp mode and fixed dumping garbage, use FIONREAD instead of MSG_PEEK for cygwin (fixes #132)