john30/ebusd #135: Installation on Raspberry Pi 3

Nijlpaard, 2017-12-03:
With my old RPI 2 the installation of the file ebusd***.armhf.deb was succesful.
Although the same procedure with an RPI 3 gave the error of the file is was not an debian format ?
Is it possible to fix it ?

john30, 2017-12-03:
please be more specific. which file did you use exactly and on which OS exactly (uname -a, cat /etc/issue)

Nijlpaard, 2017-12-03:
Command uname -a gave : Linux raspberrypi_1 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43 BST 2017 armv7l GNU           /Linux
The file i tried to install was : https://github.com/john30/ebusd/releases/ebusd-3.0_armhf.deb
(The hardware of all RPI 3 devices is 64 bit, but Raspbean Jessie is 32 bit)

ronan-fontenay, 2017-12-03:
Unable to reproduce on a freshly install Raspbian on a PI3 : 

pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.9.41-v7+ #1023 SMP Tue Aug 8 16:00:15 BST 2017 armv7l GNU/Linux
pi@raspberrypi:~ $ wget https://github.com/john30/ebusd/releases/download/v3.0/ebusd-3.0_armhf.deb
....
pi@raspberrypi:~ $ sudo dpkg -i ebusd-3.0_armhf.deb 
Selecting previously unselected package ebusd.
(Reading database ... 34363 files and directories currently installed.)
Preparing to unpack ebusd-3.0_armhf.deb ...
Unpacking ebusd (3.0) ...
Setting up ebusd (3.0) ...
Instructions:
1. Edit /etc/default/ebusd if necessary
   (especially if your device is not /dev/ttyUSB0)
2. Place CSV configuration files in /etc/ebusd/
   (see https://github.com/john30/ebusd-configuration)
3. To start the daemon, enter 'service ebusd start'
4. Check the log file /var/log/ebusd.log

john30, 2017-12-04:
okay, maybe you did not download the binary completely

closed 2017-12-04 by john30