john30/ebusd #138: unknown MS cmd:

derAuge, 2017-12-09:
meine Installation : 
pi@RPi2-2:~ $ ebusctl info
version: ebusd 3.0.595c7c0
update check: revision v3.0-23-g4c1aefb available, broadcast.csv: newer version available, vaillant/08.bai.csv: newer version available, vaillant/bai.308523.inc: newer version available, vaillant/broadcast.csv: different version available, vaillant/errors.inc: newer version available, vaillant/hcmode.inc: newer version available
access: *
signal: acquired
symbol rate: 23
max symbol rate: 169
reconnects: 4
masters: 3
messages: 212
conditional: 3
poll: 0
update: 8
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0204;HW=9602", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0419;HW=4603"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address ec: slave, scanned "MF=Vaillant;ID=70000;SW=0419;HW=4603"

bekomme ich folgende "[update notice] unknown"

2017-12-09 08:46:01.302 [update notice] unknown BC cmd: 10feb510020601
2017-12-09 08:46:59.514 [update notice] unknown MS cmd: 1008b5110100 / 087c031036040f0081
2017-12-09 08:48:00.283 [update notice] unknown MS cmd: 1008b5110100 / 087d031036040f0081
2017-12-09 08:49:00.105 [update notice] unknown MS cmd: 1008b5110100 / 0886031038040f0081
2017-12-09 08:50:00.406 [update notice] unknown MS cmd: 1008b5110100 / 0892031038040f0081
2017-12-09 08:51:00.701 [update notice] unknown MS cmd: 1008b5110100 / 0895031038040f0081
2017-12-09 08:51:01.161 [update notice] unknown MS cmd: 1008b5100305ff01 / 0101
2017-12-09 08:51:01.396 [update notice] unknown MS cmd: 1008b512020064 / 00
2017-12-09 08:51:01.638 [update notice] unknown MS cmd: 1008b5120204ff / 0101
2017-12-09 08:51:01.875 [update notice] unknown MS cmd: 1008b513020508 / 00
2017-12-09 08:51:02.087 [update notice] unknown BC cmd: 10feb510020601
2017-12-09 08:52:01.042 [update notice] unknown MS cmd: 1008b5110100 / 089c031038040f0081
2017-12-09 08:53:01.340 [update notice] unknown MS cmd: 1008b5110100 / 08a4031038040f0081
2017-12-09 08:54:01.634 [update notice] unknown MS cmd: 1008b5110100 / 08ab031038040f0081
2017-12-09 08:55:01.974 [update notice] unknown MS cmd: 1008b5110100 / 08b8031038040f0081
2017-12-09 08:56:02.274 [update notice] unknown MS cmd: 1008b5110100 / 08ba031038040f0081

wie können wir herausfinden was dahinter steckt?

john30, 2017-12-17:
recherchieren, beobachten, aufzeichnen, interpretieren und dann [eine CSV daraus machen](https://github.com/john30/ebusd/wiki/HowTos#creating-new-message-definition-files)

john30, 2018-02-27:
closed due to inactivity and because this is a ebusd-configuration issue actually

closed 2018-02-27 by john30