john30/ebusd #140: scan config 15: ERR: CRC error

bennnnnnnn, 2017-12-21:
Hi John,

First of all, thanks for this fantastic program! Really appreciate all the hard work you've put into it.

I have an issue with my f370 CalorMatic thermostat. I keep getting CRC errors. The strange thing is that it used to work. I used to be able to read and set values.
There does not seem to be an issue with the Vaillant heating unit(SW0604, HW5502).

I do have a special setup where ebusd is running in a docker container on a Xen vm. I'm using the eBus Koppler USB from eserviceonline. The entire USB controller is passed through to the VM via VT-D so as introduce as little latency/overhead as possible.
However I would think if the setup was the cause I would have issues communicating with the Vaillant heating unit as well. Also it did used to work.

I'm trying to understand why this issue is now suddenly appearing and if there is anything I can do to help it.

Logs are attached

[ebusd.log](https://github.com/john30/ebusd/files/1578841/ebusd.log)

Thanks!

Kind regards,
Ben

john30, 2017-12-22:
thanks :)
in order to really see, what's going on, you'd have to enable raw logging on byte level (i.e. start ebus with "--lograwdata=bytes").
However, it seems that some bytes are just not transferred correctly, which might either be related to the potentiometer not adjusted good enough (most probable) or something else.
We'll see that better with the bytes log

michapr, 2017-12-22:
CRC error for ebusd - this is in most cases a connection problem between ebus hardware and PC/Raspberry (if potentiometer calibration is ok).
I was using the LAN version of ebus interface - and tried via Powerline adapter - got many CRC errors (ping was between 5ms and 12ms). I had to place a Raspberry directly at same Ethernet segment.

bennnnnnnn, 2017-12-27:
Hi guys,

Thanks for your feedback!
I thought my eBus adapter didn't have a potentiometer, but I now noticed it in fact does. My bad! I'll try adjusting it and get back to you.

Best wishes!

bennnnnnnn, 2017-12-27:
Hi,

I did check the potentiometer, but as far as I could see from the instructions [here](https://github.com/john30/ebusd/wiki/6.-Hardware#adjusting-the-potentiometer) that was already ok.

I've ran ebusd with the lograwdata=bytes option:
[ebusd.log](https://github.com/john30/ebusd/files/1589722/ebusd.log)

Hopefully this will provide you some info.

I'm still wondering why there would only be a problem with the thermostat, there don't seem to be any CRC errors for messages coming from the heating unit itself. I do have to mention the thermostat is wireless, don't know if that could have something to do with it.

Thanks a lot!

john30, 2017-12-28:
The problem seems to be indeed related to the controller only. Sending an ID request to it fails with invalid CRC and even a message initiated by the controller itself yields an invalid CRC. This is really weird. From my point of view it looks like it is damaged. Either the controller itself or the base station connected to the bus.
How far is the cable distance from the controller base station to the ebus interface? Could you reduce that to the minimum possible? Maybe the wiring impedance imposes these problems.

bennnnnnnn, 2018-01-03:
Hi John,

First of all: best wishes for 2018!

I've done a change in my setup, I've now put a raspberry pi right next to the heating unit. Previously I did have quite a long cable running up to the controller (5-6m), so cable impedance could surely have been an issue.

It seems now however that my f370 thermostat is no longer getting detected at all :( although it used to be hit and miss in the previous setup as well.

I've attached a new logfile.
[ebusd.log](https://github.com/john30/ebusd/files/1599918/ebusd.log)

Is there a way we can deduce whether or not the issue lies with the controller?

Thanks!

Kind regards,
Ben

john30, 2018-01-06:
I guess this is due to the general potentiometer issue with hardware based on that concept. I'm pretty sure you won't have these issues with the new 2.0 adapter (see wiki).
anyway, you could have a look at the transferred bytes by using --lograwdata=bytes or even start with --lograwdata. if you see very short snippets like one or two bytes then something is really wrong. or if you have CRC errors in the log (without logging raw data), then also the poti is not adjusted well enough yet

bennnnnnnn, 2018-01-17:
Hi John,

Thanks for your feedback! I've tried changing the potentiometer, but I can't seem to get it to work :(

I've looked into getting the 2.0 adapter (using google translate and wishing I studied German in middle school :) ).

I'll order one when the next bulk order is coming up (do you know when that will be?)

Kind regards,
Ben

john30, 2018-01-19:
it might as well be a simple power issue, since each ebus device consumes some power from the line, each additional device will put some load on the line.
anyway, afaik new boards of the adapter are about to be ordered soon

john30, 2018-02-27:
meanwhile ordering is possible again for 2.1 adapter: https://forum.fhem.de/index.php/topic,84633.0.html

closed 2018-02-27 by john30