john30/ebusd #143: Reseting of datetime  and desired temp

pvydra, 2017-12-28:
Hi John, 

since I have connected eBus USB Koppler von Esera, current datetime and desired room temeprature are occasionally reset (lost) on my Vaillant 470f regulator.  Unfortunately, I don't know when exactly this happens and therefore I am not able to investigate log. 

Have you evere experienced this behaviour and/or could you point me where/what start to find?

I have Vaillant EcoTEC Exclusiv 146 + 470f

Many thanks, 

Petr V.

pi@hassbian:~ $ ebusctl info
version: ebusd 3.0.v3.0-26-g86a97d2
update check: version 3.1 available, broadcast.csv: different version available, memory.csv: different version available, vaillant/15.470.csv: different version available, vaillant/bai.308523.inc: different version available, vaillant/broadcast.csv: different version available, vaillant/errors.inc: different version available, vaillant/general.csv: different
signal: acquired
symbol rate: 22
max symbol rate: 145
min arbitration micros: 706
max arbitration micros: 9889
min symbol latency: 0
max symbol latency: 29
reconnects: 0
masters: 3
messages: 462
conditional: 3
poll: 1
update: 9
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0703;HW=7401", loaded "vaillant/bai.308523.inc" ([HW=7401]), "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=47000;SW=0130;HW=6002", loaded "vaillant/15.470.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd

root@hassbian:/home/pi# ebusctl grab result
10feb505022900 = 41
1008b512020000 / 00 = 983
1008b512020064 / 00 = 2286
0315b51303060000 / 0101 = 2

john30, 2018-01-01:
No, I didn't hear something like that before. You could use ebusd in "--readonly" mode if you remove all other CSV files, that are currently not needed. In readonly mode ebusd is not able to scan the devices and thus does not know, which CSV files to load.

Erbit, 2018-01-08:
Hi.

If, after switching off, the data is also lost, you need to look for the problem in Vaillant 470 (colormatic 470?). This is a known problem and results from damage to the capacitor holding the power supply. Vaillant does not fix it. You should buy a new 470 or repair it yourself as shown here https://www.elektroda.pl/rtvforum/topic3110151.html

Important! Check if it loses data after turning off the power. You can also disconnect the eBus USB Koper von Eser and check if the data is still lost.

pvydra, 2018-01-08:
Hi, thank you for the reply. 
Since I opened this issue, I haven't experienced the problem until today. 
The desired temp has disappeared again. (Time is ok)

I experienced the behavior since I have started 'experiments' with eBus Koppler, so it didn't come to my mind that it could be related to 470 and its power supply.

Erbit, 2018-01-09:
Hi.
I'm not sure but the symptoms are typical for this damage.

closed 2018-02-27 by john30

arno82, 2020-06-26:
I have the exact same issue and it was solved by disconnecting the ebus USB adapter!
I also do readonly on the bus, no writing.
Any solutions on this since 2018? It only resets my desired warm-water temperature as far as I see.

Dinth, 2020-06-26:
470f must have a really lousy RTC. Whenever the DCF77 signal is lost (for example by loosing connection to VR21), the internal clock in 470f gets terribly out of sync in a matter of days if not hours. 
I resolved it by using ebusd to manually sync datatime on 470f with an NTP server every midnight.

Kind regards

Tgroon, 2024-11-01:
> 470f must have a really lousy RTC. Whenever the DCF77 signal is lost (for example by loosing connection to VR21), the internal clock in 470f gets terribly out of sync in a matter of days if not [...]

how?

step21, 2024-12-14:
> how?

yeah would also be interested in this. My heater doesn't have a dcf, but it sometimes resets the time. Might also be because of ebus module, not sure. or because of powerloss, but I don't think there was any. Switching to only reading is not an option, as I think a main benefit (once properly integrated into home assistant) would be the writing

kjurcik, 2025-03-16:
Hi All, 

I observe the same issues and trying to isolate the loss of the time and date. If that is capacitor related to  capacitor 430 , I do nto know, but I have started to see it after conencing ebus USB adapter . Pls. are there any hints how to ebus write commant to automaticaly sync time and date ? Thank very much .