john30/ebusd #145: Conditions/Polling

rellla, 2018-01-01:
Hi,
does defining a condition in the csv automatically mean, that evaluating the condition is solved by polling with the defined interval?
https://github.com/rellla/ebusd-configuration/blob/WIP_bartl/ebusd-2.1.x/de/bartl_tem/bartl.csv#L9 are my conditions which need to execute https://github.com/rellla/ebusd-configuration/blob/WIP_bartl/ebusd-2.1.x/de/bartl_tem/bartl.csv#L6 in order to be evaluated.

Now, sometimes get a read timeout with that poll:
Here is the log:
https://gist.github.com/rellla/bbf3a53a5947a113014cb0949b553224

Having that condition is necessary, because this is the only way i currently can think of, how to differentiate between expert- and userlevel parameters. IDs depend on the currently set parameterlevel. Sadly a same ID can be a different command, so omitting the [*] condition prefixes lead to duplicaty ID errors...

So my questions:
- Is this (condition) the way i should do the differentiation between the levels, or can i do that more elegant?
- If so, how can i eliminate or minimize the amount of read timeouts?
- How is this polling internally handled? Caching etc...
- Is it possible to set the polling interval especially for conditions?

I'm referencing https://forum.fhem.de/index.php/topic,29737.msg720195.html#msg720195, which should be a result of the timeouts...

Thanks
rellla

john30, 2018-01-01:
yes, having a condition uses the polling mechanism to resolve the condition value. It is resolved in lowest priority though, so any other message with polling enabled (e.g. by using "r1" as the type in the CSV) will be queried with higher priority i.e. more often than those imposed by conditions.
Regarding your CSV: I don't think there is a better way of doing this than by conditions.
The read timeouts shouldn't impose any problem, as the conditions will be kept if the message runs into a timeout.
There is no means of setting the interval for conditions right now.

rellla, 2018-01-02:
Ok thanks. So then I'll stay with the conditions' concept.

What could be possible reasons for such timeouts? And is there a way to increase the timeout value? I tried setting "--receivetimeout=200000", but i'm not sure, if this is the related parameter because it seems to have no effect...

john30, 2018-01-04:
such a receive timeout is way beyond what the specification allows, thats why it is limited to 200ms in ebusd. furthermore, this is only the time in which an addressed slave is expected to answer. your timeout is another issue. it seems the addressed device is not willing to answer at all to the request. I can't tell you why though.

rellla, 2018-01-04:
thanks again. one last question though:

i assume it wouldn't be easily possible to implement some retrying mechanism like https://github.com/john30/ebusd/blob/master/src/ebusd/bushandler.cpp#L334 into the bushandler routines. i ask that, because from my users side of view, the polling is just a normal send to the addressed device.
for now i think i can ignore this issue.

at first i have find out, if my problems here: https://forum.fhem.de/index.php/topic,29737.msg719884.html#msg719884 and https://forum.fhem.de/index.php/topic,29737.msg720195.html#msg720195 have to do with the [condition] polling at all.

and in the meantime i will try to understand the code, especially BusHandler::handleSymbol() ;)

rellla

john30, 2018-01-06:
for the polled messages it is not desired to use a retry mechanism, as those are repeated anyway...
but this could be added as an extra commandline parameter I guess

closed 2018-12-25 by john30