john30/ebusd #146: grab did not display unknown messages

moccacup, 2018-01-01:
i tried to grab messages for a Vaillant heating pump system.

The grab command did not give any output.
In the raw log, i found many messages like this:

`2018-01-01 17:43:15.980 [bus notice] <3115b5e4f008ff00fa
2018-01-01 17:43:16.340 [bus notice] <10
2018-01-01 17:43:22.393 [bus notice] <10000950514006ffff0000fff9fa
2018-01-01 17:43:22.660 [bus notice] <10575be20009ff514006ffff0000ffac00fe
2018-01-01 17:43:22.926 [bus notice] <10579bfe000700800200801505fbfe
2018-01-01 17:43:23.136 [bus notice] <10b505fe
2018-01-01 17:43:23.438 [bus notice] <10fe
2018-01-01 17:43:23.732 [bus notice] <10e4
2018-01-01 17:43:23.997 [bus notice] <100001019a
2018-01-01 17:43:24.244 [bus notice] <100002c60281
2018-01-01 17:43:24.511 [bus notice] <10100001019a
2018-01-01 17:43:24.780 [bus notice] <10000a032543170101011840064d
2018-01-01 17:43:25.016 [bus notice] <10ff
2018-01-01 17:43:25.284 [bus notice] <10000987021600000000010075
2018-01-01 17:43:25.529 [bus notice] <100002ffffa8
`

There is no dedicated slave answer. the command "scan 15" will time out.

How may i use the decode tools?

john30, 2018-01-01:
what kind of adapter do you use? one with a potentiometer?
in this case, I guess the poti is not adjusted good enough yet your log contains pretty much nonsense

moccacup, 2018-01-02:
Thank You for the hint.
It is a "EService ebus usb" modul and it has a poti. After adjusting I can receive commands, but there are messages like above. Do you can mention if this is nosense due to the poti position or is there are another reason?

```
2018-01-02 19:41:18.790 [update notice] unknown BC cmd: 10feb505025c00
2018-01-02 19:41:18.794 [bus notice] <10feb505025c00e8
2018-01-02 19:41:19.094 [update notice] update vr_71 SensorData1 QQ=10: -;-;37.75;26.56;42.19;41.62;56.44;05 00
2018-01-02 19:41:19.097 [bus notice] <1026b5230106d00010008000805c02a90001a3029a02870305002900
2018-01-02 19:41:19.387 [update notice] update vr_71 SensorData2 QQ=10: 52.81;-;-;-;0.00;8.00;80 fc 05
2018-01-02 19:41:19.390 [bus notice] <1026b5230107d1000f4d030080008000800000800080fc05c100
2018-01-02 19:41:19.652 [update notice] unknown MS cmd: 1008b51009000000ffffff070000 / 0101
2018-01-02 19:41:19.656 [bus notice] <1008b51009000000ffffff070000520001019a00
2018-01-02 19:41:19.894 [update notice] unknown MS cmd: 1008b507020900 / 02c602
2018-01-02 19:41:19.898 [bus notice] <1008b5070209004f0002c6028100
2018-01-02 19:41:20.160 [update notice] unknown MS cmd: 1076b51009000000ffffff050000 / 0101
2018-01-02 19:41:20.165 [bus notice] <1076b51009000000ffffff050000dd0001019a00
2018-01-02 19:41:20.428 [update notice] unknown MS cmd: 1008b5040100 / 0a03224119020102186005
2018-01-02 19:41:20.432 [bus notice] <1008b50401003d000a032241190201021860051200
2018-01-02 19:41:20.665 [update notice] update broadcast vdatetime QQ=10: 19:41:18;02.01.2018
2018-01-02 19:41:20.668 [bus notice] <10feb5160800184119020102182c
2018-01-02 19:41:20.930 [update notice] unknown MS cmd: 1008b5110100 / 09870215000000000100
2018-01-02 19:41:20.934 [bus notice] <1008b5110100880009870215000000000100e900
2018-01-02 19:41:21.177 [update notice] unknown MS cmd: 1076b51303040d00 / 02ffff
2018-01-02 19:41:21.182 [bus notice] <1076b51303040d00bb0002ffffa800
2018-01-02 19:41:21.392 [update notice] update broadcast outsidetemp QQ=10: 5.375
2018-01-02 19:41:21.395 [bus notice] <10feb5160301600575
2018-01-02 19:41:21.681 [update notice] update vr_71 SetActorState QQ=10: off;-;-;off;off;off;off;off;-;-;-;-;00 00
2018-01-02 19:41:21.684 [bus notice] <1026b5230f0500ffff0000000000ffffffff0000880001019a00
2018-01-02 19:41:21.926 [update notice] update vr_71 Mc2FlowTempDesired QQ=10: on;39.0;on;5
2018-01-02 19:41:21.929 [bus notice] <1026b523040201014e4b00020105b200
2018-01-02 19:41:22.171 [update notice] update vr_71 Mc3FlowTempDesired QQ=10: off;0.0;on;-100
2018-01-02 19:41:22.175 [bus notice] <1026b5230402020000a40002019c2b00
2018-01-02 19:41:25.867 [bus notice] <00
2018-01-02 19:41:26.395 [bus notice] <10e00704006a
2018-01-02 19:41:26.921 [bus notice] <03e007040089
2018-01-02 19:41:27.619 [bus notice] <10e00704006a
2018-01-02 19:41:27.668 [bus notice] >11<03e0
2018-01-02 19:41:27.916 [bus notice] >1176b50901249d<000000>00
2018-01-02 19:41:28.477 [bus notice] >11<00fc
2018-01-02 19:41:29.004 [bus notice] <10e00704006a
2018-01-02 19:41:29.530 [bus notice] <03e007040089
2018-01-02 19:41:29.691 [bus notice] >1176b50901259c<000000>00
```
Unfortunally, in my ebus system, the address 31 was occupied. Now I used the 11. Is there a better choice or any rules for this choice? (see info below)

```
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0307;HW=0403", loaded "vaillant/08.hmu.csv"
address 0a: slave, scanned "MF=Vaillant;ID=PMW01;SW=0206;HW=8302", loaded "vaillant/0a.pmw.hwc.csv"
address 10: master #2
address 11: master #7, ebusd
address 15: slave #2
address 16: slave #7, ebusd
address 26: slave, scanned "MF=Vaillant;ID=VR_71;SW=0104;HW=0503", loaded "vaillant/26.vr_71.csv"
address 31: master #8
address 36: slave #8, scanned "MF=Vaillant;ID=KNXGW;SW=0102;HW=3603"
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZ00;SW=0307;HW=0403"
```

Further i didn't understand the error `[main error] scan config 76: ERR: invalid position`
(see context below). The error comes frequently:

```
2018-01-02 19:41:29.833 [update notice] unknown MS cmd: 1008b5110101 / 0950509005ffff0000ff
2018-01-02 19:41:29.836 [bus notice] >11<1008b511010189000950509005ffff0000ff3600
2018-01-02 19:41:30.364 [bus notice] <03e1070400d7
2018-01-02 19:41:31.024 [bus notice] >11<1074b511010116
2018-01-02 19:41:31.593 [bus notice] <03e1070400d7
2018-01-02 19:41:31.756 [bus notice] >1176b50901269f<000000>00
2018-01-02 19:41:31.858 [bus notice] <1076b5110101160009ff509005ffff0000ff6300
2018-01-02 19:41:31.931 [main error] scan config 76: ERR: invalid position
2018-01-02 19:41:31.932 [bus notice] >1176b50901279e<000000>00
2018-01-02 19:41:32.464 [bus notice] <03e1070400d7
2018-01-02 19:41:32.640 [update notice] unknown MS cmd: 1076b512030f0001 / 0700800200801505
2018-01-02 19:41:32.644 [bus notice] <1076b512030f0001fc000700800200801505fb00
2018-01-02 19:41:32.942 [update notice] update vr_71 SensorData1 QQ=10: -;-;37.69;26.56;42.12;41.50;56.44;05 00
2018-01-02 19:41:32.946 [bus notice] <1026b5230106d00010008000805b02a90001a202980287030500d400
2018-01-02 19:41:33.234 [update notice] update vr_71 SensorData2 QQ=10: 52

```

john30, 2018-01-04:
looks better now.
and you should rather use address ff instead of 11 for ebusd if 31 is already occupied by the KNX GW.
the invalid position seems to be related to the answer of 0x76, which is then probably too short and not specification conformant.
you can remove the --lograwdata option again if you don't need the hex log any more.

john30, 2018-01-06:
you should now get a grab result, too, if there are any unknown messages seen by ebusd

closed 2018-01-06 by john30

moccacup, 2018-01-06:
Thanks for the hints. I can see grabbed messages and will try to decode them. But i mention, it will take some time.

Olaf