john30/ebusd #147: Docker: cannot run daemon on Synology NAS

andig, 2018-01-04:
See below:

	admin@NAS:~$ sudo docker run --rm -it -p 8888 john30/ebusd -d 192.168.0.41:9999 -l /var/log/ebusd.log --scanconfig --latency=20000
	ebusd: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by ebusd)

If you need any info from inside the container or access please let me know.

andig, 2018-01-06:
I can confirm that adding `libstdc++6` to the runtime docker image fixes the problem for running on Synology NAS.

john30, 2018-01-06:
interesting, it was part of the base container for quite some time, but of course it doesn't hurt to add it.

john30, 2018-01-06:
is this really independant of using stretch-slim instead of jessie?

andig, 2018-01-06:
> is this really independant of using stretch-slim instead of jessie?

Yes. This issue was raised for the official ebusd docker image. I had the problem with jessie (plain, not slim) already and therefore split stretch-slim into separate container.

john30, 2018-01-07:
I see, good.

john30, 2018-01-07:
solved with latest commits (57f2b76)

closed 2018-01-07 by john30