john30/ebusd #148: Docker: build script fails (OSX)

andig, 2018-01-04:
Running `./build.sh` from contrib/docker I'm seeing multiple errors:

	~/htdocs/ebusd/contrib/docker master*
	❯ ./build.sh
	touch: illegal option -- d
	usage:
	touch [-A [-][[hh]mm]SS] [-acfhm] [-r file] [-t [[CC]YY]MMDDhhmm[.SS]] file ...
	creating debian image...
	*************
	 prepare
	*************

	./make_debian.sh: 1: ./make_debian.sh: git: not found

...

	Info:
	 new debian package, version 2.0.
	 size 224812 bytes: control archive=7980 bytes.
	   17776 bytes,   365 lines      changelog
	     405 bytes,    12 lines      control
	      62 bytes,     5 lines      dirs
	     423 bytes,     9 lines   *  postinst             #!/bin/sh
	 Package: ebusd
	 Version: 3.1
	 Section: net
	 Priority: required
	 Architecture: amd64
	 Maintainer: John Baier <ebusd@ebusd.eu>
	 Homepage: https://github.com/john30/ebusd
	 Bugs: https://github.com/john30/ebusd/issues
	 Depends: logrotate, libstdc++6 (>= 4.8.1), libc6, libgcc1, libmosquitto1
	 Description: eBUS daemon.
	  ebusd is a daemon for handling communication with eBUS devices connected to a
	  2-wire bus system.

	Content:
	drwxr-xr-x root/root         0 2018-01-04 21:14 ./
	drwxr-xr-x root/root         0 2018-01-04 21:14 ./usr/
	drwxr-xr-x root/root         0 2018-01-04 21:14 ./usr/bin/
	-rwxr-xr-x root/root     20152 2018-01-04 21:14 ./usr/bin/ebusctl
	-rwxr-xr-x root/root    561632 2018-01-04 21:14 ./usr/bin/ebusd
	drwxr-xr-x root/root         0 2018-01-04 21:14 ./etc/
	drwxr-xr-x root/root         0 2018-01-04 21:14 ./etc/init.d/
	-rwxr-xr-x root/root      2366 2018-01-04 21:14 ./etc/init.d/ebusd
	drwxr-xr-x root/root         0 2018-01-04 21:14 ./etc/default/
	-rw-r--r-- root/root      1356 2018-01-04 21:14 ./etc/default/ebusd
	drwxr-xr-x root/root         0 2018-01-04 21:14 ./etc/logrotate.d/
	-rw-r--r-- root/root       139 2018-01-04 21:14 ./etc/logrotate.d/ebusd
	drwxr-xr-x root/root         0 2018-01-04 21:14 ./etc/ebusd/
	-rw-r--r-- root/root      1457 2018-01-04 21:14 ./etc/ebusd/broadcast.csv

	mv: rename ../../ebusd-3.1__mqtt1.deb to runtime/ebusd-3.1__mqtt1.deb: No such file or directory

andig, 2018-01-05:
Got an idea where the dockerized build fails. Will try to fix and afterwards test if my assumption about #147 is true.

andig, 2018-01-05:
Problems:

  1. `touch` doesn't support `-d` on OSX. Not a big deal but leads to buildenv bulit every time.
  2. docker architecture empty in `build.sh` due to grep rule. This is the actual output on OSX:

          ❯ docker version
          Client:
           Version:      17.09.1-ce
           API version:  1.32
           Go version:   go1.8.3
           Git commit:   19e2cf6
           Built:        Thu Dec  7 22:22:25 2017
           OS/Arch:      darwin/amd64
          Server:
           Version:      17.09.1-ce
           API version:  1.32 (minimum version 1.12)
           Go version:   go1.8.3
           Git commit:   19e2cf6
           Built:        Thu Dec  7 22:28:28 2017
           OS/Arch:      linux/amd64
           Experimental: false

  3. https://github.com/john30/ebusd/blob/master/contrib/docker/runtime/Dockerfile uses hardcoded version 3.0 instead of version from `VERSION` file

andig, 2018-01-05:
2. this works on OSX in https://github.com/john30/ebusd/blob/master/contrib/docker/build.sh#L12:

          export EBUSD_ARCH=`docker version|grep -i "Arch:"|head -n 1|sed -e 's#^.*/##'`

john30, 2018-01-06:
as said, OSX is not supported (at least not by myself). feel free to change the build script as long as it will still run on debian

andig, 2018-01-06:
@john30 I can see you've noticed the PR. Closing here, the missing `touch -d` support doesn't hurt much.

closed 2018-01-06 by andig