john30/ebusd #151: Cannot read from VR70

andig, 2018-01-05:
Scan result:

	pi@server:~ $ ebusctl -s nas.fritz.box info
	version: ebusd 3.0.p20170829
	update check: version 3.1 available
	signal: acquired
	symbol rate: 23
	max symbol rate: 127
	reconnects: 0
	masters: 4
	messages: 566
	conditional: 3
	poll: 0
	update: 9
	address 00: master #1
	address 03: master #11
	address 05: slave #1, scanned "MF=Vaillant;ID=COM00;SW=0806;HW=3103"
	address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0609;HW=5502", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"
	address 10: master #2
	address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0209;HW=4103", loaded "vaillant/15.700.csv"
	address 31: master #8, ebusd
	address 36: slave #8, ebusd
	address 52: slave, scanned "MF=Vaillant;ID=VR_70;SW=0109;HW=2903", loaded "vaillant/52.vr_70.csv"

ebusd log:

	2018-01-05 15:50:56.644 [update notice] update bai Status01 QQ=10: 43.0;38.0;8.000;-;52.0;on
	2018-01-05 15:50:56.941 [update notice] update vr_70 SensorData QQ=10: 28.62;-;-;-;38.94;33.62;c8 7c 00
	2018-01-05 15:50:57.213 [update notice] update bai SetMode QQ=10: auto;43.0;-;-;0;0;0;0;0;0
	2018-01-05 15:50:57.736 [update notice] update vr_70 SetActorState QQ=10: on;-;off;off;-;-;off
	2018-01-05 15:50:57.985 [update notice] update vr_70 Mc2FlowTempDesired QQ=10: on;33.5;on;0

read:

	pi@server:~ $ ebusctl -s nas.fritz.box read -c bai Status01
	43.5;38.0;8.000;-;52.0;on

	pi@server:~ $ ebusctl -s nas.fritz.box read -c vr_70 Mc2FlowTempDesired
	ERR: element not found

Why does the one work but not the other?

andig, 2018-01-06:
I though this might have been fixed in a newer version but problem still persists with current git:

	pi@server:~ $ ebusctl -s 192.168.0.22 info
	version: ebusd 3.1.v3.0-38-g6bb4229
	update check: revision v3.1-1-g60a18d1 available
	signal: acquired
	symbol rate: 101
	max symbol rate: 151
	min arbitration micros: 35
	max arbitration micros: 1148
	min symbol latency: 1
	max symbol latency: 31
	reconnects: 0
	masters: 4
	messages: 566
	conditional: 3
	poll: 0
	update: 9
	address 00: master #1
	address 03: master #11
	address 05: slave #1, scanned "MF=Vaillant;ID=COM00;SW=0806;HW=3103"
	address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0609;HW=5502", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"
	address 10: master #2
	address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0209;HW=4103", loaded "vaillant/15.700.csv"
	address 31: master #8, ebusd
	address 36: slave #8, ebusd
	address 52: slave, scanned "MF=Vaillant;ID=VR_70;SW=0109;HW=2903", loaded "vaillant/52.vr_70.csv"

john30, 2018-01-06:
Mc2FlowTempDesired is a write message. you can't read that one.
and this is an ebusd-configuration issue, not an ebusd issue.

closed 2018-01-06 by john30