john30/ebusd #154: Feature request: Use ebusctl to test message definitions?

andig, 2018-01-18:
During https://github.com/john30/ebusd-configuration/issues/89 I was wondering if there is a quick way to test how message definitions are decoded. Running a 'productive' ebusd with TCP adapter the current process of updating config files, restarting daemon, diagnosing failure and restarting is cumbersome and impacts production operation.

Would it make sense to allow e.g. `ebusctl` to add a `-d|--definition` parameter to the `read` command that allows testing of various decodings?

john30, 2018-01-19:
for the time being, you can use `grab result decode` to get a list of potential values for each known base data type and each valid position of the master and slave data parts.
I had the option of letting ebusd decode grabbed data with a message definition transferred together with the request in mind for a long time, but it is not an easy task.
Instead, I started working on a graphical web based UI quite some time ago, that let's you interpret live data easily and edit existing or add new configuration parts.
These days I'm still busy with soldering adapters, but this is about to be finished by end of January, so I can continue with cool stuff :)
Anyway, it should not be necessary to restart ebusd when CSV files were changed. A simple `reload` command should do the trick. And before doing the reload, you can of course use ebusd to check the CSVs even if the daemon is still running.

andig, 2018-01-19:
> for the time being, you can use grab result decode to get a list of potential values for each known base data type and each valid position of the master and slave data parts.

I had this working once, but not again. My goal is to figure out decoding for `partloadhckw`. Apparently I'm doing this wrong:

	pi@server:~ $ ebusctl -s nas.fritz.box grab
	grab started

	pi@server:~ $ ebusctl -s nas.fritz.box read partloadhckw
	ERR: invalid position

	pi@server:~ $ ebusctl -s nas.fritz.box grab result decode
	done

john30, 2018-01-20:
what are you trying to achieve actually?

andig, 2018-01-20:
I would like to read `partloadhckw` on my bai. Since the config repository has multiple definitions for that message I thought that I'd just try all of them?

john30, 2018-01-23:
for such situations i currently use the hex command to see what the device is willing to answer and if the result fits to the message definition and produces valid values, i add it to the csv. but anyway, it would be good to be able to let ebusd all of that :)

john30, 2018-01-23:
note to myself: add "-t" parameter (for "test definition") followed by the message definition to be used to the read command (or better add a new command "test"?)

andig, 2018-03-11:
> for the time being, you can use grab result decode to get a list of potential values for each known base data type and each valid position of the master and slave data parts.

@john30 it would also be nice to be able to specify the `PBSB` on the command line. Use case would be to read e.g. `00xx-FFxx` to find a register that delivers a desired value. I'm thinking about using such an approach to find the `PartloadHwcKW` register for my bai.

john30, 2018-04-04:
for that you should rather use the hex command

commit 000d68f3 (2018-04-08): added "-def" option to read and write commands for using inline message definition (fixes #154), added support for replacing messages during load to FileReader and subclasses, added MessageMap::remove(), prepared decode/encode commands
closed 2018-04-08 by john30