john30/ebusd #155: ebusd cannot connect to ESP after restart

andig, 2018-01-19:
Today I have restarted the ebusd docker container in order to update the latest version. Since then it is not longer able to connect to the 2.0 adapter via wifi:

	admin@NAS:~$ sudo docker pull john30/ebusd
	latest: Pulling from john30/ebusd
	723254a2c089: Pull complete
	5a4dc9f0ec1e: Pull complete
	6bcbcd3144a0: Pull complete
	e33055e8bc56: Pull complete
	9f8f0f730abd: Pull complete
	655077b82016: Pull complete
	ce8d02e89042: Pull complete
	Digest: sha256:26d9f71282074bdbd8d7cfa9e49ac16acb25127257c606f748984926428c6c97
	Status: Downloaded newer image for john30/ebusd:latest

	admin@NAS:~$ sudo docker attach ebusd
	2018-01-19 15:10:33.766 [bus error] unable to open 192.168.0.146:9999: ERR: generic I/O error
	2018-01-19 15:10:49.294 [mqtt error] read message 700 Hc1ActualFlowTempDesired not found
	2018-01-19 15:10:59.778 [bus error] unable to open 192.168.0.146:9999: ERR: generic I/O error
	2018-01-19 15:11:19.301 [mqtt error] read message 700 Hc1ActualFlowTempDesired not found
	...

The esp device is reachable:

	pi@server:~ $ ping 192.168.0.46
	PING 192.168.0.46 (192.168.0.46) 56(84) bytes of data.
	64 bytes from 192.168.0.46: icmp_seq=13 ttl=128 time=7.02 ms
	64 bytes from 192.168.0.46: icmp_seq=14 ttl=128 time=20.4 ms

And alive:

	Build: 20171230
	Chip ID: 00125205
	CPU frequency: 80
	Free heap: 37816
	Hostname: ebus-125205
	ebusd device string: 192.168.0.46:9999
	ebusd connected: no
	eBUS signal: acquired

I'm a bit lost why ebusd can't reach it. I'm starting the container like this:

	sudo docker run -d --rm -p 8888:8888 --name ebusd john30/ebusd -f --scanconfig -d 192.168.0.146:9999 --latency 20000 --mqtthost=nas.fritz.box --mqttport=1883

john30, 2018-01-20:
the ip address is wrong. ebusd-esp says 192.168.0.46 but you configured ebusd to use 192.168.0.146

closed 2018-01-20 by john30

andig, 2018-01-20:
Wtf. Spent an hour and didn‘t see this. Wald und Bäume...

john30, 2018-01-20:
that happens sometimes :-)

andig, 2018-01-20:
Working now. However, after first restart ebusd identified the VRC700 but did not load the config. After another restart it did:

First start:

	pi@server:~ $ ebusctl -s nas.fritz.box info
	version: ebusd 3.1.b0e20b7
	update check: revision v3.1-1-g60a18d1 available
	signal: acquired
	symbol rate: 23
	max symbol rate: 168
	min arbitration micros: 18
	max arbitration micros: 123
	min symbol latency: 4
	max symbol latency: 47
	reconnects: 0
	masters: 4
	messages: 222
	conditional: 3
	poll: 0
	update: 9
	address 00: master #1
	address 03: master #11
	address 05: slave #1, scanned "MF=Vaillant;ID=COM00;SW=0902;HW=3103"
	address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0609;HW=5502", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"
	address 10: master #2
	address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0209;HW=4103"
	address 31: master #8, ebusd
	address 36: slave #8, ebusd
	address 52: slave, scanned "MF=Vaillant;ID=VR_70;SW=0109;HW=2903", loaded "vaillant/52.vr_70.csv"

	pi@server:~ $ ebusctl -s nas.fritz.box r Hc1ActualFlowTempDesired
	ERR: element not found

After restart:

	pi@server:~ $ ebusctl -s nas.fritz.box info
	version: ebusd 3.1.b0e20b7
	signal: acquired
	symbol rate: 24
	max symbol rate: 98
	min arbitration micros: 27
	max arbitration micros: 116
	min symbol latency: 5
	max symbol latency: 40
	reconnects: 0
	masters: 3
	messages: 556
	conditional: 3
	poll: 0
	update: 9
	address 03: master #11
	address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0609;HW=5502", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"
	address 10: master #2
	address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0209;HW=4103", loaded "vaillant/15.700.csv"
	address 31: master #8, ebusd
	address 36: slave #8, ebusd
	address 52: slave

	pi@server:~ $ ebusctl -s nas.fritz.box r Hc1ActualFlowTempDesired
	60

Not a big thing, but thought I'd let you know.

john30, 2018-01-23:
details about that should be in the log file. without that i have no idea what was going on...

marcin17w, 2024-10-29:
need help also 
ebus:
Chip ID: 7c2c67a9a27c, [ESP32-C3, rev 4](http://192.168.0.109/#/status?verbose)
Hostname: ebus1
Up time: 322
Free heap: 88208 / 257124
ebusd connected: yes (inactive)
eBUS signal: acquired
MQTT connected: yes

ebused in HA
[bus notice] device status: resetting
2024-10-29 16:36:21.493 [bus notice] re-opened 192.168.0.109:9999
2024-10-29 16:36:21.497 [bus notice] device status: reset, supports info
2024-10-29 16:36:21.501 [bus notice] device status: extra info: firmware 1.1[4a08].1[4a08], jumpers 0x0b
2024-10-29 16:36:21.523 [bus notice] signal acquired
2024-10-29 16:38:46.840 [update notice] received update-read broadcast id QQ=37: ENCON;  ;-;-
2024-10-29 16:39:24.042 [update notice] received update-read broadcast error QQ=37: E000      
2024-10-29 16:44:51.614 [bus notice] device invalid
2024-10-29 16:45:01.620 [bus error] unable to open 192.168.0.109:9999: ERR: generic I/O error
2024-10-29 16:45:01.620 [bus error] signal lost