john30/ebusd #156: Invalid 0 values using MQTT/Wemos adapter

andig, 2018-01-26:
Since upgrading to ebus adapter 2.0 and using Wemos via Wifi I've noticed that for certain messages I'm getting invalid 0 messages that don't exist in reality:

	{"topic":"Hc1FlowTemp","payload":"17.75"}
	{"topic":"Hc1FlowTemp","payload":"17.8"}
	{"topic":"Hc1FlowTemp","payload":"0.0"}
	{"topic":"Hc1FlowTemp","payload":"0.0"}
	{"topic":"Hc1FlowTemp","payload":"0.0"}
	{"topic":"Hc1FlowTemp","payload":"47.9375"}
	{"topic":"Hc1FlowTemp","payload":"47.9"}
	{"topic":"Hc1FlowTemp","payload":"46.4375"}
	{"topic":"Hc1FlowTemp","payload":"46.4"}
	{"topic":"Hc1FlowTemp","payload":"0.0"}
	{"topic":"Hc1FlowTemp","payload":"0.0"}
	{"topic":"Hc1FlowTemp","payload":"45"}
	{"topic":"Hc1FlowTemp","payload":"45.0"}

john30, 2018-01-26:
Please provide some ebusd log around the received 0.0 values

andig, 2018-01-28:
@john30 would really like to but I'm stuck with producing logfiles from the docker container. If I run without `-f` the container quits (since the entrypoint script exits) and with `-f` the log settings are ignored.

Even when forcing the entrypoint to ebusd the container is stopped right away:

	sudo docker run --rm -d \
	-p 8888:8888 \
	-v /volume1/data/docker/ebusd/ebusd.log:/var/log/ebusd.log \
	--entrypoint /usr/bin/ebusd \
	--name ebusd john30/ebusd \
	--logfile /var/log/ebusd.log --loglevel=debug \
	--scanconfig -d 192.168.0.46:9999 --latency 20000 \
	--mqtthost=nas.fritz.box --mqttport=1883 

Any idea how to get logs on docker?

john30, 2018-01-28:
sure, run it normally and then use `docker logs ebusd`

andig, 2018-01-28:
Ok, I have an invalid value at 13:54:31.

From the node-red log:

	{"topic":"ebusd/bai/Status01","payload":"58.5;36.5;9.938;-;48.5;off"}","timestamp":"2018-01-28T12:54:08.492Z"}
	{"topic":"ebusd/700/Hc1ActualFlowTempDesired","payload":"37.6707"}","timestamp":"2018-01-28T12:54:11.033Z"}

	{"topic":"Hc1FlowTemp","payload":"37.25"}","timestamp":"2018-01-28T12:54:11.037Z"}
	{"topic":"ebusd/700/Hc1FlowTemp","payload":"37.25"}","timestamp":"2018-01-28T12:54:11.037Z"}

	{"topic":"ebusd/700/Hc1MixerMovement","payload":"-15"}","timestamp":"2018-01-28T12:54:11.220Z"}
	{"topic":"ebusd/700/Hc1PumpStatus","payload":"1"}","timestamp":"2018-01-28T12:54:11.393Z"}
	{"topic":"ebusd/700/Hc2ActualFlowTempDesired","payload":"32.4035"}","timestamp":"2018-01-28T12:54:11.567Z"}
	{"topic":"ebusd/700/Hc2FlowTemp","payload":"34.5"}","timestamp":"2018-01-28T12:54:11.749Z"}
	{"topic":"ebusd/700/Hc2MixerMovement","payload":"-25"}","timestamp":"2018-01-28T12:54:11.935Z"}
	{"topic":"ebusd/700/Hc2PumpStatus","payload":"1"}","timestamp":"2018-01-28T12:54:12.109Z"}
	{"topic":"ebusd/700/HwcStorageTemp","payload":"48.5"}","timestamp":"2018-01-28T12:54:12.285Z"}
	{"topic":"ebusd/700/HwcTempDesired","payload":"55"}","timestamp":"2018-01-28T12:54:12.467Z"}
	{"topic":"ebusd/bai/CirPump","payload":"off"}","timestamp":"2018-01-28T12:54:12.620Z"}
	{"topic":"ebusd/bai/FlowTempDesired","payload":"52.00"}","timestamp":"2018-01-28T12:54:12.771Z"}
	{"topic":"ebusd/bai/HcHours","payload":"3825"}","timestamp":"2018-01-28T12:54:12.925Z"}
	{"topic":"ebusd/bai/HwcHours","payload":"237"}","timestamp":"2018-01-28T12:54:13.076Z"}
	{"topic":"ebusd/bai/ModulationTempDesired","payload":"46.5"}","timestamp":"2018-01-28T12:54:13.230Z"}

	{"topic":"Hc1FlowTemp","payload":"37.25"}","timestamp":"2018-01-28T12:54:14.229Z"}
	{"topic":"ebusd/700/Hc1FlowTemp","payload":"37.25"}","timestamp":"2018-01-28T12:54:14.229Z"}

	{"topic":"ebusd/700/Hc1MixerMovement","payload":"-15.00"}","timestamp":"2018-01-28T12:54:14.232Z"}
	{"topic":"ebusd/700/Hc2FlowTemp","payload":"34.50"}","timestamp":"2018-01-28T12:54:14.235Z"}
	{"topic":"ebusd/700/Hc2MixerMovement","payload":"-25.00"}","timestamp":"2018-01-28T12:54:14.237Z"}
	{"topic":"ebusd/bai/Status01","payload":"53.0;39.5;9.938;-;48.5;off"}","timestamp":"2018-01-28T12:54:19.286Z"}
	{"topic":"ebusd/bai/ModulationTempDesired","payload":"46.5"}","timestamp":"2018-01-28T12:54:19.290Z"}
	{"topic":"ebusd/bai/Status01","payload":"47.0;40.5;9.938;-;48.5;off"}","timestamp":"2018-01-28T12:54:29.334Z"}
	{"topic":"ebusd/bai/Status01","payload":"42.5;40.5;9.938;-;48.5;off"}","timestamp":"2018-01-28T12:54:39.343Z"}
	{"topic":"ebusd/700/Hc1ActualFlowTempDesired","payload":"37.6707"}","timestamp":"2018-01-28T12:54:41.058Z"}

	{"topic":"Hc1FlowTemp","payload":"3.875"}","timestamp":"2018-01-28T12:54:41.065Z"}
	{"topic":"ebusd/700/Hc1FlowTemp","payload":"3.875"}","timestamp":"2018-01-28T12:54:41.065Z"}

	{"topic":"ebusd/700/Hc1MixerMovement","payload":"100"}","timestamp":"2018-01-28T12:54:41.239Z"}
	{"topic":"ebusd/700/Hc1PumpStatus","payload":"1"}","timestamp":"2018-01-28T12:54:41.417Z"}
	{"topic":"ebusd/700/Hc2ActualFlowTempDesired","payload":"32.4035"}","timestamp":"2018-01-28T12:54:41.594Z"}
	{"topic":"ebusd/700/Hc2FlowTemp","payload":"31.375"}","timestamp":"2018-01-28T12:54:41.775Z"}
	{"topic":"ebusd/700/Hc2MixerMovement","payload":"5"}","timestamp":"2018-01-28T12:54:41.958Z"}
	{"topic":"ebusd/700/Hc2PumpStatus","payload":"1"}","timestamp":"2018-01-28T12:54:42.149Z"}
	{"topic":"ebusd/700/HwcStorageTemp","payload":"48.5"}","timestamp":"2018-01-28T12:54:42.353Z"}
	{"topic":"ebusd/700/HwcTempDesired","payload":"55"}","timestamp":"2018-01-28T12:54:42.553Z"}
	{"topic":"ebusd/bai/CirPump","payload":"off"}","timestamp":"2018-01-28T12:54:42.706Z"}
	{"topic":"ebusd/bai/FlowTempDesired","payload":"52.00"}","timestamp":"2018-01-28T12:54:42.862Z"}
	{"topic":"ebusd/bai/HcHours","payload":"3825"}","timestamp":"2018-01-28T12:54:43.017Z"}
	{"topic":"ebusd/bai/HwcHours","payload":"237"}","timestamp":"2018-01-28T12:54:43.203Z"}
	{"topic":"ebusd/bai/ModulationTempDesired","payload":"46.5"}","timestamp":"2018-01-28T12:54:43.355Z"}

	{"topic":"Hc1FlowTemp","payload":"3.9"}","timestamp":"2018-01-28T12:54:44.356Z"}
	{"topic":"ebusd/700/Hc1FlowTemp","payload":"3.9"}","timestamp":"2018-01-28T12:54:44.357Z"}

	{"topic":"ebusd/700/Hc1MixerMovement","payload":"100.0"}","timestamp":"2018-01-28T12:54:44.363Z"}
	{"topic":"ebusd/700/Hc2FlowTemp","payload":"31.4"}","timestamp":"2018-01-28T12:54:44.366Z"}
	{"topic":"ebusd/700/Hc2MixerMovement","payload":"5.0"}","timestamp":"2018-01-28T12:54:44.369Z"}
	{"topic":"ebusd/bai/Status01","payload":"41.5;38.5;9.938;-;48.5;off"}","timestamp":"2018-01-28T12:54:48.412Z"}
	{"topic":"ebusd/bai/Status01","payload":"41.5;36.5;9.938;-;48.5;off"}","timestamp":"2018-01-28T12:54:58.465Z"}
	{"topic":"ebusd/bai/Status01","payload":"41.5;35.0;9.938;-;48.5;off"}","timestamp":"2018-01-28T12:55:08.516Z"}
	{"topic":"ebusd/700/Hc1ActualFlowTempDesired","payload":"37.6707"}","timestamp":"2018-01-28T12:55:11.052Z"}

	{"topic":"Hc1FlowTemp","payload":"43.9375"}","timestamp":"2018-01-28T12:55:11.057Z"}

It looks as if there is a problem with the decimal digit. I fact, in this incident the value is 3.x, not 0.

From the ebusd log- I'll switch to debug level or dump binary?

[... 89 lines cut ...]
	2018-01-28 12:55:12.978 [mqtt notice] read bai HcHours: 1
	2018-01-28 12:55:13.130 [mqtt notice] read bai HwcHours: 1
[... cut ...]
	2018-01-28 12:55:35.073 [update notice] update vr_70 SensorData QQ=10: 28.75;-;-;-;41.25;28.81;c8 7c 00
	2018-01-28 12:55:35.344 [update notice] update bai SetMode QQ=10: auto;57.5;-;-;0;0;0;0;0;0
	2018-01-28 12:55:35.600 [update notice] update bai Status02 QQ=10: auto;60;52.0;70;70.0
	2018-01-28 12:55:35.906 [update notice] update vr_70 SetActorState QQ=10: -;-;-;-;-;-;off
	2018-01-28 12:55:36.155 [update notice] update vr_70 Mc1FlowTempDesired QQ=10: on;37.5;on;-25
	2018-01-28 12:55:36.406 [update notice] update vr_70 Mc2FlowTempDesired QQ=10: on;32.0;on;25

andig, 2018-01-28:
Seems this would need a better logging level- should I switch to debug level or add binary dump?

john30, 2018-01-28:
I don't see any invalid Hc1FlowTemp in the node-red log, so I guess there is some misconfiguration in node-red when interpreting the payload.
you meant 12:54:41 instead of 12:54:31, right?

andig, 2018-01-28:
> I don't see any invalid Hc1FlowTemp in the node-red log

It's there:

````
{"topic":"Hc1FlowTemp","payload":"3.875"}","timestamp":"2018-01-28T12:54:41.065Z"}
{"topic":"ebusd/700/Hc1FlowTemp","payload":"3.875"}","timestamp":"2018-01-28T12:54:41.065Z"}
````

The temp cannot have dropped below outside :O

> you meant 12:54:41 instead of 12:54:31, right?

Yes, sorry.

Can I provider better/different logs?

john30, 2018-01-29:
I still don't see the problem on ebus side here. According to node-red log, everything is fine IMHO, so the question is: why does node-red interpret the value so badly?

andig, 2018-01-29:
I have no clue where it gets wrong might even be the vrc700 (had some issues lately with night mode broken until unit was reset) or mqtt inbetween. I‘m just guessing that its ebus since this only started to happen after switching to the new adapter.

I cannot tell anything for the ebus logs since they don‘t show the actual values, only the mqtt read commands?

What logging setting on ebus side might be betterbsuited to rule out problems on the receiving side?

andig, 2018-02-02:
> I still don't see the problem on ebus side here. According to node-red log, everything is fine IMHO, so the question is: why does node-red interpret the value so badly?

@john30 the message is just received from MQTT and logged. I could dig out another tool but I'm convinced that it would read the same from MQTT. I'm suspecting that the error might be earlier in the chain.

I'll add `--lograwdata --loglevel` for time being to produce better ebusd log. Thanks for bearing with me.

andig, 2018-02-02:
Here we go- I hope this is a meaningful log:

	2018-02-02 17:53:30.537 [mqtt debug] received topic ebusd/700/Hc1FlowTemp/get
	2018-02-02 17:53:30.537 [mqtt info] received topic for 700 Hc1FlowTemp
	2018-02-02 17:53:30.537 [bus info] send message: 3115b52406020002000800
	2018-02-02 17:53:30.537 [bus debug] switching from send response ACK to send SYN
	2018-02-02 17:53:30.538 [bus notice] >3115b524060200020007007d<000801020700544c314229>00
	2018-02-02 17:53:30.543 [bus debug] send/receive symbol latency 5 ms
	2018-02-02 17:53:30.592 [bus debug] arbitration delay 36 micros
	2018-02-02 17:53:30.592 [bus debug] switching from ready to send command
	2018-02-02 17:53:30.598 [bus debug] send/receive symbol latency 5 ms
	2018-02-02 17:53:30.604 [bus debug] send/receive symbol latency 5 ms
	2018-02-02 17:53:30.609 [bus debug] send/receive symbol latency 5 ms
	2018-02-02 17:53:30.614 [bus debug] send/receive symbol latency 5 ms
	2018-02-02 17:53:30.619 [bus debug] send/receive symbol latency 5 ms
	2018-02-02 17:53:30.625 [bus debug] send/receive symbol latency 5 ms
	2018-02-02 17:53:30.631 [bus debug] send/receive symbol latency 5 ms
	2018-02-02 17:53:30.636 [bus debug] send/receive symbol latency 4 ms
	2018-02-02 17:53:30.641 [bus debug] send/receive symbol latency 5 ms
	2018-02-02 17:53:30.646 [bus debug] send/receive symbol latency 5 ms
	2018-02-02 17:53:30.647 [bus debug] switching from send command to send command CRC
	2018-02-02 17:53:30.653 [bus debug] send/receive symbol latency 6 ms
	2018-02-02 17:53:30.653 [bus debug] switching from send command CRC to receive command ACK
	2018-02-02 17:53:30.694 [bus debug] switching from receive command ACK to receive response
	2018-02-02 17:53:30.704 [bus debug] switching from receive response to receive response CRC
	2018-02-02 17:53:30.705 [bus debug] switching from receive response CRC to send response ACK
	2018-02-02 17:53:30.711 [bus debug] send/receive symbol latency 5 ms
	2018-02-02 17:53:30.711 [bus debug] notify request: done
	2018-02-02 17:53:30.711 [bus debug] read res: 0801020800ffffff7f
	2018-02-02 17:53:30.713 [mqtt notice] read 700 Hc1FlowTemp: 1
	2018-02-02 17:53:30.713 [mqtt debug] publish ebusd/700/Hc1FlowTemp nan

I'll be able to handle the `nan` on the node-red side. However I've never seen such messages with the ebus adapter 1.0 (directly attached).

Working as designed?

john30, 2018-02-03:
hm well, since the crc was correct I would strongly assume that the data was actually sent by the device. And ffffff7f in EXP format is just not a valid number, that's why it is resolved to "nan". Seems OK from ebusd side, the only question is why you get this now and never had before (I assume?). Maybe you could try to reproduce the same behaviour while running another ebusd instance simultaneously with the old adapter. But again, I expect the same value to be read. Maybe it is related to winter time with unusual values. I really can't tell.

andig, 2018-02-13:
I've just had the VRC700 and VR70 (where the sensors are attached) replaced which should exclude hardware error on Vaillant side. I still see the problem with the one sensor; will try to reproduce with alternative ebusd adapter.

andig, 2018-02-22:
I have meanwhile installed the original ebusd 1.0 adapter to the bus and started with

    --address 1f

after having figured out what format address needs to have and that it's actually hex (hint :). Both adapters are now listening for `700/Hc1FlowTemp` and publishing to different MQTT topics. Will update once I see the next `nan`.

Meanwhile- could you kindly check what happens when ebusd receives a `nan` from the device and how that is actually being published? Will it be an invalid byte pattern or a `nan`string or an empty message? I couldn't find out on node-red side as there's too much javascript magic inbetween.

andig, 2018-02-22:
I have meanwhile installed the 2nd adapter (original v1 hardware via FTDI, mqtt topic `ebus2`).

The flow via MQTT is this:

  - subscribe to `ebusd/700/Hc1FlowTemp` (wemos adapter)
  - subscribe to `ebus2/700/Hc1FlowTemp` (v1 adapter)
  - repeatedly trigger read by publishing `ebusd/700/Hc1FlowTemp`
  - then both the `ebusd` and the `ebus2` topics are updated with the result

What happens now is that I'm getting two curves for one topic, one from each `ebusd` adapter/instance:

[image]
I'm noticing two things here:

  - both topics show invalid values (e.g. the drops to 0.0)
  - both topics are not identical

The way it is right now the temperature curve is pretty much useless and I've lost any ideas where the mistake might be. It looks to me as if the device is actually sending wrong data (..but VRC700 and VR70 have already been replaced and the actual sensor is nothing but a stupid thermo resistor).

Totally confused....

john30, 2018-02-25:
since the MQTT sending of ebusd is asynchronouse with some delay, this might be the cause for the discrepancies between the two instances.
Currently NaN is treated as valid value by ebusd as the number is converted to a string for transfer. This could be changed to instead producing an error message and thus preventing the value from being sent via MQTT.

andig, 2018-03-15:
Further investigation shows that the invalid readings are generated by the VR70 `SensorData`. Root cause is to be determined.

Closing as not related to ebusd.

Thank you for the great support!

closed 2018-03-15 by andig