john30/ebusd #159: Inversion of Values (350/CirPump)

l33tn00b, 2018-02-02:
I'm currently integrating the heating system (BAI/350) with my home automation.
One of my goals is automatically turning off the heating while absent.
So, one of the ways of achieving that would be by turning off the Circualation Pump for the heating when leaving the house and turning it back on when returning.
ebusctl info gives:`

pi@heizpi:~ $ ebusctl info
`version: ebusd 3.1.p20180120
update check: revision v3.1-1-g60a18d1 available, broadcast.csv: newer version available, vaillant/08.bai.csv: newer version available, vaillant/15.350.csv: newer version available, vaillant/bai.308523.inc: newer version available, vaillant/broadcast.csv: different version available, vaillant/errors.inc: newer version available, vaillant/hcmode.inc: newer veaccess: *
signal: acquired
symbol rate: 23
max symbol rate: 115
min arbitration micros: 52
max arbitration micros: 141
min symbol latency: 6
max symbol latency: 10
reconnects: 0
masters: 3
messages: 409
conditional: 7
poll: 0
update: 8
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0608;HW=5502", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=35000;SW=0114;HW=7102", loaded "vaillant/15.350.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd`

Which fits with the installation.

The file selected by ebusd for control of the room thermostat is 
https://github.com/john30/ebusd-configuration/blob/master/ebusd-2.1.x/de/vaillant/15.350.csv
The line (#19) according to the above file:

`r;wi,,CirPump,,,,,"0B00",,,onoff,,,`

Now, entering the command
`ebusctl write -c 350 CirPump off`
gives
`done`.
But there is no reaction. Doing
'ebusctl read -c bai CirPump'
gives
`off`.
Trying the opposite:
`ebusctl write -c 350 CirPump on`
gives
`done`.
Et voilà, the VRT display gives 
`Absence exceptionelle actif`.
To set the room thermostat back to normal (present) mode:
`ebusctl write -c 350 CirPump off`.
Is the inversion on purpose?

john30, 2018-02-03:
certainly not. But I doubt that this is the right message for setting the absence mode. How did you identify the CirPump being responsible for the absence mode?
Futhermore, you are mixing bai and 350 CirPump messages. Try focusing on either of both.

l33tn00b, 2018-02-03:
I took a look at the VRT. And to my surprise it said I had actually activated the absence mode. 
Yes, I mentioned both. I was toying around with both. 
Since there is virtually no documentation except for your excellent work I have no other way than just try..  And that's what I discovered... 

Am 3. Februar 2018 14:11:19 MEZ schrieb John <notifications@github.com>:
>certainly not. But I doubt that this is the right message for setting
>the absence mode. How did you identify the CirPump being responsible
>for the absence mode?
>Futhermore, you are mixing bai and 350 CirPump messages. Try focusing
>on either of both.

john30, 2018-02-03:
hm, I think the absence mode is controlled differently, but of course I don't own a 350, so I really don't know...

l33tn00b, 2018-02-03:
Thought so, too. Of course, I could post a picture. Yes, you're right.  Misplaced. Can you move it?

john30, 2018-02-03:
issues can't be moved

l33tn00b, 2018-02-03:
Well, close it.

closed 2018-02-03 by l33tn00b