john30/ebusd #161: ebusd-3.1_armhf-jessie_mqtt1.deb does not generate a logfile

ChrisHanuta, 2018-02-06:
Hello,

I have a problem with the logfile in the "ebusd-3.1_armhf-jessie_mqtt1.deb" on my PI
If I use the "ebusd-3.1_armhf-jessie.deb", a logfile is created and described, as soon as you use the "ebusd-3.1_armhf-jessie_mqtt1.deb", no logfile will be created.
Parameters at ebusd-3.1_armhf-jessie_mqtt1.deb:
--enablehex --scanconfig --configpath = / opt / ebus.conf / ebusd2 --loglevel = info --mqtthost = 192.168.2.10 --mqttport = 1883 --mqttjson --mqttuser = * --mqttpass = * - mqtttopic = ebusd /% circuit /% name /% field
Parameters at ebusd-3.1_armhf-jessie.deb:
--enablehex --scanconfig --configpath = / opt / ebus.conf / ebusd2 --loglevel = info

Also adding the parameter --logfile = / opt / ebus.log / ebusd.log teaches "ebusd-3.1_armhf-jessie_mqtt1.deb" unsuccessful. The "ebusd-3.1_armhf-jessie.deb" writes the log file properly.

ChrisHanuta, 2018-02-10:
Only a single ebusd process is running at the same time. it works well with the ebusd-3.1_armhf-jessie.deb.
It did not work with the ebusd-3.1_armhf-jessie_mqtt1.deb, I did not know why.
I use Docker with Supervisor, so the setup is reproducible.

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND
    7 root      20   0   14448   7996   4500 S   1.8  0.8   8:28.12 supervisord
    1 root      20   0    1820      8      8 S   0.0  0.0   0:00.04 sh
   10 root      20   0    5100   1760   1564 S   0.0  0.2   0:00.33 cron
   56 root      20   0   52564   3032   2536 S   0.0  0.3  11:01.44 ebusd
 1657 root      20   0    5200   2912   2680 S   0.0  0.3   0:00.03 bash
 1662 root      20   0    5112   2236   1964 R   0.0  0.2   0:00.03 top

[program:ebusd]97:/opt/ebusd# cat /etc/supervisor/conf.d/supervisord-ebusd.conf
command=ebusd -f --enablehex --scanconfig --configpath=/opt/ebus.conf/ebusd2 --loglevel=info --logfile=/opt/ebus.log/ebusd.log --mqtthost=192.168.2.10 --mqttport=1883 --mqttjson --mqttuser=MQTT --mqttpass=MQTT --mqtttopic=ebusd/%%circuit/%%name/%%field
numprocs=1
autostart=true
autorestart=true

root@11febd65d197:/opt/ebusd# cat /opt/ebus.log/ebusd.log

john30, 2018-02-11:
why don't you use the ebusd docker image?

closed 2018-02-27 by john30