john30/ebusd #163: Verbose output for TCP listen mode

msteiger, 2018-02-07:
I appreciate the work you put into this thing. It really helps me a lot.

So, I'm writing a client similar to [john30/vwmon](https://github.com/john30/vwmon). It listens on raw TCP:8888 for events and writes broadcast data to a DB. 

A critical point here is bringing field names and values together and I think that this could be improved. Afaics, I have to do the following today:

* Call `find -v` and remove the values to get the field names
* Call `listen`, parse the list of values and map them, based on their index, to the field names from above

It would be convenient, if ..
* there was an option for `find` that just gives the field names.
* even better: a verbose version of `listen` that does what `read -v` does and map field names to values

Thanks

john30, 2018-02-10:
I'm not sure I fully understand.
Have a look at the documentation of e.g. the find and the read commands here:
https://github.com/john30/ebusd/wiki/3.1.-TCP-client-commands#read
https://github.com/john30/ebusd/wiki/3.1.-TCP-client-commands#find
especially the find command reveals a lot of the fields a message is composed of and you can use "-F" to limit to certain parts of the definition

msteiger, 2018-02-13:
I was hoping that I just missed the right combination of parameters. Here's what I get:
```
find -v betrd       -->       broadcast betrd = status=7;zustand=79;stellgrad=25...
listen              -->       broadcast betrd = 10;41;0;49.0;128;49;2
```
This is what I hope for (at least one of them):
```
find --no-values(?) betrd     --> broadcast betrd = status;zustand;stellgrad
listen --verbose(?)           --> broadcast betrd = status=7;zustand=79;stellgrad=25...
```
I'm using `listen` instead of `read` so I get notified immediately when the broadcast comes in.
Does that make sense?

john30, 2018-02-17:
for find you can use the "-f" option that yields in the csv format
for the listen command extension, i'll put it on my todo list

commit af77f4bc (2019-10-27): add adjustable verbosity to listen command (fixes #163)
closed 2019-10-27 by john30