john30/ebusd #165: ebusd doesn't find Wemos after wemos restart

andig, 2018-02-12:
Today I had my heating repaired and powered down. Ebusd in turn lost connection:

	2018-02-12 12:32:02.390 [bus notice] <1052b523040201014ac800020164d300
	2018-02-12 12:32:03.133 [mqtt debug] publish ebusd/global/uptime 848208
	2018-02-12 12:32:03.134 [mqtt debug] publish ebusd/bai/Status01 46.0;46.0;3.250;-;52.0;off
	2018-02-12 12:32:03.135 [mqtt debug] publish ebusd/vr_70/SensorData 27.56;-;-;-;-12.44;23.94;c8 7c 00
	2018-02-12 12:32:07.750 [main debug] performing regular tasks
	2018-02-12 12:32:08.321 [bus debug] ERR: read timeout during receive command, switching to skip
	2018-02-12 12:32:10.022 [bus debug] ERR: read timeout during skip, switching to no signal
	2018-02-12 12:32:10.022 [bus error] signal lost
	2018-02-12 12:32:13.162 [mqtt debug] received topic ebusd/700/Hc1ActualFlowTempDesired/get
	2018-02-12 12:32:13.163 [mqtt info] received topic for 700 Hc1ActualFlowTempDesired
	2018-02-12 12:32:13.163 [bus info] send message: 3115b52406020002000700
	2018-02-12 12:32:13.266 [bus error] send to 15: ERR: no signal, give up
	2018-02-12 12:32:13.266 [bus error] send message part 0: ERR: no signal
	2018-02-12 12:32:13.266 [mqtt error] read 700 Hc1ActualFlowTempDesired: ERR: no signal
	2018-02-12 12:32:13.266 [mqtt debug] received topic ebusd/700/Hc1FlowTemp/get
	2018-02-12 12:32:13.266 [mqtt info] received topic for 700 Hc1FlowTemp
	2018-02-12 12:32:13.267 [bus info] send message: 3115b52406020002000800
	2018-02-12 12:32:13.536 [bus error] send to 15: ERR: no signal, give up
	2018-02-12 12:32:13.536 [bus error] send message part 0: ERR: no signal
	2018-02-12 12:32:13.536 [mqtt error] read 700 Hc1FlowTemp: ERR: no signal
	2018-02-12 12:32:13.536 [mqtt debug] received topic ebusd/700/Hc1MixerMovement/get
	2018-02-12 12:32:13.536 [mqtt info] received topic for 700 Hc1MixerMovement
	2018-02-12 12:32:13.537 [bus info] send message: 3115b52406020002001a00
	2018-02-12 12:32:13.807 [bus error] send to 15: ERR: no signal, give up
	2018-02-12 12:32:13.807 [bus error] send message part 0: ERR: no signal
	2018-02-12 12:32:13.807 [mqtt error] read 700 Hc1MixerMovement: ERR: no signal
	2018-02-12 12:32:13.807 [mqtt debug] received topic ebusd/700/Hc1PumpStatus/get
	2018-02-12 12:32:13.807 [mqtt info] received topic for 700 Hc1PumpStatus
	2018-02-12 12:32:13.808 [bus info] send message: 3115b52406020002001e00
	2018-02-12 12:32:14.077 [bus error] send to 15: ERR: no signal, give up
	2018-02-12 12:32:14.077 [bus error] send message part 0: ERR: no signal
	2018-02-12 12:32:14.077 [mqtt error] read 700 Hc1PumpStatus: ERR: no signal
	2018-02-12 12:32:14.077 [mqtt debug] received topic ebusd/700/Hc2ActualFlowTempDesired/get
	2018-02-12 12:32:14.077 [mqtt info] received topic for 700 Hc2ActualFlowTempDesired
	...
	
After restarting the bai/wemos Ebusd continued failing to connect to the wemos. At the same time the wemos was online, the ebusd docker container remained unchanged:

	eBUS adapter 2.0: Welcome

	Build: 20171230
	Chip ID: 00125205
	CPU frequency: 80
	Free heap: 37648
	Hostname: ebus-125205
	ebusd device string: 192.168.0.46:9999
	ebusd connected: no
	eBUS signal: acquired
	Configuration:

	WIFI SSID: 
	WIFI secret: 
	WIFI IP address: , mask length: , gateway: 
	ebusd TCP/UDP mode: 
	ebusd TCP port: 
	RX+TX PINs: 
	Management TCP port: (0 to disable)
	LED PINs: RX: , TX: 

After rebooting the docker container (NOT the wemos), connection is fine again.

Something prevented ebusd from re-attaching to the wemos.

john30, 2018-02-13:
this might be related to the hard cut of the connection, so maybe ebusd does not recognize the other end being disconnected already. can you try to reproduce by simply pressing the reset switch on the ebusd-esp wemos?

andig, 2018-02-17:
Unable to repro. Took power away a couple of times and ebusd always reconnected. Seems I was unfortunate last time...

john30, 2018-02-17:
hm, weird

andig, 2019-09-29:
I have just been able to repro again. Restarting Wemos does not fix the problem, restarting ebusd does. Not sure this is a real issue in the wild, but it is reproducible.

john30, 2019-10-27:
did you wait more than 60 seconds? this is the time ebusd waits before trying a reconnect

andig, 2019-11-04:
> did you wait more than 60 seconds? this is the time ebusd waits before trying a reconnect

@john30 I can wait any amount of time- WeMOS is not able to re-attach to EBUSD. It is currently reproducible every single time: 

- restarting WeMOS breaks EBUSD connection to the bus
- restarting EBUSD re-establishes connection

This is an unfortunate situation as e.g. Wifi outages will lead to dropped EBUSD connection.

If there is any detail I can supply please let me know what you need.

john30, 2019-11-05:
please provide all config details for a start, i.e. Wemos config screenshot or text copy as well as complete ebusd command line together with all complete version numbers

andig, 2019-11-09:
> please provide all config details for a start

Here we go. To recap: everything is working fine until the wemos restarts. At this point only an ebusd restart works:

Wemos config:

    WIFI SSID: ANDIHOME
    WIFI secret: ****
    WIFI IP address: DHCP, mask length: 24, gateway: 255.255.255.0
    WIFI hostname: ("." for default): ebus2
    eBUS RX+TX PINs: direct
    ebusd connection: TCP on port: 9999
    HTTP TCP port: 80
    LED PINs: RX: disabled, TX: disabled 

Wemos status (initial):

    Chip ID: 00cfb492
    Hostname: ebus2
    Up time: 88
    Free heap: 35072
    ebusd connected: yes (inactive)
    eBUS signal: acquired 
    eBUS initial signal: yes

ebusd version:

    admin@NAS:~$ sudo docker exec ebusd ebusd --version
    ebusd 3.4.v3.3-51-g57eae05

ebusd docker config:

    ebusd:
        container_name: ebusd
        image: john30/ebusd
        ports:
        - 8888:8888
        command: --enabledefine --enablehex --accesslevel='*' --scanconfig -f -d ebus2.fritz.box:9999 --latency 20000 --mqtthost=nas.fritz.box --mqttport=1883
        restart: always

logfile when wemos is being reset:

    2019-11-09 10:48:30.615 [update notice] received read vr_70 SensorData QQ=10: -;-;-;-;32.50;29.00;c8 7d 00
    2019-11-09 10:48:30.881 [update notice] received update-write bai SetMode QQ=10: auto;33.5;-;-;0;0;0;0;0;0
    2019-11-09 10:48:31.141 [update notice] received write vr_70 SetActorState QQ=10: -;-;-;-;-;-;off
    2019-11-09 10:48:31.390 [update notice] received write vr_70 Mc1FlowTempDesired QQ=10: on;33.5;on;5
    2019-11-09 10:48:31.634 [update notice] received write vr_70 Mc2FlowTempDesired QQ=10: on;29.5;on;0
    2019-11-09 10:48:33.015 [bus error] signal lost
    2019-11-09 10:48:44.096 [bus error] send to 15: ERR: no signal, give up
    2019-11-09 10:48:44.096 [bus error] send message part 0: ERR: no signal
    2019-11-09 10:48:44.366 [bus error] send to 15: ERR: no signal, give up
    2019-11-09 10:48:44.366 [bus error] send message part 0: ERR: no signal
    2019-11-09 10:48:44.366 [mqtt error] read 700 Hc1ActualFlowTempDesired: ERR: no signal
    2019-11-09 10:48:44.637 [bus error] send to 15: ERR: no signal, give up
    2019-11-09 10:48:44.637 [bus error] send message part 0: ERR: no signal
    2019-11-09 10:48:44.907 [bus error] send to 15: ERR: no signal, give up
    2019-11-09 10:48:44.907 [bus error] send message part 0: ERR: no signal
    2019-11-09 10:48:45.177 [bus error] send to 15: ERR: no signal, give up
    2019-11-09 10:48:45.177 [bus error] send message part 0: ERR: no signal
    2019-11-09 10:48:45.177 [mqtt error] read 700 Hc1PumpStatus: ERR: no signal

at this point the wemos is unable to connect (or vice versa):

    Chip ID: 00cfb492
    Hostname: ebus2
    Up time: 21
    Free heap: 35992
    ebusd connected: no
    eBUS signal: acquired 
    eBUS initial signal: yes

john30, 2019-11-10:
I've added the keepalive option to the latest ebusd-esp binary, but it didn't change the behaviour during my tests. as of now I'm out of ideas

andig, 2019-11-11:
Is there any logging I could enable? Afaiks ebusd will (re)connect to the wemos once the connection drops. I can probably compile my own ebusd and add some logging around the connection attempts?

john30, 2019-11-16:
I did that already, but the mechanism for detecting dead end doesn't seem to work as expected (i.e. FIONREAD call will return an error)

john30, 2019-12-29:
this shuld be fixed with 177568c here and the latest https://github.com/john30/ebusd-esp/commit/418e3d1 binary

andig, 2019-12-29:
Thank you for your efforts. Closing as fixed until I can find the time to properly test. Would repoen if it reappears.

closed 2019-12-29 by andig