john30/ebusd #166: I can't start daemon

szczukot, 2018-02-15:
Hi
I installed ebusd:
git clone https://github.com/john30/ebusd.git
cd ebusd
./autogen.sh
make 
sudo make install

pi@raspberrypi:/usr/bin $ sudo ebusd -f -d /dev/ttyUSB2 --scanconfig
2018-02-15 18:13:27.033 [main notice] ebusd 3.1.v3.1-22-gdaf771e started with auto scan
2018-02-15 18:13:27.054 [bus notice] bus started with own address 31/36
2018-02-15 18:13:27.104 [bus notice] signal acquired
2018-02-15 18:13:37.962 [bus notice] new master 03, master count 2
2018-02-15 18:13:46.982 [bus notice] scan 08: ;Vaillant;BAI00;0518;7401
2018-02-15 18:13:46.983 [update notice] store 08 ident: done
2018-02-15 18:13:46.984 [update notice] sent scan.08  QQ=31: Vaillant;BAI00;0518;7401
2018-02-15 18:13:46.985 [bus notice] scan 08: ;Vaillant;BAI00;0518;7401
2018-02-15 18:13:47.014 [main notice] read common config file /etc/ebusd/vaillant/general.csv
2018-02-15 18:13:47.016 [main notice] read common config file /etc/ebusd/vaillant/scan.csv
2018-02-15 18:13:47.021 [main notice] read common config file /etc/ebusd/vaillant/broadcast.csv
2018-02-15 18:13:47.029 [main notice] read scan config file /etc/ebusd/vaillant/08.bai.csv for ID "bai00", SW0518, HW7401
2018-02-15 18:13:47.114 [bus error] send to 08: ERR: read timeout, retry
2018-02-15 18:13:47.713 [update notice] sent scan.08 id QQ=31:
2018-02-15 18:13:47.949 [update notice] sent scan.08 id QQ=31:
2018-02-15 18:13:48.120 [update notice] sent scan.08 id QQ=31:
2018-02-15 18:13:48.295 [update notice] sent scan.08 id QQ=31: 21;10;40;0010003869;3100;005327;N7
2018-02-15 18:13:48.383 [main notice] found messages: 200 (0 conditional on 23 conditions, 0 poll, 9 update)
2018-02-15 18:13:48.516 [update notice] sent scan.08 id QQ=31: 21;10;40;0010003869;3100;005327;N7
2018-02-15 18:13:49.426 [update notice] sent scan.08 id QQ=31: 21;10;40;0010003869;3100;005327;N7
2018-02-15 18:15:47.814 [main notice] update check: revision v3.1-1-g60a18d1 available, broadcast.csv: different version available, memory.csv: different version available, vaillant/bai.0010003857.inc: different version available, vaillant/broadcast.csv: different version available, vaillant/errors.inc: different version available, vaillant/general.csv: different version available, vaillant/hcmode.inc: different version available

sudo systemctl enable ebusd
i get "Failed to execute operation: No such file or directory"
I can't run ebusd as daemon

and many times, when i start ebusd in foreground i get only:
2018-02-15 18:18:41.820 [main notice] ebusd 3.1.v3.1-22-gdaf771e started with auto scan
2018-02-15 18:18:41.845 [bus notice] bus started with own address 31/36
2018-02-15 18:18:41.862 [bus notice] signal acquired
and nothing else

regards
Fantom

john30, 2018-02-17:
please provide the full output of "sudo systemctl enable ebusd" as well "ls -la lib/systemd/system/"

szczukot, 2018-02-17:
pi@raspberrypi:~ $ sudo systemctl enable ebusd
Failed to execute operation: No such file or directory

pi@raspberrypi:~ $ sudo ls -la /lib/systemd/system
total 836
drwxr-xr-x 24 root root 36864 Feb  9 16:54 .
drwxr-xr-x  8 root root  4096 Jul 25  2017 ..
-rw-r--r--  1 root root   476 Jul 17  2014 alsa-restore.service
-rw-r--r--  1 root root   526 Jul 17  2014 alsa-state.service
-rw-r--r--  1 root root   373 Jul 17  2014 alsa-store.service
lrwxrwxrwx  1 root root     9 Jul 17  2014 alsa-utils.service -> /dev/null
lrwxrwxrwx  1 root root    14 May  7  2017 autovt@.service -> getty@.service
-rw-r--r--  1 root root  1044 Apr 18  2015 avahi-daemon.service
-rw-r--r--  1 root root   874 Apr 18  2015 avahi-daemon.socket
-rw-r--r--  1 root root   524 May  7  2017 basic.target
drwxr-xr-x  2 root root  4096 May 27  2016 basic.target.wants
-rw-r--r--  1 root root   338 May 27  2016 bluetooth.service
-rw-r--r--  1 root root   379 May  7  2017 bluetooth.target
lrwxrwxrwx  1 root root     9 May  7  2017 bootlogd.service -> /dev/null
lrwxrwxrwx  1 root root     9 May  7  2017 bootlogs.service -> /dev/null
lrwxrwxrwx  1 root root     9 May  7  2017 bootmisc.service -> /dev/null
lrwxrwxrwx  1 root root     9 May  7  2017 checkfs.service -> /dev/null
lrwxrwxrwx  1 root root     9 May  7  2017 checkroot-bootclean.service -> /dev/null
lrwxrwxrwx  1 root root     9 May  7  2017 checkroot.service -> /dev/null
-rw-r--r--  1 root root   770 May  7  2017 console-getty.service
-rw-r--r--  1 root root   741 May  7  2017 console-shell.service
-rw-r--r--  1 root root   783 May  7  2017 container-getty@.service
-rw-r--r--  1 root root   251 Sep  5  2015 cron.service
lrwxrwxrwx  1 root root     9 May  7  2017 cryptdisks-early.service -> /dev/null
lrwxrwxrwx  1 root root     9 May  7  2017 cryptdisks.service -> /dev/null
-rw-r--r--  1 root root   394 May  7  2017 cryptsetup-pre.target
-rw-r--r--  1 root root   366 May  7  2017 cryptsetup.target
lrwxrwxrwx  1 root root    13 May  7  2017 ctrl-alt-del.target -> reboot.target
lrwxrwxrwx  1 root root    25 May  7  2017 dbus-org.freedesktop.hostname1.service -> systemd-hostnamed.service
lrwxrwxrwx  1 root root    23 May  7  2017 dbus-org.freedesktop.locale1.service -> systemd-localed.service
lrwxrwxrwx  1 root root    22 May  7  2017 dbus-org.freedesktop.login1.service -> systemd-logind.service
lrwxrwxrwx  1 root root    24 May  7  2017 dbus-org.freedesktop.machine1.service -> systemd-machined.service
lrwxrwxrwx  1 root root    25 May  7  2017 dbus-org.freedesktop.timedate1.service -> systemd-timedated.service
-rw-r--r--  1 root root   366 Jan 15  2017 dbus.service
-rw-r--r--  1 root root   106 Jan 15  2017 dbus.socket
drwxr-xr-x  2 root root  4096 Apr 22  2017 dbus.target.wants
-rw-r--r--  1 root root   402 May  7  2017 debian-fixup.service
-rw-r--r--  1 root root  1010 May  7  2017 debug-shell.service
lrwxrwxrwx  1 root root    16 May  7  2017 default.target -> graphical.target
-rw-r--r--  1 root root   636 May  7  2017 dev-hugepages.mount
-rw-r--r--  1 root root   590 May  7  2017 dev-mqueue.mount
-rw-r--r--  1 root root   241 Oct 17  2015 dhcpcd.service
-rw-r--r--  1 root root   986 May  7  2017 emergency.service
-rw-r--r--  1 root root   431 May  7  2017 emergency.target
-rw-r--r--  1 root root   353 Mar 16  2015 fake-hwclock.service
-rw-r--r--  1 root root   440 May  7  2017 final.target
lrwxrwxrwx  1 root root     9 May  7  2017 fuse.service -> /dev/null
-rw-r--r--  1 root root  1536 May  7  2017 getty@.service
-rw-r--r--  1 root root   342 May  7  2017 getty-static.service
-rw-r--r--  1 root root   460 May  7  2017 getty.target
drwxr-xr-x  2 root root  4096 Jul 25  2017 getty.target.wants
-rw-r--r--  1 root root   490 May  7  2017 graphical.target
drwxr-xr-x  2 root root  4096 Jul 25  2017 graphical.target.wants
-rw-r--r--  1 root root   565 May  7  2017 halt-local.service
lrwxrwxrwx  1 root root     9 May  7  2017 halt.service -> /dev/null
-rw-r--r--  1 root root   487 May  7  2017 halt.target
drwxr-xr-x  2 root root  4096 May 27  2016 halt.target.wants
-rw-r--r--  1 root root   271 Feb 16  2017 hciuart.service
-rw-r--r--  1 root root   447 May  7  2017 hibernate.target
lrwxrwxrwx  1 root root     9 May  7  2017 hostname.service -> /dev/null
lrwxrwxrwx  1 root root     9 May  7  2017 hwclockfirst.service -> /dev/null
-rw-r--r--  1 root root   398 May  7  2017 hwclock-save.service
lrwxrwxrwx  1 root root     9 May  7  2017 hwclock.service -> /dev/null
-rw-r--r--  1 root root   468 May  7  2017 hybrid-sleep.target
-rw-r--r--  1 root root   380 May  7  2017 ifup@.service
[... 167 lines cut ...]
-rw-r--r--  1 root root   575 May  7  2017 systemd-udevd-kernel.socket
-rw-r--r--  1 root root   826 May  7  2017 systemd-udevd.service
-rw-r--r--  1 root root   823 May  7  2017 systemd-udev-settle.service
-rw-r--r--  1 root root   715 May  7  2017 systemd-udev-trigger.service
-rw-r--r--  1 root root   757 May  7  2017 systemd-update-utmp-runlevel.service
-rw-r--r--  1 root root   821 May  7  2017 systemd-update-utmp.service
-rw-r--r--  1 root root   588 May  7  2017 systemd-user-sessions.service
-rw-r--r--  1 root root   433 May  7  2017 system.slice
-rw-r--r--  1 root root   652 May  7  2017 system-update.target
-rw-r--r--  1 root root   355 May  7  2017 timers.target
drwxr-xr-x  2 root root  4096 Jul 25  2017 timers.target.wants
-rw-r--r--  1 root root   395 May  7  2017 time-sync.target
-rw-r--r--  1 root root   661 May  7  2017 tmp.mount
[... cut ...]
lrwxrwxrwx  1 root root     9 May  7  2017 umountroot.service -> /dev/null
-rw-r--r--  1 root root   417 May  7  2017 umount.target
lrwxrwxrwx  1 root root    27 May  7  2017 urandom.service -> systemd-random-seed.service
-rw-r--r--  1 root root   150 Aug 18  2014 usb_modeswitch@.service
-rw-r--r--  1 root root   497 May  7  2017 user@.service
-rw-r--r--  1 root root   392 May  7  2017 user.slice
-rw-r--r--  1 root root   244 Oct 16 12:41 wpa_supplicant.service
lrwxrwxrwx  1 root root     9 May  7  2017 x11-common.service -> /dev/null

john30, 2018-02-17:
oh I see, since you didn't use a release package but built it yourself, you'll have to copy the https://github.com/john30/ebusd/blob/master/contrib/debian/default/ebusd and https://github.com/john30/ebusd/blob/master/contrib/debian/systemd/ebusd.service to the right folders (/etc/default for default/ebusd, /lib/systemd/system for ebusd.service).

john30, 2018-02-17:
what do you mean by that? did you copy the files already?

szczukot, 2018-02-17:
ls /etc/default
....
ebusd 
....

ls /lib/systemd/system
...
ebusd.service
....

john30, 2018-02-17:
does ebusd appear in `systemctl list-units`?

szczukot, 2018-02-17:
After reeboot, i see:

> systemctl list-units
> ebusd.service         loaded active     running         ebusd, the daemon for communication with eBUS

still
pi@raspberrypi:~ $ sudo chkconfig ebusd on
ebusd: unknown service

and

pi@raspberrypi:~ $ tail /var/log/ebusd.log -f
2018-02-17 15:12:42.123 [main notice] ebusd 3.1.v3.1-22-gdaf771e started with auto scan
and nothing else

closed 2018-02-17 by szczukot