john30/ebusd #167: Add support for crc8 checksum fields in messages

msteiger, 2018-02-18:
I admit that this an odd request, but I think that it's required to support write support for Wolf proprietary messages. This is how it works:

The command `xx yy 5023 09 d8 4201 04 00 5d010000` sets parameter `HG 07` to 4 minutes.

The value `5023` is the main Wolf write command.
Then `09` is the number of bytes  to come (as usual).
The `d8` is a **CRC-8 checksum of the 8 bytes that follow**.
The `4201` indicates the parameter `HG 07` as described by the user manual.
The value is in the two bytes `04 00` in LL HH format.
The rest of the message seems to be a constant suffix.

In order to specify those messages in CSV I would need to be able to define the checksum field somehow. I have a short C snippet that computes the CRC8 (tested successfully) I can share. Does that make sense?

baycom, 2018-02-20:
I am not sure if this byte is a crc over the whole message. It even occurs with exactly the same value when reading from the slave.
Maybe you are interested in my overview: https://docs.google.com/spreadsheets/d/1S0difDnnMiqyjwa7TPG0M8KY0L4JfTgNukiMdSh7rx0/edit?usp=sharing

baycom, 2018-02-20:
What kind of CRC algorithm is being used for that? In my case it does not make any difference what value the first byte of the message has. The components still answer / ack the request if the first byte is 0x00. Here are my findings - I think we should consolidate the whole Wolf stuff.

msteiger, 2018-02-20:
It's a CRC8 based on the value `0x5C`. It's what I discovered what the ART and the BM devices use. I just tried out what you say and I can confirm it: the checksum value is really ignored by my heating unit (CGU-2K).

I still think that we should implement it correctly instead of sending a constant dummy value.

I agree that we should consolidate the data for Wolf! All of the HG parameters I know are proposed here: https://github.com/john30/ebusd-configuration/pull/96

baycom, 2018-02-21:
Today I took a little time and extracted the remaining parameters for all Wolf heat pump types (air, water, split, etc.). Now there is nothing left undocumented for heat pumps. Further I can confirm that the crc8 polynom 0x5c works for all messages.

msteiger, 2018-02-24:
Nice! Can you create a PR for the [ebus-configuration](https://github.com/john30/ebusd-configuration) repo so I (and others) can use it, too?

john30, 2018-02-25:
if the wolf devices ignore tha checksum, then why should we add a special data type for that one?
just make it IGN:1 and that's it.

msteiger, 2018-02-26:
Yes, that's a very valid point. We could just ignore it and carry on, because it will work for the CGU-2K at least.

That said, just because it works, doesn't mean that it's correct ... and it isn't. We're just exploiting the fact that some heating units don't verify the checksum. The controlling devices ART and BM do send the checksum to the heating units, so chances are that some units will require it to be correct.

The good thing is that the hard part has already been done: the meaning of the value has been identified, the algorithm is implemented and even tested on two real devices. Integrating it into ebus is the last piece of the puzzle. Why not complete it?

john30, 2018-02-27:
sure, but someone has to develop the data type and the tests for it. I don't have time for that right now, but I'll keep this issue as a reminder/TODO

kopierschnitte, 2023-09-23:
Sorry for bumping an old issue but it seems that my setup indeed requires a correct crc8 checksum for at least one dataset.
I'm talking about "HG15" (water hysteresis) on a Wolf CGW-20.

Every other command I've tried works with 0x00 or any other crc but this one doesn't. It works if I manually calculate it and use the hex-command.

kopierschnitte, 2023-11-14:
Unfortunately, I couldn't calculate it for myself yet. I was just reply previously sniffed CRCc

msteiger, 2023-11-20:
There are several CRC7 or CRC8 available which slightly differ. Maybe you could provide some "valid" checksums, so we can identify the correct implementation? Should not be too hard.

FanDjango, 2023-12-31:
Here are the python routines I use for my Wolf ebus code concerning crc fields, for incoming and outgoing:

```py
def crc(b, crc):
    for i in range(8):
        if crc & 0x80:
            p = 0x9B
        else:
            p = 0x00
        crc = (crc & 0x7F) << 1
        if b & 0x80:
            crc = crc | 0x01
        crc = crc ^ p
        b = (b << 1) & 0xFF
    return crc

def crc_check(data, check_crc):
    calc_crc = 0x00
    for i in range(len(data)):
        calc_crc = crc(data[i], calc_crc)
    return calc_crc == check_crc

def crc_make(data):
    calc_crc = 0x00
    for i in range(len(data)):
        calc_crc = crc(data[i], calc_crc)
    return calc_crc
```

lildadou, 2024-03-30:
The CRC is calculated on PSPB and the ID on 2 bytes. Both are "static" data. The CRC can therefore be calculated in advance, without precluding the possibility of changing QQ, ZZ and DD. This is what has been done in [ebus-configuration](https://github.com/john30/ebusd-configuration/blob/master/ebusd-2.x.x/de/wolf/08.csv)

In this example, the ID is announced over 3 bytes and the first byte actually contains the CRC.

To calculate my CRCs, I used [a project that implemented the calculation on a complete telegram](https://github.com/csowada/ebus-configuration/blob/master/src/test/java/de/csdev/ebus/cfg/KW_CRC_Test.java) (it'll reassure you that changing QQ, ZZ or DD doesn't change the CRC).

PLCHome, 2025-12-16:
> The CRC is calculated on PSPB and the ID on 2 bytes. Both are "static" data. The CRC can therefore be calculated in advance, without precluding the possibility of changing QQ, ZZ and DD. This is [...]

An exception is made when writing the value, since the value is part of the checksum, or in other words, the checksum changes with each value. With d84201 and  5d010000 you can write 4 not 5 or 3.

Something else caught my eye while reading. I don't know if it's the same check amount, but...

At that point, there's sometimes a number and sometimes "0xCC" = 204, depending on who's querying the value. It would also be practical if EBusD could take this into account. I've now created everything twice and am managing it via the topic.

```
root@ebusd:/var/log/supervisor# ebusctl g result all|grep -i -E "..08502203..2D00"
3108502203142d00 / 029600 = 540
3008502203cc2d00 / 029600 = 9588
7008502203cc2d00 / 029600 = 2
```

For example, the parameter HG 25 is queried either with cc or 14 before the ID; sometimes one value arrives faster than the other.
where the value is often automatically transmitted as CC.

I know the PTID 180131 is HG25 from my ISM7. Otherwise, it's very difficult to read it from this file: https://github.com/zivillian/ism7mqtt/blob/master/src/ism7mqtt/Resources/DeviceTemplates.xml

If you have an ISM7, it's easy because the project creates a file containing the addresses and PTIDs. However, I don't find it desirable to only receive an update every few minutes. The eBus runs in extended time.

I only ever had a few values ​​until I found this:
https://raw.githubusercontent.com/zivillian/ism7mqtt/refs/heads/master/src/ism7mqtt/Resources/ParameterTemplates.xml

```
    <NumericParameterDescriptor PTID="180131">
      <Name>Kesselübertemperatur bei Speicherladung</Name>
      <UnitName>°C</UnitName>
      <RemoveConditionId>False</RemoveConditionId>
      <ReadOnlyConditionId>False</ReadOnlyConditionId>
      <InactiveConditionId>False</InactiveConditionId>
      <Comment>HG25</Comment>
      <IsSnapshotTransmitEnabled>true</IsSnapshotTransmitEnabled>
      <MinValueCondition>1</MinValueCondition>
      <MaxValueCondition>30</MaxValueCondition>
      <StepWidth>0.5</StepWidth>
      <Decimals>1</Decimals>
      <ControlType>NumericInput</ControlType>
    </NumericParameterDescriptor>
```

The ID can be found here:
https://github.com/zivillian/ism7mqtt/blob/master/src/ism7mqtt/Resources/ConverterTemplates.xml
```
    <NumericConverter16Template CTID="180131">
      <TelegramNr>45</TelegramNr>
      <Type>SS10</Type>
    </NumericConverter16Template>
```

`<TelegramNr>45</TelegramNr>` is 0x002D in HEX. Now you have to turn low and high bytes. ID=2D00

You will find it with the grep: `ebusctl g result all|grep -i -E "..08502203..2D00"`
08 is the bus address for the HG area. Some IDs exist on multiple devices.

The translation will be available here:
```
    <TextTableEntry>
      <DEU>Kesselübertemperatur bei Speicherladung</DEU>
      <CHN>缓冲水箱加热过程中的锅炉温度过高</CHN>
      <GRC>Υπερθέρμανση λέβητα στη φόρτιση μπόιλερ</GRC>
      <EST>Boileri liigne temperatuur akumulatsioonipaagi laadimisel</EST>
      <HRV>Nadtemperatura kotla pri punjenju spremnika</HRV>
      <LVA>Apkures katla papildus temperatūra amortizatora cilindra uzlādes laikā</LVA>
      <LTU>Katilo perviršinė temperatūra buferinio cilindro įkrovimo metu</LTU>
      <ROU>Temperatură excesivă boiler în timpul încărcării cilindrului tamponului</ROU>
      <ITA>Sovratemperatura caldaia con carico accumulatore</ITA>
      <ESP>Diferencia temperatura de caldera con carga acumulador</ESP>
      <FRA>Surtempérature de chaudière pour chargement ballon</FRA>
[... 92 lines cut ...]
    return crc

def crc_make(data):
    calc_crc = 0x5C
    for i in range(len(data)):
        calc_crc = crc(data[i], calc_crc)
    return calc_crc

if __name__ == "__main__":
#xxyy502309 d8  420104005d010000
    data = crc_make([0x42,0x01,0x04,0x00,0x5d,0x01,0x00,0x00])
    print("")
    print(f"{'%#04x' % data}")
    data = crc_make([0x00,0x00,0x01,0x5d,0x00,0x04,0x01,0x42])
    print("")
    print(f"{'%#04x' % data}")
```

Return:
```
0x9c
0xc9
```

Noting equals d8

aghulas, 2026-03-13:
The same CRC is also used with weishaupt read commands  (0x5022,, 0x5000) and write commands (0x5023, 0x5001)
Some devices aren't checking the CRC but the WTC burners definitively are. 

If you are interested, I partially solved this requirement by making an addon to the ebus-typespec to allow the CRC to be computed and inserted at compile to the extension id : 

[weishaupt.ts](https://github.com/user-attachments/files/25962613/weishaupt.ts)

and then in a weishaupt.tsp file : 

import "./weishaupt.js";

extern dec crcext(target: TypeSpec.Reflection.Model, ...dd: valueof uint8[]);

then I use @crcext instead of @ext and this nicely compute and insert the CRC value when I compile my files. 

It works for the read messages, but not for the write ones as the CRC value needs to be computed on the extid + outgoing values.