john30/ebusd #168: Cannot run docker on raspi3

andig, 2018-02-19:
This happens:

	pi@ebus2:~ $ sudo docker run john30/ebusd
	Unable to find image 'john30/ebusd:latest' locally
	latest: Pulling from john30/ebusd
	723254a2c089: Pull complete
	fb30d9e9be5b: Pull complete
	840892f6e7c7: Pull complete
	aa49144da917: Pull complete
	15d48e829feb: Pull complete
	d3d382abe717: Pull complete
	abfed5a15f06: Pull complete
	Digest: sha256:24337cacac622d3cfca3e125b9c2bba34e2f285f173a5b743bf7d11136646aae
	Status: Downloaded newer image for john30/ebusd:latest
	standard_init_linux.go:195: exec user process caused "exec format error"

Did I somehow catch the wrong architecture?

andig, 2018-02-20:
Thinking about it an ARM build might not be too beneficial, at least not for Raspbian since ebus already provides installation packages. Closing for time being.

closed 2018-02-20 by andig

semenoh, 2021-01-07:
this is exactly my case. I'm trying to setup ebusd in HomeAssistant using docker but I can not because of amd64 architecture. Is it possible to make arm version beside amd ?

john30, 2021-02-20:
arm is available now as well in the docker images