john30/ebusd #169: 23.solsy.cc error with arm package

andig, 2018-02-20:
I've just installed the 3.1 armhf package on a fresh raspi and copied the 2.1 DE config file:

    cp -r ebusd-configuration/ebusd-2.1.x/de/* /etc/ebusd

Starting ebusd gives following error:

	pi@ebus2:~ $ ebusd -f --loglevel=debug	2018-02-20 09:31:36.728 [main notice] ebusd 3.0pre.bbc4d04 started
	2018-02-20 09:31:36.728 [main info] loading configuration files from /etc/ebusd
	2018-02-20 09:31:36.729 [main info] reading templates /etc/ebusd
	2018-02-20 09:31:36.733 [main info] read templates in /etc/ebusd
	2018-02-20 09:31:36.733 [main info] reading file /etc/ebusd/broadcast.csv
	2018-02-20 09:31:36.737 [main info] successfully read file /etc/ebusd/broadcast.csv
	2018-02-20 09:31:36.737 [main info] reading file /etc/ebusd/memory.csv
	2018-02-20 09:31:36.739 [main info] successfully read file /etc/ebusd/memory.csv
	2018-02-20 09:31:36.739 [main info] reading dir  /etc/ebusd/vaillant
	2018-02-20 09:31:36.742 [main info] reading templates /etc/ebusd/vaillant
	2018-02-20 09:31:36.754 [main info] read templates in /etc/ebusd/vaillant
	2018-02-20 09:31:36.755 [main info] reading file /etc/ebusd/vaillant/23.vr630.cc.csv
	2018-02-20 09:31:36.758 [main info] successfully read file /etc/ebusd/vaillant/23.vr630.cc.csv
	2018-02-20 09:31:36.758 [main info] reading file /etc/ebusd/vaillant/64.v65.csv
	2018-02-20 09:31:36.759 [main info] successfully read file /etc/ebusd/vaillant/64.v65.csv
	2018-02-20 09:31:36.759 [main info] reading file /etc/ebusd/vaillant/broadcast.csv
	2018-02-20 09:31:36.761 [main info] successfully read file /etc/ebusd/vaillant/broadcast.csv
	2018-02-20 09:31:36.761 [main info] reading file /etc/ebusd/vaillant/26.solsy.hc.csv
	2018-02-20 09:31:36.776 [main info] successfully read file /etc/ebusd/vaillant/26.solsy.hc.csv
	2018-02-20 09:31:36.776 [main info] reading file /etc/ebusd/vaillant/15.e7f.csv
	2018-02-20 09:31:36.791 [main info] successfully read file /etc/ebusd/vaillant/15.e7f.csv

	2018-02-20 09:31:36.791 [main info] reading file /etc/ebusd/vaillant/23.solsy.cc.csv
	2018-02-20 09:31:36.792 [main error] error reading config files: ERR: duplicate entry, last error: /etc/ebusd/vaillant/23.solsy.cc.csv:6: ERR: duplicate entry, duplicate ID
	2018-02-20 09:31:36.792 [bus error] unable to open /dev/ttyUSB0: ERR: element not found

	2018-02-20 09:31:36.792 [bus notice] bus started with own address 31/36
	2018-02-20 09:31:36.793 [main info] registering data handlers
	2018-02-20 09:31:36.793 [main info] registered data handlers

I've always been confused with using the 2.x or 2.1 config files but the latter are the last updated ones. File in question updated a year ago.

john30, 2018-02-21:
you also need to copy all _templates.csv from both the base directory and the sub directory.

closed 2018-02-21 by john30

andig, 2018-02-22:
@john30 I realise I've done this a number of times and I may still get it wrong. However the files are in place and identical:

	pi@ebus2:~ $ ls -l /etc/ebusd/
	total 16
	-rw-r--r-- 1 pi pi 1253 Feb 20 09:26 broadcast.csv
	-rw-r--r-- 1 pi pi  754 Feb 20 09:26 memory.csv
	-rw-r--r-- 1 pi pi 1228 Feb 20 09:26 _templates.csv
	drwxr-xr-x 2 pi pi 4096 Feb 20 09:26 vaillant

	pi@ebus2:~ $ ls -l ebusd-configuration/ebusd-2.1.x/de/
	total 16
	-rw-r--r-- 1 pi pi 1253 Feb 20 09:25 broadcast.csv
	-rw-r--r-- 1 pi pi  754 Feb 20 09:25 memory.csv
	-rw-r--r-- 1 pi pi 1228 Feb 20 09:25 _templates.csv
	drwxr-xr-x 2 pi pi 4096 Feb 20 09:25 vaillant