john30/ebusd #171: ebusd 3.1.v3.1-22-gdaf771e does not load csv file

CrazYoshi, 2018-02-26:
Hi,
after the update to version 3.1.v3.1-22-gdaf771e ebusd is not able to load the csv file for my VRC700 anymore.
Before with version 3.1.v3.0-35-gb0e20b7
```
pi@homedaemon:~ $ ebusctl info
version: ebusd 3.1.v3.0-35-gb0e20b7
signal: acquired
symbol rate: 25
max symbol rate: 85
min arbitration micros: 224
max arbitration micros: 262
min symbol latency: 7
max symbol latency: 12
reconnects: 0
masters: 3
messages: 363
conditional: 0
poll: 0
update: 8
address 03: master #11
address 08: slave #11
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0110;HW=2103", loaded "vaillant/15.700.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
```
and now:
```
version: ebusd 3.1.v3.1-22-gdaf771e
update check: revision v3.1-1-g60a18d1 available, broadcast.csv: different version available, memory.csv: different version available
signal: acquired
symbol rate: 24
max symbol rate: 118
min arbitration micros: 213
max arbitration micros: 1082
min symbol latency: 2
max symbol latency: 27
reconnects: 0
masters: 3
messages: 13
conditional: 0
poll: 0
update: 4
address 03: master #11
address 08: slave #11
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0110;HW=2103"
address 31: master #8, ebusd
address 36: slave #8, ebusd
```

john30, 2018-02-27:
please provide the ebusd.log file for the case when the csv is not loaded

CrazYoshi, 2018-02-27:
Hi john,
below the log after ebusd have found the Vaillant VRC700:
```
2018-02-26 22:32:17.623 [bus notice] scan 15: ;Vaillant;70000;0110;2103
2018-02-26 22:32:17.624 [update notice] store 15 ident: done
2018-02-26 22:32:17.624 [update notice] received scan.15  QQ=03: Vaillant;70000;0110;2103
2018-02-26 22:32:24.537 [update notice] received unknown MS cmd: 1008b5110101 / 0984830003ffff0000ff
2018-02-26 22:32:24.808 [update notice] received unknown MS cmd: 1008b51009000000ffffff050000 / 0101
2018-02-26 22:32:34.539 [update notice] received unknown MS cmd: 1008b5110101 / 0984820003ffff0000ff
2018-02-26 22:32:34.800 [update notice] received unknown MS cmd: 1008b51009000000ffffff050000 / 0101
2018-02-26 22:32:35.074 [update notice] received unknown MS cmd: 1008b5110102 / 06011496467864
2018-02-26 22:32:44.538 [update notice] received unknown MS cmd: 1008b5110101 / 0983820003ffff0000ff
2018-02-26 22:32:44.806 [update notice] received unknown MS cmd: 1008b51009000000ffffff050000 / 0101
2018-02-26 22:32:54.595 [update notice] received unknown MS cmd: 1008b5110101 / 0982810003ffff0000ff
2018-02-26 22:32:54.869 [update notice] received unknown MS cmd: 1008b51009000000ffffff050000 / 0101
2018-02-26 22:33:05.360 [update notice] received unknown MS cmd: 1008b5040100 / 0a00ffffffffffffff0003
2018-02-26 22:33:05.594 [update notice] received unknown BC cmd: 10feb516080024322226020118
2018-02-26 22:33:05.873 [update notice] received unknown MS cmd: 1008b5110101 / 0982810003ffff0000ff
2018-02-26 22:33:06.133 [update notice] received unknown MS cmd: 1008b51009000000ffffff050000 / 0101
2018-02-26 22:33:06.395 [update notice] received unknown MS cmd: 1008b5110102 / 06011496467864
2018-02-26 22:33:06.665 [update notice] received unknown MS cmd: 1008b5110100 / 09120411002218002200
2018-02-26 22:33:06.908 [update notice] received unknown MS cmd: 1008b5100305ff01 / 0101
2018-02-26 22:33:07.148 [update notice] received unknown MS cmd: 1008b512020000 / 00
2018-02-26 22:33:07.398 [update notice] received unknown MS cmd: 1008b5120204ff / 0101
2018-02-26 22:33:07.612 [update notice] received unknown BC cmd: 10feb51603010003
2018-02-26 22:33:07.856 [update notice] received unknown MS cmd: 1008b513020508 / 0101
2018-02-26 22:33:15.778 [update notice] received unknown MS cmd: 1008b5110101 / 0981800003ffff0000ff
```
Could be something related to this?
`2018-02-26 19:32:08.805 [main notice] update check: revision v3.1-1-g60a18d1 available, broadcast.csv: different version available`
I get the latest version from github and I have follow the instructions using make to install on my raspi, why it is keep telling me that there is a newer version available?

john30, 2018-03-03:
no, its not related to the update log entry.
but you might have just not waited long enough. please provide the ebusd.log with another minute of data after the "received scan.15" entry.
Anyway, it looks as ig you have started ebusd in readonly mode. please also provide the ebusd start parameters.

CrazYoshi, 2018-03-03:
I already seen that the version is newer than the one suggested by the update check, so maybe also this has to be fix later on.
Here the logs of 20 minutes later the successfull scan:
```
2018-02-28 13:15:51.000 [bus notice] scan 15: ;Vaillant;70000;0110;2103
2018-02-28 13:15:51.001 [update notice] store 15 ident: done
2018-02-28 13:15:51.001 [update notice] received scan.15  QQ=03: Vaillant;70000;0110;2103
2018-02-28 13:16:41.014 [main notice] update check: revision v3.1-1-g60a18d1 available, broadcast.csv: different version available, memory.csv: different version available
2018-02-28 13:17:53.395 [update notice] received unknown MS cmd: 1008b5110100 / 09b1041138040f000116
2018-02-28 13:17:53.643 [update notice] received unknown MS cmd: 1008b5100305ff01 / 0101
2018-02-28 13:17:53.894 [update notice] received unknown MS cmd: 1008b512020064 / 00
2018-02-28 13:17:54.138 [update notice] received unknown MS cmd: 1008b5120204ff / 0101
2018-02-28 13:17:54.349 [update notice] received unknown BC cmd: 10feb5160301f004
2018-02-28 13:17:54.602 [update notice] received unknown MS cmd: 1008b513020508 / 0101
2018-02-28 13:17:56.379 [update notice] received unknown MS cmd: 1008b5110101 / 0996832005ffff0100ff
2018-02-28 13:17:57.400 [update notice] received unknown MS cmd: 1008b510090000a0ffffff000000 / 0101
2018-02-28 13:18:02.910 [update notice] received unknown MS cmd: 1008b5110101 / 0995832005ffff0100ff
2018-02-28 13:18:03.175 [update notice] received unknown MS cmd: 1008b510090000a0ffffff000000 / 0101
2018-02-28 13:18:03.441 [update notice] received unknown MS cmd: 1008b5110102 / 06011496467864
2018-02-28 13:18:12.211 [update notice] received unknown MS cmd: 1008b5110101 / 0996832005ffff0100ff
2018-02-28 13:18:12.411 [update notice] received unknown MS cmd: 1008b510090000a0ffffff000000 / 0101
2018-02-28 13:18:22.155 [update notice] received unknown MS cmd: 1008b5110101 / 0995832005ffff0100ff
2018-02-28 13:18:22.424 [update notice] received unknown MS cmd: 1008b510090000a0ffffff000000 / 0101
2018-02-28 13:18:32.156 [update notice] received unknown MS cmd: 1008b5110101 / 0996832005ffff0100ff
2018-02-28 13:18:32.431 [update notice] received unknown MS cmd: 1008b5040100 / 0a00ffffffffffffff2005
2018-02-28 13:18:32.667 [update notice] received unknown BC cmd: 10feb516080013181328020318
2018-02-28 13:18:32.943 [update notice] received unknown MS cmd: 1008b510090000a0ffffff000000 / 0101
2018-02-28 13:18:33.197 [update notice] received unknown MS cmd: 1008b5110102 / 06011496467864
2018-02-28 13:18:33.475 [update notice] received unknown MS cmd: 1008b5110100 / 09b1041138040f000116
2018-02-28 13:18:33.718 [update notice] received unknown MS cmd: 1008b5100305ff01 / 0101
2018-02-28 13:18:33.970 [update notice] received unknown MS cmd: 1008b512020064 / 00
2018-02-28 13:18:34.200 [update notice] received unknown MS cmd: 1008b5120204ff / 0101
2018-02-28 13:18:34.422 [update notice] received unknown BC cmd: 10feb51603012005
2018-02-28 13:18:42.163 [update notice] received unknown MS cmd: 1008b5110101 / 0995832005ffff0100ff
2018-02-28 13:18:42.430 [update notice] received unknown MS cmd: 1008b510090000a0ffffff000000 / 0101
2018-02-28 13:18:52.450 [update notice] received unknown MS cmd: 1008b510090000a0ffffff000000 / 0101
2018-02-28 13:19:02.243 [update notice] received unknown MS cmd: 1008b5110101 / 0996832005ffff0100ff
2018-02-28 13:19:02.494 [update notice] received unknown MS cmd: 1008b510090000a0ffffff000000 / 0101
2018-02-28 13:19:02.755 [update notice] received unknown MS cmd: 1008b5110102 / 06011496467864
2018-02-28 13:19:12.235 [update notice] received unknown MS cmd: 1008b5110101 / 0995832005ffff0100ff
2018-02-28 13:19:12.496 [update notice] received unknown MS cmd: 1008b510090000a0ffffff000000 / 0101
2018-02-28 13:19:22.232 [update notice] received unknown MS cmd: 1008b5110101 / 0996832005ffff0100ff
2018-02-28 13:19:22.500 [update notice] received unknown MS cmd: 1008b510090000a0ffffff000000 / 0101
2018-02-28 13:19:32.231 [update notice] received unknown MS cmd: 1008b5110101 / 0996832005ffff0100ff
2018-02-28 13:19:32.506 [update notice] received unknown MS cmd: 1008b5040100 / 0a00ffffffffffffff2005
2018-02-28 13:19:32.744 [update notice] received unknown BC cmd: 10feb516080013191328020318
2018-02-28 13:19:33.021 [update notice] received unknown MS cmd: 1008b510090000a0ffffff000000 / 0101
2018-02-28 13:19:33.286 [update notice] received unknown MS cmd: 1008b5110102 / 06011496467864
2018-02-28 13:19:33.567 [update notice] received unknown MS cmd: 1008b5110100 / 09b1041138040f000116
2018-02-28 13:19:33.799 [update notice] received unknown MS cmd: 1008b5100305ff01 / 0101
2018-02-28 13:19:34.031 [update notice] received unknown MS cmd: 1008b512020064 / 00
2018-02-28 13:19:34.275 [update notice] received unknown MS cmd: 1008b5120204ff / 0101
2018-02-28 13:19:34.495 [update notice] received unknown BC cmd: 10feb51603012005
2018-02-28 13:19:42.238 [update notice] received unknown MS cmd: 1008b5110101 / 0995832005ffff0100ff
2018-02-28 13:19:42.508 [update notice] received unknown MS cmd: 1008b510090000a0ffffff000000 / 0101
2018-02-28 13:19:52.253 [update notice] received unknown MS cmd: 1008b5110101 / 0996832005ffff0100ff
2018-02-28 13:19:52.525 [update notice] received unknown MS cmd: 1008b510090000a0ffffff000000 / 0101
2018-02-28 13:20:02.307 [update notice] received unknown MS cmd: 1008b5110101 / 0995832005ffff0100ff
2018-02-28 13:20:02.581 [update notice] received unknown MS cmd: 1008b510090000a0ffffff000000 / 0101
[... cut ...]
2018-02-28 13:35:35.090 [update notice] received unknown MS cmd: 1008b5100305ff01 / 0101
2018-02-28 13:35:35.330 [update notice] received unknown MS cmd: 1008b512020064 / 00
2018-02-28 13:35:35.588 [update notice] received unknown MS cmd: 1008b5120204ff / 0101
2018-02-28 13:35:35.784 [update notice] received unknown BC cmd: 10feb51603012005
```
As you can see only unknown command are received.
Ebusd is started without readonly mode in the following way (taken from  /etc/default/ebusd):
`EBUSD_OPTS="-d 192.168.1.23:8889 --httpport=9092 -l /var/log/ebusd.log --scanconfig"`

john30, 2018-03-03:
ebusd does not send a single byte to the bus, so something is wrong.
please issue the following command and post the result:
`ps aux|grep ebusd`

CrazYoshi, 2018-03-03:
Here the result:
pi@homedaemon:~ $ ps aux|grep ebusd
pi        9359  0.0  0.4   4364  1916 pts/0    S+   10:22   0:00 grep --color=auto ebusd
root     16171  1.2  0.5  39728  2616 ?        Ssl  Mar01  33:05 /usr/bin/ebusd --pidfile /var/run/ebusd.pid -d 192.168.1.23:8889 --httpport=9092 -l /var/log/ebusd.log --scanconfig

john30, 2018-03-07:
ok that seems fine. can you post the same log as in your second post with the older ebusd version again? And without chaning anything else (like CSV files or so).

CrazYoshi, 2018-03-07:
I have done the following:
- removed the latest version taken from git (`sudo make uninstall`) 
- backup all csv files in `/etc/ebusd` folder
- install deb package `ebusd-3.1_armhf-wheezy.deb` (I have to remove the `/etc/ebusd` folder that cause an error during installation)
- Override `/etc/ebusd` folder with the one I have saved before
- Change `/etc/default/ebusd` to have the same configuration as before

After starting ebusd service it found and load csv file immediately:
```
pi@homedaemon:~ $ cat /var/log/ebusd.log
2018-03-07 09:37:36.891 [main notice] ebusd 3.1.v3.0-35-gb0e20b7 started with auto scan
2018-03-07 09:37:37.040 [bus notice] bus started with own address 31/36
2018-03-07 09:37:37.054 [bus notice] signal acquired
2018-03-07 09:37:42.771 [bus notice] new master 10, master count 2
2018-03-07 09:37:42.841 [bus notice] new master 03, master count 3
2018-03-07 09:37:42.842 [update notice] unknown MS cmd: 1008b5110101 / 094544e00cffff0000ff
2018-03-07 09:37:47.154 [main error] scan config 08: ERR: read timeout
2018-03-07 09:37:49.352 [bus notice] scan 15: ;Vaillant;70000;0110;2103
2018-03-07 09:37:49.407 [main notice] read common config file /etc/ebusd/vaillant/broadcast.csv
2018-03-07 09:37:49.411 [main notice] read common config file /etc/ebusd/vaillant/scan.csv
2018-03-07 09:37:49.414 [main notice] read common config file /etc/ebusd/vaillant/general.csv
2018-03-07 09:37:49.770 [main notice] read scan config file /etc/ebusd/vaillant/15.700.csv for ID "70000", SW0110, HW2103
2018-03-07 09:37:49.776 [main notice] found messages: 365 (0 conditional on 0 conditions, 0 poll, 11 update)
2018-03-07 09:37:52.864 [update notice] unknown MS cmd: 1008b5110101 / 094544e00cffff0000ff
2018-03-07 09:37:53.141 [update notice] unknown MS cmd: 1008b51009000000ffffff010000 / 0101
2018-03-07 09:38:02.871 [update notice] unknown MS cmd: 1008b5110101 / 094544e00cffff0000ff
2018-03-07 09:38:03.143 [update notice] unknown MS cmd: 1008b51009000000ffffff010000 / 0101

```

john30, 2018-03-11:
please go back to the version built from git and in addition to posting the ebusd.log file provide the output of the following commands:
```
ls -la /etc/ebusd/
ls -la /lib/systemd/system/ebusd.service && cat /lib/systemd/system/ebusd.service
ls -la /etc/init.d/ebusd && cat /etc/init.d/ebusd
cat /etc/default/ebusd

```

CrazYoshi, 2018-03-11:
Below the command result:
```
pi@homedaemon:~/ebusd $ ls -la /etc/ebusd/
total 24
drwxr-xr-x   3 root root 4096 Mar  7 09:36 .
drwxr-xr-x 103 root root 4096 Mar  7 09:17 ..
-rw-r--r--   1 root root 1457 Dec 26 17:38 broadcast.csv
-rw-r--r--   1 root root  744 Mar  7 09:21 memory.csv
-rw-r--r--   1 root root 1229 Mar  7 09:21 _templates.csv
drwxr-xr-x   2 root root 4096 Mar  7 09:21 vaillant
pi@homedaemon:~/ebusd $ ls -la /lib/systemd/system/ebusd.service && cat /lib/systemd/system/ebusd.service
ls: cannot access '/lib/systemd/system/ebusd.service': No such file or directory
pi@homedaemon:~/ebusd $ ls -la /etc/init.d/ebusd && cat /etc/init.d/ebusd
-rwxr-xr-x 1 root root 2366 Dec 26 17:38 /etc/init.d/ebusd
#!/bin/bash
#
### BEGIN INIT INFO
# Provides:        ebusd
# Required-Start:  $network $remote_fs $syslog
# Required-Stop:   $network $remote_fs $syslog
# Default-Start:   2 3 4 5
# Default-Stop:    0 1 6
# Short-Description: controls ebusd, the daemon for communication with eBUS heating systems.
### END INIT INFO

DAEMON=/usr/bin/ebusd
PIDFILE_PREFIX=/var/run/ebusd
PIDFILE_SUFFIX=.pid

. /lib/lsb/init-functions
[ -r /etc/default/ebusd ] && . /etc/default/ebusd

if [ ! -x $DAEMON ]; then
        log_failure_msg "$DAEMON is not available or not executable."
        exit 5
fi

ALL_OPTS=${!EBUSD_OPTS*}
if [ ${#ALL_OPTS[@]} -lt 1 ]; then
  ALL_OPTS[0]=EBUSD_OPTS
fi
instance="EBUSD_OPTS$2"
instance=${!instance}

start_instance () {
        local opts suffix
        opts=$1
        shift
        suffix="${opts#EBUSD_OPTS}"
        log_daemon_msg "Starting ebusd${suffix}" ebusd
        pidfile=$PIDFILE_PREFIX${suffix}$PIDFILE_SUFFIX
        start-stop-daemon --start --quiet --oknodo --pidfile $pidfile --exec $DAEMON -- --pidfile $pidfile ${!opts}
        log_end_msg $?
}

stop_instance () {
        local opts suffix
        opts=$1
        shift
        suffix="${opts#EBUSD_OPTS}"
        log_daemon_msg "Stopping ebusd${suffix}" ebusd
        pidfile=$PIDFILE_PREFIX${suffix}$PIDFILE_SUFFIX
        start-stop-daemon --stop --quiet --oknodo --pidfile $pidfile
        log_end_msg $?
        rm -f $PIDFILE_PREFIX${suffix}$PIDFILE_SUFFIX
}

case $1 in
        start)
                if [ -z "$2" ]; then
[... 163 lines cut ...]
2018-03-11 19:24:28.263 [update notice] received unknown MS cmd: 1008b5110102 / 06011496467864
2018-03-11 19:24:37.354 [update notice] received unknown MS cmd: 1008b5110101 / 096b77a00bffff0000ff
2018-03-11 19:24:37.610 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101
2018-03-11 19:24:47.633 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101
2018-03-11 19:24:57.387 [update notice] received unknown MS cmd: 1008b5110101 / 097174a00bffff0000ff
2018-03-11 19:24:57.649 [update notice] received unknown MS cmd: 1008b5040100 / 0a00ffffffffffffffa00b
2018-03-11 19:24:57.880 [update notice] received unknown BC cmd: 10feb516080027241911030718
2018-03-11 19:24:58.151 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101
2018-03-11 19:24:58.438 [update notice] received unknown MS cmd: 1008b5110102 / 06011496467864
2018-03-11 19:24:58.687 [update notice] received unknown MS cmd: 1008b5110100 / 09890312001138000200
2018-03-11 19:24:58.935 [update notice] received unknown MS cmd: 1008b5100305ff01 / 0101
2018-03-11 19:24:59.175 [update notice] received unknown MS cmd: 1008b512020064 / 00
2018-03-11 19:24:59.414 [update notice] received unknown MS cmd: 1008b5120204ff / 0101
2018-03-11 19:24:59.640 [update notice] received unknown BC cmd: 10feb5160301a00b
2018-03-11 19:25:07.383 [update notice] received unknown MS cmd: 1008b5110101 / 097271a00bffff0000ff
2018-03-11 19:25:07.659 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101
2018-03-11 19:25:17.394 [update notice] received unknown MS cmd: 1008b5110101 / 097270a00bffff0000ff
2018-03-11 19:25:17.657 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101
2018-03-11 19:25:27.401 [update notice] received unknown MS cmd: 1008b5110101 / 097270a00bffff0000ff
2018-03-11 19:25:27.681 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101
2018-03-11 19:25:27.931 [update notice] received unknown MS cmd: 1008b5110102 / 06011496467864
2018-03-11 19:25:37.440 [update notice] received unknown MS cmd: 1008b5110101 / 097270a00bffff0000ff
2018-03-11 19:25:37.705 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101
2018-03-11 19:25:48.267 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101
```

john30, 2018-03-14:
I'm afraid I can't tell you the reason for this behaviour right now.
Could you set the log level to debug and post that log file (as file attachment please)?

CrazYoshi, 2018-03-15:
I have collected the log as requested with loglevel debug.
[ebusd.log](https://github.com/john30/ebusd/files/1816761/ebusd.log)

john30, 2018-03-18:
it seems this is related to a rather arbitrary timing issue.
please add the parameter "--latency=50000" to /etc/default/ebusd and try again.
why do you do a full scan anyway? it does not reveal any further but takes a lot of time (up to 3 minutes).

CrazYoshi, 2018-03-18:
I will change the configuration file and I will to another try.
I usually do a full scan cause the initial one seems doesn't find VR700, so I perform a new one.

CrazYoshi, 2018-03-24:
I have tried and now ebusd have loaded the csv file. Thanks john!

closed 2018-03-24 by CrazYoshi