john30/ebusd #174: Connection lost after couple of days / restart required

stoxn, 2018-03-10:
Hi,

Thanks a lot for ebusd - it finally makes my Vaillant heating system working as per our requirements and helps saving energy.

However, now that my rules are stable and I just let it run, I receiver no data every couple of days and only ebusd restart resolves it.

Here is my rough setup:
- Commercial Ethernet adapter from esera
- ebusd runs on CentOS with an edomi installation
- Connection to ebusd via PHP / Socket (edomi LBS)
- Reading data from eBus every 10 minutes
- Writing data to eBus on demand - couple of times per day

After one week, I have seen in my Graphs that no further data was received. Below is the issue that occurred during the failure.

My current plan is, to restart every day or in case no signal received for 1h.
Any suggestion would be appreciated.

Best,
Patrick

stoxn, 2018-03-10:
At time of issue - the ACK errors were there over whole week:

2018-03-09 17:36:12.256 [main error] scan config 23: ERR: ACK error
2018-03-09 17:36:24.768 [main error] scan config 23: ERR: ACK error
2018-03-09 17:36:37.026 [main error] scan config 23: ERR: ACK error
2018-03-09 17:36:49.323 [main error] scan config 23: ERR: ACK error
2018-03-09 17:37:01.614 [main error] scan config 23: ERR: ACK error
2018-03-09 17:37:13.773 [main error] scan config 23: ERR: ACK error
2018-03-09 17:37:23.038 [bus error] poll ui YieldThisYear failed: ERR: read timeout
2018-03-09 17:37:26.012 [bus error] signal lost
2018-03-09 17:37:26.012 [main error] scan config 23: ERR: no signal
2018-03-09 17:37:56.206 [main error] scan config 23: ERR: no signal
2018-03-09 17:38:26.399 [main error] scan config 23: ERR: no signal
2018-03-09 17:38:39.400 [bus error] unable to open 192.168.15.120:5000: ERR: generic I/O error
2018-03-09 17:38:52.401 [bus error] unable to open 192.168.15.120:5000: ERR: generic I/O error
2018-03-09 17:39:05.402 [bus error] unable to open 192.168.15.120:5000: ERR: generic I/O error
2018-03-09 17:39:05.402 [main error] scan config 23: ERR: no signal
2018-03-09 17:39:18.403 [bus error] unable to open 192.168.15.120:5000: ERR: generic I/O error
2018-03-09 17:39:31.404 [bus error] unable to open 192.168.15.120:5000: ERR: generic I/O error
2018-03-09 17:39:44.405 [bus error] unable to open 192.168.15.120:5000: ERR: generic I/O error
2018-03-09 17:39:44.405 [main error] scan config 23: ERR: no signal
2018-03-09 17:39:57.406 [bus error] unable to open 192.168.15.120:5000: ERR: generic I/O error
2018-03-09 17:40:10.408 [bus error] unable to open 192.168.15.120:5000: ERR: generic I/O error
2018-03-09 17:40:10.408 [bus error] send to 06: ERR: no signal, give up

One day later:
2018-03-10 12:00:01.562 [bus error] send to 08: ERR: no signal, give up
2018-03-10 12:00:01.562 [bus error] send message part 0: ERR: no signal
2018-03-10 12:00:01.566 [bus error] send to 06: ERR: no signal, give up
2018-03-10 12:00:01.566 [bus error] send message part 0: ERR: no signal
2018-03-10 12:00:01.573 [bus error] send to 0a: ERR: no signal, give up
2018-03-10 12:00:01.573 [bus error] send message part 0: ERR: no signal
2018-03-10 12:00:01.579 [bus error] send to 50: ERR: no signal, give up
2018-03-10 12:00:01.579 [bus error] send message part 0: ERR: no signal
2018-03-10 12:00:03.318 [main error] scan config 23: ERR: no signal
2018-03-10 12:00:03.318 [bus error] send to 50: ERR: no signal, give up
2018-03-10 12:00:03.318 [bus error] send message part 0: ERR: no signal
2018-03-10 12:00:03.318 [main error] write mc OperatingMode: ERR: no signal
2018-03-10 12:00:33.320 [main error] scan config 23: ERR: no signal
2018-03-10 12:01:03.321 [main error] scan config 23: ERR: no signal
2018-03-10 12:01:33.323 [main error] scan config 23: ERR: no signal
2018-03-10 12:02:03.325 [main error] scan config 23: ERR: no signal
2018-03-10 12:02:33.327 [main error] scan config 23: ERR: no signal
2018-03-10 12:03:03.329 [main error] scan config 23: ERR: no signal
2018-03-10 12:03:33.331 [main error] scan config 23: ERR: no signal

john30, 2018-03-11:
this seems to be related to the not officially supported esera Ethernet adapter. You could try to switch to UDP mode in the adapter and add "--initsend" to the ebusd parameters.

stoxn, 2018-03-11:
Ah, I was not aware it is not supported. I have restarted the device and started ebusd with --initsend as suggested.

I was not able to run it in UDP mode as different settings did not work. However, it runs smooth again and I will monitor it. With a watchdog I will be informed in case the connection is not working again.

Thanks for the help - this issue can be closed for now.

closed 2018-03-11 by stoxn