john30/ebusd #177: Support constant fields

msteiger, 2018-03-18:
Many of Wolf's proprietary messages have a constant suffix. Here's an example:

`3003502309d8420104005d010000`

from 30 to 03, 
the "Wolf-Write-Command": `50 23`
9 bytes following
The crc8 of the following 8 bytes (see https://github.com/john30/ebusd/issues/167)
The ID for "HG07" (Heizgerät Parameter 07) `42 01`
The actual value: 04 00
**The 4-byte suffix:** `5d 01 00 00`

I can specify it as a unsigned integer to make it work, but it's actually a constant. So I suggest adding a constant field type, together with some way of defining the constant. Maybe in the "divider/values" column?

john30, 2018-03-18:
already available, e.g. use `HEX:4,=5d010000`

closed 2018-03-18 by john30

msteiger, 2018-03-18:
Oh, that's perfect. Thanks for helping me out.