john30/ebusd #178: Weishaupt WTC 15-A problem

Dollargrin, 2018-04-01:
Hello

I got a Weishaupt WTC 15-A heater, an eBus Koppler USB from Esera and a raspberrypi running ebusd 3.0pre.bbc4d04. The hardware seems to be working, I get 20 <aa per second. I was hoping configuration from JOEK3R would work for my Weishaupt device so I cloned his configuration files like this

git clone https://github.com/J0EK3R/ebusd-configuration-weishaupt.git /etc/ebusd/

“ebusctl scan result” gives me

1 scan(s) still running
08;Kromschroeder;W ;0306;0301
f6;Kromschroeder;WWST?;0306;0301

There is only the Weishaupt heater on the line + the raspberry. The problem is that I don’t know if things are working. I would like to try getting data with 
ebusctl read XXX but don’t know what is available. Can somebody give me some pointers? Here’s what scanconfig gave:

ebusd -f --scanconfig
2018-04-01 05:18:36.289 [main notice] ebusd 3.0pre.bbc4d04 started with auto scan
2018-04-01 05:18:36.300 [bus notice] bus started with own address 31/36
2018-04-01 05:18:36.304 [bus notice] signal acquired
2018-04-01 05:20:46.499 [main notice] update check: version 3.1 available
2018-04-01 05:29:25.083 [bus notice] new master f1, master count 2
2018-04-01 05:29:25.085 [update notice] unknown BC cmd: f1fe500a0d01067f52545aff6c0001ff0108
2018-04-01 05:29:26.631 [main error] scan config f6: ERR: CRC error
2018-04-01 05:29:38.723 [main error] scan config f6: ERR: read timeout
2018-04-01 05:31:40.906 [main notice] update check: version 3.1 available
2018-04-01 05:39:26.410 [update notice] unknown BC cmd: f1fe500a0d01067f522070ff6c0001ff0108

I’m new at this so beginners mistake is not out of question

thanks

john30, 2018-04-01:
for what's available, [see the find command](https://github.com/john30/ebusd/wiki/3.1.-TCP-client-commands#find)

closed 2018-04-01 by john30

Dollargrin, 2018-04-12:
an "ebusctl find -v" gives many lines similar to "hc3 HeatingDemand = no data stored" but also this line:

sc Act = Status1=1;Operatingphase=BrennerInBetrieb;Ukn2_1=1;Ukn2_2=1;Ukn2_3=1;Flame=1;GasValve1=1;GasValve2=1;Pump=1;Error=0;Ukn3_1=0;SoWi=Winter;Ukn3_3=0;Ukn3_4=0;Ukn3_5=0;Ukn3_6=0;Ukn3_7=1;SettingUV=Heating;Load=56;SupplyTemp=35.0;FlueGasTemp=-;DHWTemp=53.0;UknTemp=0.0;ExternalTemp=11;TrendTemp=14.707;SupplySetTemp=8

For example ExternalTemp and DHWTemp seem to be working here but not others, e.g. Flame which is always true.  Can anyone point me in the right direction to get more data out of my heater?

john30, 2018-04-15:
ebusd does not automatically issue reads on the bus unless a message is set to a poll priority and polling is enabled in ebusd. other than that, you have to query the messages yourself