john30/ebusd #181: Can't find vrc 700 on ebus?

minitrue-nl, 2018-04-07:
Just started with trying to get ebusd to work on my Vaillant Heatpump. I'm half way through I think:
- installed and started all the software
- seems to work fine, but only finds two devices: MHU01 on address 08 (loaded csv) and VWZIO on address 76 (no config available)
I was hoping however also to get a read from my VRC 700 (kind of the point of this all). However I don't see anything popping up on starting ebusd both manually and automatically.
Am I doing something wrong? Or could it be possible that I've somehow hooked up to a wrong ebus-wire?
Help is appreciated!

john30, 2018-04-08:
which adapter do you have?
if it has a potentiometer, you need to adjust that very good first

minitrue-nl, 2018-04-12:
Hi John, I bought the commercial adapter by esera. It picked up the signal right away without adjustment, and I just calibrated again just now to check. However it only picks up the MHU01 (succesfull) and VWZIO (no configfile). But I also have a VRC 700 in my system.

Could it be that I plugged in at the wrong ebus that the VRC700 is not connected to? I thought that (because its a bus) every device can be picked up from all connections?
Or do I actually need to do something on the VRC 700 so that it will send a signal while configuring ebusd?

minitrue-nl, 2018-04-12:
And an update after trying to set the potentiometer again... It now found the VRC 700 as well!
However I found it very hard to adjust the potentiometer to find all the devices on the network. In the instructions it says you're good to go as soon as you don't see any 'signals lost', but I guess it was not finely tuned enough to pick up all devices...

modenet, 2018-04-13:
same problem for VWZIO: no config file

minitrue-nl, 2018-04-13:
@modenet Yes, but that was not really my problem here... There is not much info to get from the VWZIO in my opinion.

john30, 2018-04-15:
the poti adjustment is a little horror, thats why we developed a new hardware without poti, see here: https://ebus.github.io/adapter/index.en.html

closed 2018-04-15 by john30