john30/ebusd #183: problem identifying slaves, possible config

caviarbag, 2018-04-16:
Hi, for about 2 months ebusd has been running fine with the 2.0 adapter. 3 days ago I restarted the machine running ebusd and it stopped working. Over the weekend I also restarted the whole shabang by cutting the power to everything without getting any progress.

What is strange is that ebusd was running in a container, allthough not the official image from the hub it is built on same Dockerfile but on centos-image. Messagefiles was pulled from 2.1.x/en.

I have tried all the images from dockerhub (latest,devel,v3.1) and rebuilding my own.

One thing that baffles me is that if container is started using -f -scanconfig it seems to pull some configfiles from http://ebusd.eu/config/
```2018-04-16 13:53:00.387 [main notice] ebusd 3.1.p20180416 started with auto scan\n
2018-04-16 13:53:00.387 [main info] loading configuration files from http://ebusd.eu/config/
2018-04-16 13:53:00.601 [main info] reading templates /
2018-04-16 13:53:00.679 [main info] read templates in /
2018-04-16 13:53:00.679 [main info] reading file memory.csv
2018-04-16 13:53:00.756 [main info] successfully read file memory.csv
2018-04-16 13:53:00.756 [main info] reading file broadcast.csv
2018-04-16 13:53:00.834 [main info] successfully read file broadcast.csv
2018-04-16 13:53:00.834 [main info] read config files
```
The container that ran ok for a long time was created by
`docker run --name ebusd -p 8888:8888 -p 8080:8080 --restart=unless-stopped -d centos-ebusd -f --scanconfig -d tcp:192.168.150.37:9999 --latency=80000 --httpport=8080;`

After playing around i found that setting `--configpath=/etc/ebusd` will prevent automatic download.

Running checkconfig returns
```
[root@6b0b00fc3c5e /]# ebusd --checkconfig --configpath=/etc/ebusd
2018-04-16 14:04:08.858 [main notice] ebusd 3.1.p20180416 performing configuration check...
2018-04-16 14:04:10.133 [main notice] found messages: 11134 (437 conditional on 154 conditions, 17 poll, 64 update)
2018-04-16 14:04:10.179 [main notice] ebusd stopped
```

Attaching ebusd debuglevel output.
[ebusd.txt](https://github.com/john30/ebusd/files/1915169/ebusd.txt)

Most likely I am overlooking something and its a silly bruno, but Im out of ideas and hope someone can help.

caviarbag, 2018-04-18:
I spent two more evenings trying to figure this out. My ebus have these devices connected
```
address 00: master #1
address 03: master #11
address 05: slave #1, scanned "MF=Vaillant;ID=VR920;SW=0209;HW=5703"
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0307;HW=0403", loaded "vaillant/08.hmu.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0419;HW=4603", loaded "vaillant/15.700.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 52: slave, scanned "MF=Vaillant;ID=VR_70;SW=0109;HW=2903", loaded "vaillant/52.vr_70.csv"
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZ00;SW=0307;HW=0403"
```
Sometimes after an hour one or two devices was identified but not always with a messagefile.

Alot of read timeouts, ACK, SYN, CRC and so on. The adapter showed no sign of trouble; steady yellow light, fast flickering green and no red flashes.

Tonight I setup the USB adapter and everything started to work instantly. What I dont understand is why it worked so well for so long, then after a reboot of the machine running ebusd everything stopped.

closed 2018-04-18 by caviarbag

john30, 2018-04-19:
is this still an issue or did you solve it (since you closed it again)?

caviarbag, 2018-04-19:
I suspect its something with the adapter, not ebusd.