john30/ebusd #187: HTTP does not work

semenoh, 2018-04-29:
Here is my `/etc/default/ebusd` file:
```
EBUSD_OPTS="--scanconfig -d /dev/ttyUSB0 -p 8888 --httpport=8889 --latency=20000 --loglevel=error
```
But ebusd works only for TCP:

`pi@raspberrypi:~ $ sudo netstat -tulpn | grep LISTEN`
```
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      562/sshd            
tcp        0      0 0.0.0.0:8888            0.0.0.0:*               LISTEN      903/ebusd           
tcp6       0      0 :::22                   :::*                    LISTEN      562/sshd            
tcp6       0      0 :::443                  :::*                    LISTEN      783/domoticz        
tcp6       0      0 :::6144                 :::*                    LISTEN      783/domoticz        
tcp6       0      0 :::8880                 :::*                    LISTEN      783/domoticz 
```
Do you have any suggestions on how do I figure out why is it so ?
Thanks in advance.

john30, 2018-04-30:
seems to me as if the configuration file is not being read. can you try running ebusd in foreground to check? so first stop the daemon and then use the same options together with the prefix "ebusd -f " in a separate session

semenoh, 2018-04-30:
Long story short `ebusctl scan result` shows different result:
```
08;Vaillant;BAI00;0316;0903;21;17;06;0010015325;3100;005007;N3
-----------
1 scan(s) still running
08;Vaillant;BAI00;0316;0903
```
`ebusctl find` has one line different in the end:
```
scan.08 id = 21;17;06;0010015325;3100;005007;N3
--------
scan.15  = no data stored
scan.26  = no data stored
```
`ebusctl i` has different output:
```
version: ebusd 3.1.v3.1-22-gdaf771e
signal: acquired
symbol rate: 14
max symbol rate: 117
min arbitration micros: 3215
max arbitration micros: 3333
min symbol latency: 5
max symbol latency: 5
reconnects: 0
masters: 2
messages: 209
conditional: 3
poll: 0
update: 8
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0316;HW=0903", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
-----------------------------
version: ebusd 3.1.v3.1-22-gdaf771e
signal: acquired
symbol rate: 13
max symbol rate: 34
min arbitration micros: 3221
max arbitration micros: 3347
min symbol latency: 4
max symbol latency: 5
reconnects: 0
masters: 2
messages: 179
conditional: 7
poll: 0
update: 8
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0316;HW=0903"
address 31: master #8, ebusd
address 36: slave #8, ebusd
````

Is there a chance it runs different versions or use different configs for foreground and daemon ?
Or it may be due to the permissions ?

john30, 2018-05-01:
I think you might have installed the one release version that didn't care for the settings in /etc/default/ebusd. can you try building from source code?
or did you install a release in parallel to a build from source?

semenoh, 2018-05-01:
At the beginning I tried to build from sources, but then I figured out than it is possible to install .deb package so I installed it. 
Do you mean that I installed version that contains a bug and I should reinstall or update ebusd  by this  
> installed the one release version that didn't care for the settings in /etc/default/ebusd
?

john30, 2018-05-02:
then you might just have two ebusd executables on your system in different paths and if you start it as a service, the custom-built executable is probably missing as it is not in the path I guess

semenoh, 2018-05-03:
OK, I reinstalled ebusd and configs and now it works. I'll describe what exactly I did just in case someone else will experience the same issue.
And thanks a lot for your support.

I've uninstalled `ebusd` like this
```
pi@raspberrypi:~ $ sudo dpkg -P  ebusd-configuration
(Reading database ... 36066 files and directories currently installed.)
Removing ebusd-configuration (2.1.b143f39-en) ...
```
```
pi@raspberrypi:~ $ sudo dpkg -P  ebusd
(Reading database ... 35976 files and directories currently installed.)
Removing ebusd (3.1) ...
```
and than installed them again like this:
```
pi@raspberrypi:~ $ sudo dpkg -i --force-overwrite ebusd-3.1_armhf-jessie.deb
Selecting previously unselected package ebusd.
(Reading database ... 35970 files and directories currently installed.)
Preparing to unpack ebusd-3.1_armhf-jessie.deb ...
Unpacking ebusd (3.1) ...
Setting up ebusd (3.1) ...
Instructions:
1. Edit /etc/default/ebusd if necessary
   (especially if your device is not /dev/ttyUSB0)
2. Place CSV configuration files in /etc/ebusd/
   (see https://github.com/john30/ebusd-configuration)
3. To start the daemon, enter 'service ebusd start'
4. Check the log file /var/log/ebusd.log
```
changed config:
```
pi@raspberrypi:~ $ sudo nano /etc/default/ebusd
pi@raspberrypi:~ $ cat /etc/default/ebusd
# /etc/default/ebusd:
# config file for ebusd service.

# Options to pass to ebusd (run "ebusd -?" for more info):
EBUSD_OPTS="--scanconfig -d /dev/ttyUSB0 -p 8888 --httpport 8889 --latency=20000 --loglevel=error"

# MULTIPLE EBUSD INSTANCES WITH SYSV
# In order to run multiple ebusd instances on a SysV enabled system, simply
# define several EBUSD_OPTS with a unique suffix for each. Recommended is to
# use a number as suffix for all EBUSD_OPTS settings. That number will then be
# taken as additional "instance" parameter to the init.d script in order to
# start/stop an individual ebusd instance instead of all instances.
# Example: (uncomment the EBUSD_OPTS above)
#EBUSD_OPTS1="--scanconfig -d /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A50285BI-if00-port0 -p 8888 -l /var/log/ebusd1.log"
#EBUSD_OPTS2="--scanconfig -d /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A900acTF-if00-port0 -p 8889 -l /var/log/ebusd2.log"
#EBUSD_OPTS3="--scanconfig -d /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A900beCG-if00-port0 -p 8890 -l /var/log/ebusd3.log"

# MULTIPLE EBUSD INSTANCES WITH SYSTEMD
# In order to run muiltiple ebusd instances on a systemd enabled system, just
# copy the /usr/lib/systemd/system/ebusd.service file to /etc/systemd/system/
# with a different name (e.g. ebusd-2.service), remove the line starting with
# 'EnvironmentFile=', and replace the '$EBUSD_OPTS' with the options for that
# particular ebusd instance.
```
installed configs
```
pi@raspberrypi:~ $ sudo dpkg -i --force-overwrite ebusd-configuration-2.1.b143f39-en_all.deb
Selecting previously unselected package ebusd-configuration.
(Reading database ... 35977 files and directories currently installed.)
Preparing to unpack ebusd-configuration-2.1.b143f39-en_all.deb ...
Unpacking ebusd-configuration (2.1.b143f39-en) ...
dpkg: warning: overriding problem because --force enabled:
dpkg: warning: trying to overwrite '/etc/ebusd/broadcast.csv', which is also in package ebusd 3.1
Setting up ebusd-configuration (2.1.b143f39-en) ...
```
restarted the system and ran `ebusd`:
```
pi@raspberrypi:~ $ sudo service ebusd start
pi@raspberrypi:~ $ ebusdctl scan full
-bash: ebusdctl: command not found
pi@raspberrypi:~ $ ebusctl scan full
done

pi@raspberrypi:~ $ ebusctl scan result
1 scan(s) still running
08;Vaillant;BAI00;0316;0903
```

closed 2018-05-03 by semenoh