john30/ebusd #188: mqtt dash data not refreshing

marnikvd, 2018-05-01:
Hi John,

How and when do the values in [mqtt dash](https://github.com/john30/ebusd/wiki/7.-Integrations) refresh?
I tried reading temperature data following issue #52 but it is not refreshing in mqtt dash.
I can force a refresh by restarting the ebusd daemon and then reading the temperature :
 ```
$ :~/brink $ sudo service ebusd restart
 $ :~/brink $ ebusctl r -f -c kwl InsideTemperature
20.5
```

I have an uptime (ebusd/global/uptime) value which works fine and refreshes every ±16 seconds.

My configuration is described [here](https://github.com/marnikvd/brink-renovent-excellent-RPI-ebusd).
My ebusd config is :
`/usr/bin/ebusd -d /dev/ttyebus -p 8888 -l /var/log/ebusd.log --scanconfig --httpport=8080 --mqttport=1883 --mqtthost=localhost --sendretries=5`
When I do this command on the command line :
 `$ : ebusctl r -f -p 9 -c kwl InsideTemperature`
I see in the log file that the temperature is read every 6 seconds (shouldn't this be every 5(default)x9=45 seconds ?)
I also tried a crontab to read temperatures every 15 minutes but mqtt dash is not updated neither.  What am I missing?

andig, 2018-05-01:
You can force a refresh by adding a `/get` (https://github.com/john30/ebusd/wiki/3.3.-MQTT-client).

andig, 2018-05-01:
I'm not using mqtt dash. I have noticed though when using node-red- that I need to send a value to `/get`, too to make it actually send the request.

marnikvd, 2018-05-01:
I'm sorry, but this doesn't make sense to me.  I wouldn't know where I must add this `/get`.

andig, 2018-05-01:
Its simply the postfix of the topic:

    ebusd/700/Hc1FlowTemp/get

for example.

marnikvd, 2018-05-01:
As mentioned before, putting `/get` at the end of the topic in MQTT Dash has no influence in my case.
Nevertheless, I guess it's working now.  Don't ask me what the difference is with my settings before : I don't know...

I have a crontab as follows :
`0,15,30,45 * * * * ~/ebusd/contrib/scripts/readall.sh -c kwl InsideTemperature`

ebusd options In /etc/default/ebusd are as in my first post in this issue.

In MQTT Dash :
`Topic = ebusd/kwl/InsideTemperature`

Although one should expect the temperature to refresh every 15 minutes, this isn't always the case.  The refresh often happens after 30 minutes (or more).  When i check the log in this case i can't see any errors.  Why there are missing refreshes remains a mystery for me.  I will keep an eye on this the following days.

andig, 2018-05-01:
As I wrote above- did you try using ˋ/getˋ and actually send a value instead of an empty message? With eclipse as broker that was necessary for me.

john30, 2018-05-02:
the mqtt messages are only sent when the actual content changes, so if e.g. the temperature stays the same over 30 minutes, no mqtt topic will be updated at all.

closed 2018-05-02 by john30

andig, 2018-05-02:
> the mqtt messages are only sent when the actual content changes, so if e.g. the temperature stays the same over 30 minutes, no mqtt topic will be updated at all.

@john30 I assume that means:

  - a forced read via `ebusctl` does not trigger a mqtt message if the value does not change
  - a forced read via mqtt `/get`  does not trigger a mqtt message if the value does not change (???)

is that correct?

john30, 2018-05-06:
reading via ebusctl does not force a topic to be sent on mqtt unless the data has changed. reading via /get will always force an mqtt topic being sent

marnikvd, 2020-12-20:
Hi,
I left this problem to sleep for several years, but now I was able to solve it.
You have to add `--mqttretain` in the  ebusd config (mentioned in my first post) in order to tell the mqtt broker to keep the last values.  When [mqtt dash](https://github.com/john30/ebusd/wiki/7.-Integrations) connects, you get the most recent values.
Next you have to keep the values in the mqtt broker up to date.
To update the broker with the most recent values I put this line in the crontab, so the value is refreshed each 15 minutes :
(for instance : YourReadValue can be a temperature and YourWriteValue can be an fan speed)
`*/15 * * * * ebusctl read -f YourReadValue > /dev/null`
You can keep track of these values in a terminal with the command :
`mosquitto_sub -v -t  ebusd/yourcircuit/YourReadValue`
As john30 mentioned above, keep in mind that this will only write a value to the broker if it has changed.
In mqtt dash :
In a read widget you have to set `Topic(sub)` to `ebusd/yourcircuit/YourReadValue`
In a read/write widget you have to set `Topic(sub)` to `ebusd/yourcircuit/YourWriteValue` and in addition, with publishing enabled, you set `Topic(pub)` to `ebusd/yourcircuit/YourWriteValue/set`

About reading via /get :
This was confusing for me although John explains it [here](https://github.com/john30/ebusd/wiki/3.3.-MQTT-client).
To experiment with it you can open 2 terminal windows.  In the first one you log the broker values with the mosquitto_sub command from above.
In the second window you can **force** a write to the broker (even if the value hasn't changed) whith this command :
`mosquitto_pub -n -t ebusd/yourcircuit/YourReadValue/get`
So you do a write command (publish) in one window in order to read the wanted result in another window !