john30/ebusd #190: How do I write FlowTempDesired ?

semenoh, 2018-05-06:
I tried to change desired temperature on the boiler and read data to figure out which one to change. So I figured out that `FlowTempDesired` is changing. Than I tried to write it but got failed:
```
pi@raspberrypi:~ $ ebusctl w -c bai FlowTempDesired 32
ERR: element not found
```
I've tried to lookup config file
```
pi@raspberrypi:~ $ cat /etc/ebusd/vaillant/bai.308523.inc | grep FlowTempDesired
r,,FlowTempDesired,d.05 Vorlaufsolltemperatur,,,,"3900",,,temp,,,Vorlaufsolltemperatur oder Rücklaufsolltemperatur (wenn Rücklaufregelung aktiviert wurde). Der Maximalwert wird über d.71 und einem eBUS Regler begrenzt.
r,,ExtFlowTempDesiredMin,d.09 Min. ext. Vorlaufsollwert,,,,"6E04",,,temp,,,minimum out of Kl.7 and eBus flow setpoint
r,,externalFlowTempDesired,ext. Vorlaufsollwert,,,,"2500",,,temp,,,Vorlaufsollwert von einem externen Regler an Klemme 7-8-9
```
And It looks like it is read only field. So the question is why is it read only and which one should I write to tune temperature ?
Here is all I can write:
```
pi@raspberrypi:~ $ ebusctl f -w
bai AccessoriesOne = no data stored
bai AccessoriesTwo = no data stored
bai BlockTimeHcMax = no data stored
bai clearerrorhistory = no data stored
bai DSNOffset = no data stored
bai FanSpeedOffsetMax = no data stored
bai FanSpeedOffsetMin = no data stored
bai FlowsetHcMax = no data stored
bai FlowsetHwcMax = no data stored
bai HcPumpMode = no data stored
bai HoursTillService = no data stored
bai HwcPostrunTime = no data stored
bai HwcTempMax = no data stored
bai PartloadHcKW = no data stored
bai PartloadHwcKW = no data stored
bai ReturnRegulation = no data stored
bai SecondPumpMode = no data stored
bai SetFactoryValues = no data stored
bai SolarPostHeat = no data stored
bai StorageLoadTimeMax = no data stored
bai ValveMode = no data stored
bai WarmstartOffset = no data stored
bai WPPostrunTime = no data stored
broadcast ident = no data stored
broadcast queryexistence = no data stored
memory eeprom = no data stored
memory ram = no data stored
```

BTW `/ets/default/ebusd` has `--accesslevel="*"` option:

```
EBUSD_OPTS="--scanconfig -d /dev/ttyUSB0 -p 8888 --httpport 8889 --latency=20000 --loglevel=error --accesslevel="*""
```

john30, 2018-05-10:
to be honest: I don't know. this highly depends on the variant of heater you are using. you can try extending the message to support the write as well, but -as always- there is no guarantee that this will work.

semenoh, 2018-05-22:
There is no special use. I'm new to all this stuff and I have no idea what all these `bai`, `Hc`, `Hcw` etc mean. By trying to figure out how to control my boiler  I noticed that `FlowTempDesired` is the only field that correspond to value I set on boiler...
Regarding `FlowsetHcMax` something strange going on:

```
pi@raspberrypi:~ $ ebusctl r FlowsetHcMax
75.00

pi@raspberrypi:~ $ ebusctl w -c bai FlowsetHcMax 50
done

pi@raspberrypi:~ $ ebusctl r FlowsetHcMax
75.00
```

And it does not start heating (ATM boiler is in summer mode). 
What I expect is something I do manually on boiler by adjusting temperature but done programmatically.

semenoh, 2018-12-16:
Hi
I'm still not able to write anything to my boiler. Currently I'm trying to use `SetMode` as I see this message in `ebusctl listen` when I change temperature on thermostat.
so here is what I see:
```
bai Status01 = 38.5;38.0;-;-;19.0;off
bai Status01 = 37.5;38.0;-;-;19.0;off
bai SetMode = auto;90.0;-;-;0;0;0;0;0;0
bai Status01 = 41.5;38.0;-;-;19.0;off
bai Status02 = on;60;75.0;70;65.0
bai Status01 = 40.5;37.5;-;-;19.0;on
bai Status01 = 41.5;38.0;-;-;19.0;on
```
Here is what I'm trying to do:
```
pi@raspberrypi:~ $ ebusctl write -c bai SetMode auto;90.0;-;-;0;0;0;0;0;0
ERR: element not found

pi@raspberrypi:~ $ ebusctl write -c bai SetMode auto\;90.0\;-\;-\;0\;0\;0\;0\;0\;0
ERR: element not found

pi@raspberrypi:~ $ ebusctl write -c SetMode 0\;90.0\;-\;-\;0\;0\;0\;0\;0\;0
ERR: element not found

pi@raspberrypi:~ $ ebusctl write -c SetMode "0\;90.0\;-\;-\;0\;0\;0\;0\;0\;0"
ERR: element not found
```
Here are some other outputs that might be halpefull:
```
pi@raspberrypi:~ $ ebusctl f -f -w -c bai SetMode
ERR: element not found

pi@raspberrypi:~ $ ebusctl f -f -w
w,broadcast,id,identification,,fe,0704,,
w,broadcast,queryexistence,Inquiry of existence,,fe,07fe,,

pi@raspberrypi:~ $ ebusctl f -f SetMode
uw,bai,SetMode,Operation Mode,,08,b510,00,hcmode,m,UCH,0=auto;1=off;2=water,,boiler mode,flowtempdesired,m,D1C,,°C,temperature,hwctempdesired,m,D1C,,°C,temperature,hwcflowtempdesired,m,UCH,,°C,temperature,,m,IGN:1,,,,disablehc,m,BI0:1,,,,disablehwctapping,m,BI1:1,,,,disablehwcload,m,BI2:1,,,,,m,IGN:1,,,,remoteControlHcPump,m,BI0:1,,,,releaseBackup,m,BI1:1,,,,releaseCooling,m,BI2:1,,,

pi@raspberrypi:~ $ ebusctl i
version: ebusd 3.2.v3.2
signal: acquired
symbol rate: 23
max symbol rate: 114
min arbitration micros: 2083
max arbitration micros: 3338
min symbol latency: 4
max symbol latency: 6
reconnects: 0
masters: 3
messages: 212
conditional: 3
poll: 0
update: 9
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0316;HW=0903", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=05000;SW=0113;HW=9601"
address 31: master #8, ebusd
address 36: slave #8, ebusd

pi@raspberrypi:~ $ ebusctl scan result
08;Vaillant;BAI00;0316;0903;21;17;06;0010015325;3100;005007;N3
15;Vaillant;05000;0113;9601;21;17;31;0020018266;0082;006956;N9
```

Can someone spot what I'm doing wrong ? I've already spent a ton of time studying useless docs and surfing internet but with no success :(

john30, 2018-12-16:
SetMode is defined as update write message, i.e. you can't write it from ebusd as your controller will override it automatically

john30, 2018-12-16:
depends on your controller, usually you set the heat curve and the controller will do the rest

semenoh, 2018-12-16:
The controller is very dumb. It has no curves and modes. Only one knob to set a temperature.
`bai Status02 = on;60;75.0;70;65.0` here you can see that it keeps boiler somewhere between 70 and 75 degrees. So temperature in the house changes rapidly. 
That's why I decided to replace it with something smarter with ability to connect it with more stuff in the future (for instance smart radiator valves, conditioning system etc).
When I set a temperature on the boiler to 45-50 it feels really comfortable in house. But it changes with the outside temperature. So I have to go to the boiler and change it again. I'd like to delegate it to some script. But when I change temperature on the boiler ebusd does not show me any commands. So I decided to make my script to pretend it is a controller. But it turned out these messages are read only. So the question is how can I send the same messages instead of controller in order to control a temperature.

andig, 2018-12-17:
@semenoh you can use SetMode to write FlowTempDesired. See https://github.com/john30/ebusd/issues/214 for my experience with custom message definition to write SetMode myself.

john30, 2018-12-17:
if you're going to replace your controller with a script, then you'd have to make sure that the controller does not intervene the changes you made with your scripting. that would mean switching it off, i.e. disconnect it from the bus. not sure if you really want to do that.

semenoh, 2018-12-17:
> > The controller is very dumb. It has no curves and modes. Only one knob to set a temperature.
> 
> Probably we all bought our Vaillant boilers wanting to do the same thing as you want to do. Unfortunately it's not that simple - ebusd is reverse engineered (at least in some degree) opensource project which is still being in development. On the top of that, while the ebus protocol is open (?), many messages and the way ebusd devices interpret them are still a mystery.
> Dont give up, tinker with it and hopefully you will find out a way of doing what you want to do. Once this happens, please dont forget to share your findings with the community :)

You are totally right. Thanks for your support. I do realize that everything here is reverse-engineered and a bit of mystery. And despite a lot of things were demystified by the community still remain a mystery for me. Unfortunately there is not single place for knowledge sharing. Everyone should gather information bit by bit from different sources. Something is on blog posts, something here in issues, something in wiki etc. I will share my experience, not sure when ))

semenoh, 2018-12-17:
> if you're going to replace your controller with a script, then you'd have to make sure that the controller does not intervene the changes you made with your scripting. that would mean switching it off, i.e. disconnect it from the bus. not sure if you really want to do that.

Unfortunately for me you are working with MQTT. I have to rebuild all my setup with different build of `ebusd`. Also you mentioned that you `have a custom definition SetModeOverride that is writable`. Could you please elaborate a little bit on this. Probably I will be able to map what you are doing with MQTT to TCP.

semenoh, 2018-12-17:
> if you're going to replace your controller with a script, then you'd have to make sure that the controller does not intervene the changes you made with your scripting. that would mean switching it off, i.e. disconnect it from the bus. not sure if you really want to do that.

In fact this controller make things only worse. So it is switched off all the time except cases when I'm trying to intercept messages. The only thing I'm worried about is that my script can miss something important and boiler will malfunction.
BTW I've been thinking of sending binary data, but I need to intercept it first. Is it possible to listen to binary messages? (running `ebusd` in foreground mode prints byte by line and most of them are 0xAA so it's really difficult to read)

john30, 2018-12-17:
just switch ebusd to print complete messages rather than every byte by using "--lograwdata" instead of "--lograwdata=bytes"

semenoh, 2018-12-20:
@john30 is it possible that ebusd see newer version in repo and override my one with it? Or it read config from different place ?

Erbit, 2019-01-10:
Hi.
I think that FlowTempDesired is set by the boiler controller and can not be changed using ebusctl. FlowTempDesired depends on: Hc1HeatCurve, current changes in ReturnTemp and can not exceed FlowsetHcMax.

I deduced this from the observation of my own charts.

The problem is that the maximum HC temperature I have set in the controller at 50 and it works. FlowTempDesired does not exceed 50. Unfortunately, "ebusctl read -c bai FlowsetHcMax" returns me 75. The change of FlowsetHcMax value by ebustl is without errors but the re-reading of FlowsetHcMax still shows 75. I think this is some error in .inc files for our boilers, which may be too new and do not have a properly prepared configuration in .inc files.

I think you should only change FlowsetHcMax, but you can not do it correctly. I can not too.

Erbit, 2019-01-10:
> bai FlowsetHcMax - this is for the heating circuit
> bai FlowsetHwcMax - this is for the hot water circuit
> 
> i dont think there is a minimum value you can set. doesnt make sense to me...
> i've set those values in my heater and never touched them again

I agree with Dinth.

HubertJH, 2019-05-06:
Just stumbled across this discussion while searching for information to achieve the same thing - setting FlowTempDesired (on a new boiler Vaillant VCI installed just recently. I have done the equivalent on the previous boiler using the 7-8-9 interface).
The new boiler has no controller connected and FlowTempDesired can be set manually from the UI on the boiler itself. In so far 
> FlowTempDesired depends on: Hc1HeatCurve, current changes in ReturnTemp and can not exceed FlowsetHcMax.

does not apply in my case.
I have started playing with some of the options described here with no success so far. Guess I have to digest SetMode first...
Did you made any progress in the meantime?

andig, 2019-05-07:
> The new boiler has no controller connected and FlowTempDesired can be set manually from the UI on the boiler itself.

Unfortunately, in my experiments, the boiler-internal changes are not reflected on the bus.

> Guess I have to digest SetMode first...

Imho `SetMode` is the way to go. It is what the VRC700 issues to the BAI. Since it's not writeable define your own `SetModeOverride` command and use that.

HubertJH, 2019-05-08:
Thanks for your response, andig.
> Unfortunately, in my experiments, the boiler-internal changes are not reflected on the bus.

Not quite sure what you mean with "reflected" - changes of FlowTempDesired on the unit can be read from the eBus. Anyway, will give SetMode a try.

Erbit, 2019-05-08:
Hi HubertJH .
The graph shows the FlowTempDesired changes.

At the beginning of January 2019, FlowsetHcMax was set at 55 °C. On January 7, I changed FlowsetHcMax to 50 °C. As you can see, FlowTempDesired does not exceed FlowsetHcMax and is often below FlowsetHcMax, especially now when the outside temperature varies from 3 °C to 17 °C.

**I think that FlowTempDesired is set by the internal program of the boiler and the user can only limit FlowTempDesired via FlowsetHcMax.**

The FlowsetHcMax change can not be seen immediately in FlowTempDesired. FlowTempDesired will not change until the next measurement cycle. It can be within 30 minutes.
[image]

Erbit, 2019-05-08:
In the next graph I added the outside temperature. See that FlowTempDesired decreases as the outside temperature increases. I do not change FlowTempDesired, it does the boiler program. An external temperature sensor is connected to the boiler.

[image]

cbspidi, 2019-11-11:
Hi 

Just stumbled across this discussion, did anyone got any success in steering flow temperature on valiant boilers, i have Uk version of 618(system boiler) without any additional components.
Boiler is directly attached to ebus interface.
I get all readings correct but im suffering from what was previously mentioned in the discussion which is no change in temperature when i write to FlowsetHcMax, temperature does not change(not in parameters reported by ebus nor on radiators)
Not sure how to use SetMode mentioned by andig, does anyone have a place where that is explained?

andig, 2019-11-11:
> Not sure how to use SetMode mentioned by andig, does anyone have a place where that is explained?

The related discussion was in https://github.com/john30/ebusd/issues/179.

cbspidi, 2019-11-11:
Hey andig 
I dont have SeMode in my list of write options, where do i go from here ?

andig, 2019-11-11:
> I don't have SeMode in my list of write options, where do i go from here ?

You'll have to follow https://github.com/john30/ebusd/issues/179#issuecomment-434628658.

john30, 2021-10-06:
closed due to inactivity