john30/ebusd #191: Vaillant ecoTec Plus 825

Dinth, 2018-05-21:
Hi. Im struggling with figuring out how to use ebusd together with my combi boiler, as most values which normally should be writable are read only (for example HeatingSwitch).
Im using ebusd 3.2 and latest config (online). Access level is set to *.

Even values which are writable cannot really be modified.
```
localhost: r FlowsetHcMax
75.00

localhost: w -c bai FlowsetHcMax "65.00"
done

localhost: r -f FlowsetHcMax
75.00
```

Here's my boiler's code:
```
localhost: scan result
08;Vaillant;BAI00;0204;9602;21;17;17;0010018353;1300;257134;N7
```

And output of ebusctl find -V: 
```
localhost: f -V 
bai AccessoriesOne = =circulationpump [Funktion des Zubehörrelais 1] [ZZ=08, lastup=2018-05-21 21:30:03, active read]
bai AccessoriesTwo = =extheatingpump [Funktion des Zubehörrelais 2] [ZZ=08, lastup=2018-05-21 21:30:04, active read]
bai AntiCondensValue = =4 kW / % [Anhebung der minimalen Leistung] [ZZ=08, lastup=2018-05-21 21:30:05, active read]
bai averageIgnitiontime = =1.8 s [Mittlere Zündzeit] [ZZ=08, lastup=2018-05-21 21:30:06, active read]
bai BlockTimeHcMax = minutes0=20 min [Max. Brennersperrzeit bei einem Vorlaufsollwert von 20°C] [ZZ=08, lastup=2018-05-21 21:30:07, active read]
bai BoilerType = =2 [Boiler typ of the bmu] [ZZ=08, lastup=2018-05-21 21:30:08, active read]
bai ChangesDSN = =0 [Anzahl der DSN (Gerätekennung) Änderungen] [ZZ=08, lastup=2018-05-21 21:30:09, active read]
bai CirPump = onoff=off [Status Zirkulationspumpe (über ein externes Modul ansteuerbar)] [ZZ=08, lastup=2018-05-21 21:30:10, active read]
bai CodingResistor = =08 f7 00 [Therme Identifikation Widerstand] [ZZ=08, lastup=2018-05-21 21:30:11, active read]
bai CounterStartattempts1 = temp0=4 °C [Anzahl der erfolglosen Zündversuche (im 1. Versuch)] [ZZ=08, lastup=2018-05-21 21:30:13, active read]
bai CounterStartattempts2 = temp0=4 °C [Anzahl der erfolglosen Zündversuche (im 2. Versuch)] [ZZ=08, lastup=2018-05-21 21:30:14, active read]
bai CounterStartAttempts3 = temp0=3 °C [Anzahl der erfolglosen Zündversuche (im 3. Versuch)] [ZZ=08, lastup=2018-05-21 21:30:15, active read]
bai CounterStartAttempts4 = temp0=4 °C [Anzahl der erfolglosen Zündversuche (im 4. Versuch)] [ZZ=08, lastup=2018-05-21 21:30:16, active read]
bai currenterror = error=- [Fehlernummer];error=- [Fehlernummer];error=- [Fehlernummer];error=- [Fehlernummer];error=- [Fehlernummer] [ZZ=08, lastup=2018-05-21 21:34:01, active read]
bai DateTime = dcfstate=nosignal [DCF Empfängerstatus];btime=-:-:- [Uhrzeit];bdate=-.-.- [Datum];temp2=- °C [Temperatur] [ZZ=08, lastup=2018-05-21 21:30:18, active read]
bai dcfState = dcfstate=nosignal [DCF Status] [ZZ=08, lastup=2018-05-21 21:30:19, active read]
bai DCFTimeDate = =06:41:37 [DCF Time / date stamp struct { uchar sec; uchar min; uchar std; uchar tag; uchar mon; uchar wtg; uchar jah; } Tdcf77_time;];=24.02.2000 [ZZ=08, lastup=2018-05-21 21:30:20, active read]
bai DCRoomthermostat = onoff=on [Wärmeanforderung vom externen Regler (Klemme 3-4)] [ZZ=08, lastup=2018-05-21 21:30:21, active read]
bai DeactivationsIFC = =10 [Anzahl der Zündfehler (nicht erfolgreiche Züdung im letzten Versuch oder fehlerhaftes Flammensignal)] [ZZ=08, lastup=2018-05-21 21:30:22, active read]
bai DeactivationsTemplimiter = =0 [Anzahl der Abschaltungen durch den Sicherheitstemperaturbgrenzers] [ZZ=08, lastup=2018-05-21 21:30:24, active read]
bai DeltaFlowReturnMax = temp=22.62 °C [Wartungsdaten] [ZZ=08, lastup=2018-05-21 21:30:25, active read]
bai DisplayMode = =2 [Display mode of the aplliance] [ZZ=08, lastup=2018-05-21 21:30:26, active read]
bai DSN = =13006 [DSN: Device Specific number] [ZZ=08, lastup=2018-05-21 21:30:27, active read]
bai DSNOffset = =6 [Gerätekennung (DSN)] [ZZ=08, lastup=2018-05-21 21:30:28, active read]
bai DSNStart = =13000 [DSN Startadresse] [ZZ=08, lastup=2018-05-21 21:30:29, active read]
bai EbusSourceOn = onoff=on [Aktivierung der eBUS Speisung] [ZZ=08, lastup=2018-05-21 21:30:30, active read]
bai EbusVoltage = onoff=on [Rückmeldung der eBUS Spannung] [ZZ=08, lastup=2018-05-21 21:30:31, active read]
bai errorhistory = no data stored [ZZ=08, active read]
bai ExternalFaultmessage = onoff=off [Signal für die externe Störmeldeeinrichtung] [ZZ=08, lastup=2018-05-21 21:30:33, active read]
bai externalHwcSwitch = onoff=off [Speicheranforderung eines externen Speichers über den Speicherkontakt] [ZZ=08, lastup=2018-05-21 21:30:35, active read]
bai ExternGasvalve = onoff=240 [Externes Magnetventil] [ZZ=08, lastup=2018-05-21 21:30:36, active read]
bai ExtFlowTempDesiredMin = temp=40.00 °C [minimum out of Kl.7 and eBus flow setpoint] [ZZ=08, lastup=2018-05-21 21:30:37, active read]
bai ExtStorageModulCon = yesno=no [Externes Speichermodul (VR65) angeschlosssen] [ZZ=08, lastup=2018-05-21 21:30:38, active read]
bai extWP = onoff=on [Externe Heizungspumpe] [ZZ=08, lastup=2018-05-21 21:30:39, active read]
bai FanHours = hoursum2=56 h [Betriebsstunden des Lüfters] [ZZ=08, lastup=2018-05-21 21:30:40, active read]
bai FanMaxSpeedOperation = =6000 1/min [Lüfter Maximaldrehzahl] [ZZ=08, lastup=2018-05-21 21:30:41, active read]
bai FanMinSpeedOperation = =750 1/min [Lüfter Minimaldrehzahl] [ZZ=08, lastup=2018-05-21 21:30:42, active read]
bai FanPWMSum = =0 [Predictive Maintenance data for the fan damage recognition] [ZZ=08, lastup=2018-05-21 21:30:43, active read]
bai FanPWMTest = =0 [Predictive Maintenance data for the fan damage recognition] [ZZ=08, lastup=2018-05-21 21:30:45, active read]
bai FanSpeed = =0 1/min [Aktuelle Lüfterdrehzahl] [ZZ=08, lastup=2018-05-21 21:30:46, active read]
bai FanSpeedOffsetMax = =-290 1/min [Offset max. Drehzahl] [ZZ=08, lastup=2018-05-21 21:30:47, active read]
bai FanSpeedOffsetMin = =420 1/min [Offset min. Drehzahl] [ZZ=08, lastup=2018-05-21 21:30:48, active read]
bai FanStarts = cntstarts2=0 [Anzahl der Lüfterschaltspiele] [ZZ=08, lastup=2018-05-21 21:30:49, active read]
bai Flame = =off [Flammensignal] [ZZ=08, lastup=2018-05-21 21:30:50, active read]
bai FlameSensingASIC = =579 [Ioni/ADC Wert vom Flammenwächter] [ZZ=08, lastup=2018-05-21 21:30:51, active read]
bai FloorHeatingContact = onoff=off [Eingang Anlegethermostat] [ZZ=08, lastup=2018-05-21 21:30:52, active read]
[... 101 lines cut ...]
bai WarmstartDemand = yesno=yes [Warmstartaktivierung] [ZZ=08, lastup=2018-05-21 21:32:45, active read]
bai WaterHcFlowMax = =0 [Wartungsdaten] [ZZ=08, lastup=2018-05-21 21:32:46, active read]
bai WaterPressure = press=1.922 bar [Wasserdruck];sensor=ok [Fühlerstatus] [ZZ=08, lastup=2018-05-21 21:34:00, active read]
bai WaterpressureBranchControlOff = onoff=off [Überwachung der Druckänderung beim Schalten der Pumpe kann hiermit aktiviert oder deaktiviert werden] [ZZ=08, lastup=2018-05-21 21:32:48, active read]
bai WaterpressureMeasureCounter = =13 [Wartungsdaten] [ZZ=08, lastup=2018-05-21 21:32:49, active read]
bai WaterpressureVariantSum = pressm2=7979 mbar [Wartungsdaten] [ZZ=08, lastup=2018-05-21 21:32:51, active read]
bai WP = onoff=on [Interne Heizungspumpe] [ZZ=08, lastup=2018-05-21 21:32:52, active read]
bai WPPostrunTime = minutes0=5 min [Wasserpumpennachlaufzeit nach Heizbetrieb] [ZZ=08, lastup=2018-05-21 21:32:53, active read]
bai WPSecondStage = onoff=off [Second stage of the pump activated] [ZZ=08, lastup=2018-05-21 21:32:54, active read]
broadcast datetime = no data stored [ZZ=fe, passive read]
broadcast error = no data stored [ZZ=fe, passive read]
broadcast hwcStatus = no data stored [ZZ=fe, passive read]
broadcast id = no data stored [ZZ=fe, passive read]
broadcast id = no data stored [ZZ=any, active read]
broadcast load = no data stored [ZZ=fe, passive read]
broadcast outsidetemp = no data stored [ZZ=fe, passive read]
broadcast signoflife = no data stored [ZZ=fe, passive read]
broadcast vdatetime = no data stored [ZZ=fe, passive read]
general valuerange = no data stored [ZZ=any, active read]
memory eeprom = no data stored [ZZ=any, active read]
memory ram = no data stored [ZZ=any, active read]
scan id = no data stored [ZZ=any, active read]
scan.08  = MF=Vaillant;ID=BAI00;SW=0204;HW=9602 [ZZ=08, lastup=2018-05-21 21:13:20, active read]
scan.08 id = prefix=21;year=17;week=17;product=0010018353;supplier=1300;counter=257134;suffix=N7 [ZZ=08, lastup=2018-05-21 21:33:07, active read]
```

semenoh, 2018-05-22:
Same on my turboTEC plus VU 202

john30, 2018-05-25:
did you already check the writable messages with "find -w"?

Dinth, 2018-05-25:
Hi John30. Indeed, it may be a configuration problem, but i believe there are two issues here:
1. I cannot write to a variable which is writable (FlowsetHcMax)
2. Variables which should be writable, are not (HeatingSwitch)

Here's the output of find -w. Actually i found a really weird thing here. If i do "find -w" im getting "no data stored" on all variables, but if i just do "find" ebusctl shows the data stored in a variable.

```
localhost: find -w
bai AccessoriesOne = no data stored
bai AccessoriesTwo = no data stored
bai AntiCondensValue = no data stored
bai BlockTimeHcMax = no data stored
bai clearerrorhistory = no data stored
bai DSNOffset = no data stored
bai FanSpeedOffsetMax = no data stored
bai FanSpeedOffsetMin = no data stored
bai FlowsetHcMax = no data stored
bai HcPumpMode = no data stored
bai HoursTillService = no data stored
bai HwcTempMax = no data stored
bai OptionalRelais = no data stored
bai PartloadHcKW = no data stored
bai PumpPowerDesired = no data stored
bai ReturnRegulation = no data stored
bai SetFactoryValues = no data stored
bai StorageLoadTimeMax = no data stored
bai WPPostrunTime = no data stored
broadcast id = no data stored
broadcast queryexistence = no data stored
memory eeprom = no data stored
memory ram = no data stored

```

```
localhost: find 
bai AccessoriesOne = circulationpump
bai AccessoriesTwo = extheatingpump
(...)
```

john30, 2018-05-26:
> I cannot write to a variable which is writable (FlowsetHcMax)

it seems your bai just ignores the write. can't do anything about that I'm afraid. Do you have a controller as well? Then you could try setting the parameter there and watch the bus for corresponding messages in order to correct/adjust the message definition.

> Variables which should be writable, are not (HeatingSwitch)

the bai definitions were extracted from a manufacturer DB and that's how it is defined there. Since I don't own all of the devices I can't validate them, so if you're sure the message is writeable, then please issue a corresponding pull request on ebusd-configuration.

Dinth, 2018-05-26:
> it seems your bai just ignores the write. can't do anything about that I'm afraid. Do you have a controller as well? 
Unfortunately i dont, ive been hoping that i will be able to control the boiler using RPi and ebus. Im not a gas enginner, is it just a matter of getting a VCR700 for debugging or i would also need to install a receiver in the boiler? 

>Since I don't own all of the devices I can't validate them, so if you're sure the message is writeable, then please issue a corresponding pull request on ebusd-configuration.
Fully understand. Is there a way i could confirm if the value is writable before issuing a pull request, or help with getting a better support for my boiler in any other way?

john30, 2018-06-02:
> Unfortunately i dont, ive been hoping that i will be able to control the boiler usingjust an RPi and ebus. Im not a gas enginner, is it just a matter of getting a VCR700 for debugging or i would also need to install a wireless receiver in the boiler?

it is for finding out what messages the boiler is willing to answer to. anyway, before spending money, you should probably have a look at the "standard" messages that are understood by most of the heating devices, see here for heating [hcmode.inc](https://github.com/john30/ebusd-configuration/blob/master/ebusd-2.1.x/de/vaillant/hcmode.inc) and here for hot water unit [hwcmode.inc](https://github.com/john30/ebusd-configuration/blob/master/ebusd-2.1.x/de/vaillant/hwcmode.inc).

> Is there a way i could confirm if the value is writable before issuing a pull request, or help in any other way to make ebusd better if im not a coder ?

well just set it to writable, write the value and check that the desired effect becomes real.

Dinth, 2018-06-02:
> well just set it to writable, write the value and check that the desired effect becomes real.
Indeed, i edited 'bai.0010015600.inc' and replaced 
`r,,HeatingSwitch,d.23 (...)
`with 
`r;wi,,HeatingSwitch,d23 (...)`
And now changing the HeatingSwitch value actually switches heating on and off. 
Im not sure if i should make a pull request - if this config file is also used by different models then this change may break things for other people?

Dinth, 2018-06-06:
I created pull request #101 at ebusd-configuration repository which makes HeatingSwitch and HwcSwitch writable. Also i translated the file into english.
When i will have some spare time i will also try to check other variables if they can be made r/w

john30, 2018-09-22:
closed in favour of https://github.com/john30/ebusd/issues/101

closed 2018-09-22 by john30