john30/ebusd #192: "No data stored" when doing running -w, but data shows when running an ordinary find.

Dinth, 2018-05-25:
I already mentioned this issue in another thread.

When i run "find -w" in ebusctl i get "No data stored" in all variables:

```
localhost: find -w
bai AccessoriesOne = no data stored
bai AccessoriesTwo = no data stored
bai AntiCondensValue = no data stored
bai BlockTimeHcMax = no data stored
bai clearerrorhistory = no data stored
bai DSNOffset = no data stored
bai FanSpeedOffsetMax = no data stored
bai FanSpeedOffsetMin = no data stored
bai FlowsetHcMax = no data stored
bai HcPumpMode = no data stored
bai HoursTillService = no data stored
bai HwcTempMax = no data stored
bai OptionalRelais = no data stored
bai PartloadHcKW = no data stored
bai PumpPowerDesired = no data stored
bai ReturnRegulation = no data stored
bai SetFactoryValues = no data stored
bai StorageLoadTimeMax = no data stored
bai WPPostrunTime = no data stored
broadcast id = no data stored
broadcast queryexistence = no data stored
memory eeprom = no data stored
memory ram = no data stored
```
but when i run find without '-v' option i can see that most of those variables actually store data:

```
localhost: find 
bai AccessoriesOne = circulationpump
bai AccessoriesTwo = extheatingpump
bai AntiCondensValue = 4
bai BlockTimeHcMax = 20
(...)
```

john30, 2018-05-26:
did you try "find -p" as well?

Dinth, 2018-05-26:
find -p also gives 'no data stored' on all variables.

```
ebusctl find -p
bai SetMode = no data stored
broadcast datetime = no data stored
broadcast error = no data stored
broadcast hwcStatus = no data stored
broadcast id = no data stored
broadcast load = no data stored
broadcast outsidetemp = no data stored
broadcast signoflife = no data stored
broadcast vdatetime = no data stored
```

Ohh, actually when i done 'find -a' i found out that some variables are just doubled - one contains data and one does not. 

```
ebusctl find -a
bai AccessoriesOne = circulationpump
bai AccessoriesOne = no data stored
bai AccessoriesTwo = extheatingpump
bai AccessoriesTwo = no data stored
bai AntiCondensValue = 4
bai AntiCondensValue = no data stored
bai APCLegioProtection = no data stored (message not available due to condition)
bai APCLegioProtection = no data stored (message not available due to condition)
bai averageIgnitiontime = 1.8
bai BlockTimeHcMax = 20
bai BlockTimeHcMax = no data stored

```

john30, 2018-06-02:
this is due to the circumstance that the same information is sometimes transferred in different ways: as a "read", a "write", or an "update". Each of those types have their own definition since the message format differs for each of them, but the names are kept the same since the translated data is the same. I know this is a little bit tricky.

> That's weird but maybe this can actually explain why i cannot write to some variables (my other issue)? If one of them is readonly, other one is writable, and ebusctl write tries to send data into the read only one?

No, if you issue a write, then only a message defined as writable will be used by ebusd.

closed 2018-12-25 by john30