john30/ebusd #194: Weishaupt WRSol 2.1

sagittariusgit, 2018-05-31:
hello,

I am having a problem with the WRSol 2.1 device, which I am not able to solve - some issues here relate to the device, but I have not seen a complete solution.
- connection through the esara coupler to a raspberry pi under jessie
- reading on the bus appears to work. I have tried various address settings on the WRSOl with various data rates and changed the poti settings. I can read a message, which the WRSOl issues after boot and every  5 min or so, it only differs in byte 1 (sending address) and last byte (CRC) as it should, the byte count and content is consistent for various speeds. 

2018-05-31 16:39:49.056 [bus notice] <01fe
2018-05-31 16:39:49.445 [bus notice] <fe010b
2018-05-31 16:39:49.836 [bus notice] <455336
2018-05-31 16:39:50.228 [bus notice] <3520a0
2018-05-31 16:39:50.619 [bus notice] <45204f
2018-05-31 16:39:50.636 [bus notice] <4bff38

- ebusctl scan full does not yield any other device than the 31/36 adresses by the pi
- I had tried a self-designed interface board, which basically gave the same behaviour.
- when I enter a scan for the originating adress of the above message the adress (01)  is either unknown or an error produced.

version: ebusd 3.2.v3.2
signal: acquired
symbol rate: 12
max symbol rate: 17
min arbitration micros: 1036
max arbitration micros: 1233
min symbol latency: 5
max symbol latency: 7
reconnects: 0
masters: 1
messages: 11
conditional: 0
poll: 0
update: 4
address 31: master #8, ebusd (answering)
address 36: slave #8, ebusd (answering)

and

ebusctl scan 1
ERR: invalid address

and

ebusctl w -h 01070400 

just produces the help message, so an error message of sorts.

it appears that i) I am doing sthg wrong, but ii) also that the wrsol is not really responsive under the address .

has anyone solved the ebus communication for such device ?

thx in advance

john30, 2018-06-02:
> I have tried various address settings on the WRSOl with various data rates and changed the poti settings. I can read a message, which the WRSOl issues after boot and every 5 min or so, it only [...]

what do you mean with "various speeds"? that doesn't make sense at all since the eBUS is fixed to 2400 Baud.

> 2018-05-31 16:39:49.056 [bus notice] <01fe
2018-05-31 16:39:49.445 [bus notice] <fe010b
2018-05-31 16:39:49.836 [bus notice] <455336
2018-05-31 16:39:50.228 [bus notice] <3520a0
2018-05-31 16:39:50.619 [bus notice] <45204f
2018-05-31 16:39:50.636 [bus notice] <4bff38

this is more or less nonsense. you need to adjust the poti very well first before you'll get anything meaningful from the bus! Stick to the howto for that: https://github.com/john30/ebusd/wiki/6.-Hardware#adjusting-the-potentiometer

> ebusctl scan 1

you need to specify the address as two digits hex, i.e. "01" instead of "1"

> ebusctl w -h 01070400

this is not a valid command. For sending arbitrary hex messages, you need to add "--enablehex" to ebusd daemon start parameters and then issue "ebusctl hex 01070400".

sagittariusgit, 2018-06-02:
thank you

1. speed

obviously the bus works at 2400 baud fixed. however, the wrsol's settings permit the selection of various bus adresses, each of which is associated with different levels of bus activity from the side of the wrsol. The 0xAA sync bytes come slower or faster depending on the wrsol's  selected bus adress. that is what I meant.

2. poti settings

I actually followed your how-to in my trials. consistent 0xaa appear at 2400 baud setting both with your deaemon and a terminal emulator. I am approx. in the middle of the poti range between loss of signal on both ends.

3. status broadcast message

the message is not entirely nonsense it seems, something meaningful is being read. I even ran it through a crc check and the numbers correspond for various sync speeds. it is a broadcast to all and has some wrsol version numbers inside. so if the message is crc checked correctly and the first byte (sender adress) is consistently changed along with the bus adress setting on the wrsol and the message text is one and the same .. ? seems to be an indication of reasobably correct settings.

4. i also had looked into https://github.com/john30/ebusd-configuration/issues/9

which basically shows the same behaviour as mine here. also with a status broadcast message like the above coming through regularly. also here the wrsol apparently does not answer.

as sending is not a settings issue on the sending side (just a line being pulled down) I suspect a settings issue on the wrsol side or that it is listening on a different adress than the sending one. could that be ?

john30, 2018-06-13:
the snippets of logging you pasted indicate that the transfer of that message is interrupted by 0xAA (that's why these parts appear on separate lines in ebusd log). Since 0xAA is not a symbol that will ever occur in a message on the eBUS, the poti is not adjusted well enough yet.
BTW, sending to the master address 0x01 is usually not what you do when you want to read something from the device. Use the devices slave address 0x06 instead. That's also true for the ID request, so you should rather use `ebusctl hex 06070400` or `ebusctl scan 06`.

john30, 2018-06-30:
yes, there seem to be severe timing issues on your bus, e.g.:

> 2018-06-25 15:14:22.895 [bus notice] <aa
2018-06-25 15:14:22.900 [bus notice] <10
2018-06-25 15:14:22.904 [bus notice] <fe
2018-06-25 15:14:22.955 [bus notice] <aa

the last aa is sent by the auto-SYN generator due to 50ms silence on the bus. But afterwards it seems the master 0x10 just continues to send its original message, i.e. 10 fe fe 01 0b 45 53 36 35 20 20 45 20 4f 4b ff 64.
Actually this message seems to be valid since the CRC is correct, so you'd have to find out why the SYN bytes are thrown in or why the master 0x10 (usually the controller) is sending with such a high delay between some of the bytes in a message.

sagittariusgit, 2018-06-30:
Thank you,

exactly that was my thinking. So I have to understand what generates the autosync on the WRSol side - or what causes the delays in the message byte generation, which cause the autosync.
my first idea would be to try swithching it off on the wrsol side and letting ebusd do the autosync. what is the algorithm for the autosync you have implented ? no traffic after n msec  triggers a 0xaa ?

john30, 2018-07-01:
didn't you mention that wrsol has different settings that allow different times between the autogen SYN symbols? I'd got for that option, because then the system will run even without your ebusd instance.
anyway, the algorithm in ebusd is implemented as required by the specification.

sagittariusgit, 2018-07-01:
it appears to be the same issue:

2018-07-01 20:09:57.764 [bus notice] <aa
2018-07-01 20:09:57.768 [bus notice] <f0
2018-07-01 20:09:57.772 [bus notice] <fe
2018-07-01 20:09:57.777 [bus notice] <fe
2018-07-01 20:09:57.781 [bus notice] <01
2018-07-01 20:09:57.784 [bus notice] <0b
2018-07-01 20:09:58.084 [bus notice] <aa
2018-07-01 20:09:58.090 [bus notice] <45
2018-07-01 20:09:58.093 [bus notice] <53
2018-07-01 20:09:58.097 [bus notice] <36
2018-07-01 20:09:58.395 [bus notice] <aa
2018-07-01 20:09:58.399 [bus notice] <35
2018-07-01 20:09:58.403 [bus notice] <20
2018-07-01 20:09:58.408 [bus notice] <20
2018-07-01 20:09:58.707 [bus notice] <aa
2018-07-01 20:09:58.711 [bus notice] <45
2018-07-01 20:09:58.715 [bus notice] <20
2018-07-01 20:09:58.719 [bus notice] <4f
2018-07-01 20:09:59.017 [bus notice] <aa
2018-07-01 20:09:59.021 [bus notice] <4b
2018-07-01 20:09:59.026 [bus notice] <ff
2018-07-01 20:09:59.029 [bus notice] <03
2018-07-01 20:09

at a lower "speed" I still read 4 ms between valid msg byes and then 300msec pauses, which are filled by 0xaa. I have no idea why these pauses occur.

closed 2018-12-25 by john30