john30/ebusd #196: Vaillant calorMATIC VRT350 RoomTemp invalid position in decode

sagittarius0815, 2018-08-05:
I'm new with ebusd, but I think I've got the daemon running right.

Using configuration
```
ebusctl info
version: ebusd 3.2.v3.2-5-g671c8d2
update check: revision v3.2-2-g1737636 available
signal: acquired
symbol rate: 23
max symbol rate: 112
min arbitration micros: 1536
max arbitration micros: 2627
min symbol latency: 5
max symbol latency: 11
reconnects: 0
masters: 3
messages: 410
conditional: 7
poll: 0
update: 9
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0104;HW=7803", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=35000;SW=0114;HW=7102", loaded "vaillant/15.350.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
```

```
scan result
08;Vaillant;BAI00;0104;7803;21;18;15;0010021934;0001;006301;N9
15;Vaillant;35000;0114;7102;21;18;12;0020124472;0082;012268;N8
```
```
find -f
r,350,RoomTemp,Aktuelle Raumtemperatur,,15,b509,0d0f00,temp,s,D2C,,°C,Temperatur,sensor,s,UCH,0=ok;85=circuit;170=cutoff,,Fühlerstatus
```
But I cannot get values for "RoomTemp"
```
ebusctl r -V RoomTemp
ERR: invalid position in decode
```
What can I do to fix this problem?

john30, 2018-08-05:
probably not at all since it seems your device just doesn't support that particular message.
For the future: please post configuration issues on [ebusd-configuration](https://github.com/john30/ebusd-configuration/issues) rather than here.

closed 2018-08-05 by john30

sagittarius0815, 2018-08-05:
The device supports the RoomTemp message, but it seems so, that controller vtr 350 in this software version supports same messages as the VTR 370 controller.
I replaced content of 15.350.csv with content from 15.370.csv. Now it works for me.
So it's not a ebusd issue, but a configuration issue.
Sorry I'm worried you, but sometimes it helps to talk.
Thank you.

hansinator, 2024-12-03:
Hey I'm having the same problem, but there is no 15.370.csv @sagittarius0815
How did you get to that solution?

Koky05, 2024-12-04:
@hansinator you need to use local copy of configuration files and there are csv for 370:
[All Vaillant configurations](https://github.com/john30/ebusd-configuration/tree/master/ebusd-2.1.x/en/vaillant)