john30/ebusd #198: Vaillant VR32 support?

Nuc67UA, 2018-09-09:
Hello,
Is this module supported? I have the following setup: VR630 and two Turbotec boilers cascaded. The first one is connected to the bus directly, the second vis VR32 module. The first boiler is visible, the second seems not. Is there any simple solution around that?
Everything else is just fine!

version: ebusd 3.2.v3.2
update check: revision v3.2-2-g1737636 available, broadcast.csv: newer version available, vaillant/15.ui.csv: newer version available, vaillant/25.vr630.hwc.csv: newer version available, vaillant/bai.0010006101.inc: newer version available, vaillant/broadcast.csv: newer version available, vaillant/errors.inc: newer version available, vaillant/hcmode.inc: newer version available
signal: acquired
symbol rate: 25
max symbol rate: 104
min arbitration micros: 1004
max arbitration micros: 8590
min symbol latency: 4
max symbol latency: 9
reconnects: 0
masters: 3
messages: 745
conditional: 12
poll: 0
update: 10
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0518;HW=7401", loaded "vaillant/bai.0010006101.inc" ([PROD='']), "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=UI   ;SW=0321;HW=6201", loaded "vaillant/15.ui.csv"
address 23: slave, scanned "MF=Vaillant;ID=VR630;SW=0306;HW=6301", loaded "vaillant/23.vr630.cc.csv"
address 25: slave, scanned "MF=Vaillant;ID=VR630;SW=0306;HW=6301", loaded "vaillant/25.vr630.hwc.csv"
address 26: slave, scanned "MF=Vaillant;ID=VR630;SW=0306;HW=6301", loaded "vaillant/26.vr630.hc.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 50: slave, scanned "MF=Vaillant;ID=VR630;SW=0306;HW=6301", loaded "vaillant/50.vr630.mc.csv"
address 51: slave, scanned "MF=Vaillant;ID=VR630;SW=0306;HW=6301", loaded "vaillant/51.vr630.mc.3.csv"

Nuc67UA, 2018-09-12:
Thank you for you reply, I'll do that. BTW with --latency=100000 the VR32 gets visible.

closed 2018-09-12 by Nuc67UA

tlc76, 2021-01-26:
@Nuc67UA - I have exactly the same configuration as you (2 x Ecotec heaters in cascade and 1 x Calormatic 630).
However I would like to keep a single ebus line for all 3 devices in order to preserve the "cascade" function and full control via Calormatic.

I have some questions:
How did you configure your system? With 2 separate ebus lines (and lost cascade function) or with a single ebus line?
Is the VR32 in the 2nd heater still in use?
How did you make the wirings?

Thanks a lot for your answers!
Cristian