john30/ebusd #199: Starting on Ebusd

klona, 2018-09-17:
Hi John.
Thanks a lot for your project, all the work done, for shating that and support it. It's a huge work.

I found your project and decided to fully manage (under Jeedom) my boiler Saunier Duval ThemaPlus Condens F25 A (aka AWB ThermoElegance Advance 25-A or Bulex Thema CONDENS F25A ..)
I add an ebus to Opentherm board in the boiler ( link to spare parts , if this can help someone : http://www.klima-parts.nl/awb-opentherm-module-0020092426.html )

As first step, I build a device using infos from http://otgw.tclcode.com/ and it's working with the otmonitor from this site. 
I set up a Raspberry Pi 3, connecting through USB to my Opentherm gateway.
I get basic stuff working like starting heating but, as expected, it's a blackbox and I cant go further to manage all proprietary codes from Saunier-Duval.
08:17:40.498170  R80190000  Read-Data   Boiler water temperature: 0.00
08:17:40.616808  B40191F80  Read-Ack    Boiler water temperature: 31.50
08:17:41.524167  R10015A00  Write-Data  Control setpoint: 90.00
08:17:41.715804  BD0015A00  Write-Ack   Control setpoint: 90.00
08:17:42.550179  R80380000  Read-Data   DHW setpoint: 0.00
08:17:42.668690  B4038FFFF  Read-Ack    DHW setpoint: -0.00
08:17:43.576196  R80000100  Read-Data   Status: 00000001 00000000
08:17:43.694802  B40000100  Read-Ack    Status: 00000001 00000000
08:17:44.602316  R80000100  Read-Data   Status: 00000001 00000000
08:17:44.720933  B40000100  Read-Ack    Status: 00000001 00000000
08:17:45.626988  R00120000  Read-Data   CH water pressure: 0.00
08:17:45.746855  BC0120099  Read-Ack    CH water pressure: 0.60
08:17:46.652785  R80000100  Read-Data   Status: 00000001 00000000
08:17:46.772784  B40000100  Read-Ack    Status: 00000001 00000000
08:17:47.678903  R80190000  Read-Data   Boiler water temperature: 0.00
08:17:47.798786  B40191F80  Read-Ack    Boiler water temperature: 31.50
08:17:48.704907  R10015A00  Write-Data  Control setpoint: 90.00
08:17:48.823521  BD0015A00  Write-Ack   Control setpoint: 90.00
08:17:49.731020  R80000100  Read-Data   Status: 00000001 00000000
08:17:49.849627  B40000100  Read-Ack    Status: 00000001 00000000
08:17:50.757009  R00110000  Read-Data   Relative modulation level: 0.00
08:17:50.875628  BC0110000  Read-Ack    Relative modulation level: 0.00
08:17:51.783004  R80000100  Read-Data   Status: 00000001 00000000
08:17:51.901596  B40000100  Read-Ack    Status: 00000001 00000000
08:17:52.807829  R801C0000  Read-Data   Return water temperature: 0.00
08:17:52.926463  B401C2080  Read-Ack    Return water temperature: 32.50

Then , as hardware is working, I moved to Ebusd. After some readings and tries, I get it to the point I can see some raw data.

2018-09-17 08:06:47.180 <f4
2018-09-17 08:06:48.078 <42
2018-09-17 08:06:48.083 <21
2018-09-17 08:06:48.087 <f4
2018-09-17 08:06:48.197 <33
2018-09-17 08:06:48.202 <21
2018-09-17 08:06:48.206 <f4
2018-09-17 08:06:49.103 <42
2018-09-17 08:06:49.123 <21
2018-09-17 08:06:49.123 <f4
2018-09-17 08:06:49.222 <35
2018-09-17 08:06:49.228 <a1
2018-09-17 08:06:49.231 <f4
2018-09-17 08:06:50.129 <42
2018-09-17 08:06:50.134 <21
2018-09-17 08:06:50.138 <f4
2018-09-17 08:06:50.248 <33
2018-09-17 08:06:50.253 <21
etc..

pi@raspberry:/etc/ebusd $ ebusctl info
version: ebusd 3.2.v3.2
signal: acquired
symbol rate: 6
max symbol rate: 6
reconnects: 0
masters: 1
messages: 286
conditional: 0
poll: 0
update: 4
address 31: master #8, ebusd
address 36: slave #8, ebusd

pi@raspberry:/etc/ebusd $ ebusctl scan
empty

And now, I am lost...
As far as I understand I should have something more detected on the bus (the boiler...)
I guess it's because my boiler is not at all supported yet, but I can understand the next step .
Can you point me in right direction ?

Thanks in Advance,
/Klona

john30, 2018-09-17:
> 2018-09-17 08:06:47.180 <f4
2018-09-17 08:06:48.078 <42
2018-09-17 08:06:48.083 <21
2018-09-17 08:06:48.087 <f4
2018-09-17 08:06:48.197 <33
2018-09-17 08:06:48.202 <21
2018-09-17 08:06:48.206 <f4
2018-09-17 08:06:49.103 <42
2018-09-17 08:06:49.123 <21
2018-09-17 08:06:49.123 <f4
2018-09-17 08:06:49.222 <35
2018-09-17 08:06:49.228 <a1
2018-09-17 08:06:49.231 <f4
2018-09-17 08:06:50.129 <42
2018-09-17 08:06:50.134 <21
2018-09-17 08:06:50.138 <f4
2018-09-17 08:06:50.248 <33
2018-09-17 08:06:50.253 <21

this does not look like valid eBUS protocol data at all. Are you sure that the interface transfers the eBUS data as is?

klona, 2018-09-17:
Humm. Damned. I think you are right. I didnt realize that this hardware is doing opentherm on PIC and there is some sort of advanced commands through the serial management..

I think i will buy a eBus Koppler USB from Esera. Is that a fully compatible choice ?

/Klona

klona, 2018-09-17:
> @klona did you actually install the configuration files? For Saunier see [john30/ebusd-configuration#77](https://github.com/john30/ebusd-configuration/issues/77) fo one of the related investigative issues.

Thx for the link. Didnt see this one and it will clearly help me

john30, 2018-09-18:
see here: https://github.com/john30/ebusd/wiki/6.-Hardware

closed 2018-09-18 by john30