john30/ebusd #201: HTTP Client not working, error configpath without scanconfig!

ThoEngel, 2018-09-23:
Fresh installation on RPi, Telnet access works fine, but HTTP access no feedback.

If I config the httpport: 
sudo ebusd --httpport=8080

I get the following error message: "2018-09-23 10:44:57.824 [main error] invalid configpath without scanconfig"

What I am doing wrong? How to config configpath/scanconfig?

ThoEngel, 2018-09-23:
--> sudo ebusd --scanconfig[=ADDR]
Which value (ADDR) must be set? 
How I get the value: ADDR?

ThoEngel, 2018-09-23:
It is not possible to set -scanconfig ?!?

$ sudo ebusd --checkconfig --scanconfig
2018-09-23 14:26:11.931 [main notice] ebusd 3.2.v3.2 performing configuration check...
2018-09-23 14:26:12.166 [main notice] found messages: 11 (0 conditional on 0 conditions, 0 poll, 4 update)
2018-09-23 14:26:12.167 [main notice] ebusd stopped
$ sudo ebusd --scanconfig
$ sudo ebusd --httpport=8080
2018-09-23 14:30:21.387 [main error] invalid configpath without scanconfig
$ sudo ebusd --configpath="http://ebusd.eu/config/"
2018-09-23 14:38:05.981 [main error] invalid configpath without scanconfig
$ sudo ebusd --configpath="http://ebusd.eu/config/" --scanconfig
$ sudo ebusd --httpport=8080
2018-09-23 14:38:50.256 [main error] invalid configpath without scanconfig
$ sudo ebusd --checkconfig
2018-09-23 14:39:46.307 [main error] invalid configpath without scanconfig
$ sudo ebusd --dumpconfig
2018-09-23 14:40:43.009 [main error] invalid configpath without scanconfig

ThoEngel, 2018-09-23:
Maybe a little hint ?

ThoEngel, 2018-09-24:
Would you support for a donation. What does the support cost?

hvorragend, 2018-09-24:
First
https://github.com/john30/ebusd/wiki/4.8.-Probing-the-configuration
"With this option, ebusd will not fork into background and not open any device or network port,..."

Second
ADDR is optional. Not necessary.

hvorragend, 2018-09-24:
Third:
Try  --configpath=/etc/ebusd/config

john30, 2018-09-25:
I'm not eager to earn money with this project and the problem is just limited time, so please read the documents that are available.
basically you need to pass all options together to the call of ebusd, i.e. `sudo ebusd -f --scanconfig --httpport=8080`.
in your output provided before, the ebusd start was already working fine, so please make yourself bright by looking at the documentation and/or the command line help (`ebusd --help`) before digging around.
thanks @hvorragend for your help here, appreciated!

andig, 2018-09-25:
Nevertheless I‘ve found it good practice after intensively using ebusd for some time to make some donations as token of appreciation for all the work @john30 has put into this project.