john30/ebusd #202: Issue with writing to the Ebus

kolibrie-eric, 2018-09-24:
Hi,

I changed the definition of the SetMode message in hcmode.inc from a passive write to an active write. With Ebusctl I wrote the following message:

_Ebusctl write -s 10 -c hmu SetMode auto;0;-;-;1;1;1;0;0;0_

However the ebus daemon writes the following message to the bus:

_sent write hmu SetMode QQ=10: auto;0.0;-;-;0;0;1;0;0;0_

Notice the difference in the value string. I did some initial debugging and believe that the prepareMaster  function in message.cpp may contain a bug (called in the readFromBus function in the bus handler). The inputStr variable contains the original values. The output master object does not. Seems like an issue handling the BI type.

Would appreciate some help in correcting the issue.

Thx

john30, 2018-09-25:
did you escape the semicolon characters in the shell?

kolibrie-eric, 2018-09-25:
Yes I did. I checked with a debugger that the daemon actually receives the correct command.

I removed the boundary clause from the datatype.cpp, writeRawValue function, line 784 and then everything seems to be working again. Not sure what breaks though.

kolibrie-eric, 2018-10-01:
Ebusd version is 3.2. 
The raw logging output is _[bus notice] >1008b51009000000ffff00040000fb<0001019a>00_

The hex output is identical to _sent write hmu SetMode QQ=10: auto;0.0;-;-;0;0;1;0;0;0_ and not equal to the command I issued via EbusCtl

john30, 2018-10-01:
weird. please post the message definition as output from "ebusctl f -f -w -c hmu SetMode", just to be sure we're both looking at the same thing

kolibrie-eric, 2018-10-01:
Indeed, weird. The message definition I used is a modified version of the hcmode.inc one (changed it from passive to active):

_w,hmu,SetMode,Operation Mode,,08,b510,00,hcmode,m,UCH,0=auto;1=off;2=water,,boiler mode,flowtempdesired,m,D1C,,°C,temperature,hwctempdesired,m,D1C,,°C,temperature,hwcflowtempdesired,m,UCH,,°C,temperature,,m,IGN:1,,,,disablehc,m,BI0:1,,,,disablehwctapping,m,BI1:1,,,,disablehwcload,m,BI2:1,,,,,m,IGN:1,,,,remoteControlHcPump,m,BI0:1,,,,releaseBackup,m,BI1:1,,,,releaseCooling,m,BI2:1,,,_

john30, 2018-10-03:
thanks for finding this, that was indeed wrong for the use case write message with bit combination

commit 18bd21f2 (2018-10-03): fix for bit combinations during write to SymbolString (fixes #202)
closed 2018-10-03 by john30